fwupd/fwupd #729: Probably need to add threading :(

hughsie, 2018-09-07:
We probably need to do something I've been trying to avoid; running the transactions in a helper thread to avoid blocking the DBus interface. Updating my K780 today took 3 minutes, and that gave me several thousand of these lines in my session log:

    12:14:43:0994 As  run 0x7f36e4010850~GsPlugin::fwupd(gs_plugin_add_updates)
    12:14:44:0019 Gs  failed to get updates: Timeout was reached

I don't think we want to go full thread crazy; I think it only makes sense to update one device at a time -- but I do think we should service requests like `GetUpdates` and `GetDevices` whilst doing a long update. If we try to `Install` on a different device while another is installing I think it's perfectly file to return an error like `NOT_SUPPORTED`.

I've not looked at how this would work, but it would involve doing some planning before just jumping in. Comments welcome, and I'm certainly not going to block 1.1.2 on this.

superm1, 2018-09-07:
Of course adding threading introduces lots of risk for order of events.  I think to start out you should just have a concept of "Busy" and "Available".  

* Busy means servicing anything that can change data structures - not just `Install`.

While busy you only respond affirmatively to requests that provide static data that can't change from the command.  As an example you probably shouldn't respond to `GetUpdates` affirmatively while busy because `UpdateMetadata` might be changing that.  Similarly you shouldn't respond to `GetDevices` because `Install` might be causing devices to come and go during that time.

After that simple concept is in place and plumbed through I think more granularity can be introduced.

Whether to block 1.1.2 on this, how does G-S handle this other than those messages in logs?  Is there other bad effects?

hughsie, 2018-09-07:
> you probably shouldn't respond to GetUpdates affirmatively

Right, some care is required.

> how does G-S handle this other than those messages in logs

Just lots of CPU; g-s uses threads itself so although the main drawing thread isn't blocked we still do lots of pointless IO.

superm1, 2018-09-27:
Moving it to 1.1.4 so that 1.1.3 can be a stable roll up release

superm1, 2020-07-02:
I'm going to close this - we never came up with a true need for it..

closed 2020-07-02 by superm1