fwupd/fwupd #730: May need to move thunderbolt over to new daemon provided replug

superm1, 2018-09-07:
I was hoping to avoid doing this for 1.1.2, but I'm finding that if a Thunderbolt device is in the transaction the rest of the tree isn't back in time for composite_cleanup to work properly even with the commits from #728 which helped with just USB devices in the transaction

hughsie, 2018-09-07:
Couldn't we just do a blocking sleep at the end of the thunderbolt update() method?

superm1, 2018-09-07:
Actually it already blocks at the end waiting for the Thunderbolt device to come back.  Adding extra sleeping on top of that doesn't help because it's the other devices that disappear too.

The device IDs are invalid for other devices later in the transaction (I actually see an error that they're not found because usb_device_added didn't run again).

I'll experiment with some stuff in composite_prepare to set stuff up to wait for replug.

superm1, 2018-09-07:
OK I think i've got something that will work from `composite_prepare`, we can put off thunderbolt move to the new replug stuff until after 1.1.2.

commit 43b1578a (2018-09-24): thunderbolt: Use replugging from the daemon (Closes: #730)
commit 27446923 (2018-09-25): thunderbolt: Use replugging from the daemon (Closes: #730)
commit a3ae2229 (2018-09-25): thunderbolt: Use replugging from the daemon (Closes: #730)
commit e8b8af3d (2018-09-25): thunderbolt: Use replugging from the daemon (Closes: #730)
closed 2018-09-26 by superm1