fwupd/fwupd #737: cannot update on T480s, no space left on device?

marco-waven, 2018-09-11:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.1.1
daemon version:	1.1.1
compile-time dependency versions
	appstream-glib:	0.7.10
	gusb:	0.2.11
	efivar:	34
```

```shell
$ fwupdmgr get-devices
Thinkpad T480s Thunderbolt Controller
  DeviceId:             17844c332ef80285269c6762f47fd9c6434c2038
  Guid:                 061f2847-061b-5cfb-9c39-3dbfe2b49d69
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              12.02
  Icon:                 computer
  Created:              2018-09-11

20L7A01GHK System Firmware
  DeviceId:             e7cc234e82d08b6a14e246beb6b5f8142091e570
  Guid:                 ebfe8df8-dee7-4692-a721-cbcf5095c5cf
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.14
  VersionLowest:        0.0.1
  Icon:                 computer
  Created:              2018-09-11
  Modified:             2018-09-11
  UpdateState:          failed
  UpdateError:          {error #0} efivarfs.c:351 efivarfs_set_variable(): write failed: No space left on device	{error #1} lib.c:81 _efi_set_variable_mode(): ops->set_variable() failed: No space left on device: Error writing to file descriptor: No space left on device

UEFI Device Firmware
  DeviceId:             cf3ad1e5e6838b09cba64bb5ccb1142dd44995d2
  Guid:                 676af093-2a5c-4238-9c29-db8063a33532
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              184.50.3425
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-09-11

UEFI Device Firmware
  DeviceId:             323bda90831520867f8c02544efb9711dfca9fb6
  Guid:                 4d254d6e-cd67-477b-97d5-bc3048af45c4
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.8
  VersionLowest:        0.1.8
  Icon:                 audio-card
  Created:              2018-09-11

```

```shell
$ efibootmgr -v
BootCurrent: 0019
Timeout: 0 seconds
BootOrder: 0017,0018,0019,001A,001B,001C,001D,001E
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
[... 338 lines cut ...]
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initrd.img-4.17.0-3-amd64
├── System.map-4.17.0-3-amd64
└── vmlinuz-4.17.0-3-amd64

```

Please answer the following questions:

- Operating system and version:
 Debian testing
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
apt-get 
- Have you tried rebooting?
yes, try --offline then reboot
- Are you using an NVMe disk?
yes, only one NVMe disk
- Is secure boot enabled (only for the UEFI plugin)?
no

superm1, 2018-09-11:
Seems like efivarfs is what's out of space

hughsie, 2018-09-11:
Ohh, efivarfs indeed! I suppose this means the NV store is full?

superm1, 2018-09-11:
Either that or that boot order lock is set.  I would guess when boot order lock is set it's causing a similar behavior, but I don't know for sure.

chopin1998, 2018-09-11:
strange, when i do other reboot, and press F1 to BIOS to check settings, nothing changed, after this fwupdmgr work!

superm1, 2018-09-11:
Well I guess good that it works, i'll close

closed 2018-09-11 by superm1

lovesegfault, 2020-04-03:
I can reproduce this on my laptop, even though (and I made sure) boot order lock was not set. Entering and exiting the BIOS and then trying again seemed to fix it, although fwupd was still unable to correctly set the next boot order. I had to choose the right efi entry manually on reboot, and then it went okay.