fwupd/fwupd #743: thinkpad carbon x1 gen6: no firmware presented from `get-devices`

stuartnelson3, 2018-09-12:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.1.1
daemon version:	1.1.1
compile-time dependency versions
	appstream-glib:	0.7.10
	gusb:	0.3.0
	efivar:	35
```

```shell
$ fwupdmgr get-devices
X1 Carbon Thunderbolt Controller
  DeviceId:             e95521252b392e08d42ac4cb1082bf8fd1df0a1b
  Guid:                 5ff9544e-fce2-5fd9-9cd2-671c95ac91dc
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              30.00
  Icon:                 computer
  Created:              2018-09-12

```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,001A,0000,0018,0019,001B,001C,001D,001E,001F,0024
Boot0000* Windows Boot Manager	HD(1,GPT,c5f78380-8748-7049-859f-a274a340a993,0x800,0x32000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}..._................
Boot0001* arch_grub	HD(1,GPT,c5f78380-8748-7049-859f-a274a340a993,0x800,0x32000)/File(\EFI\arch_grub\grubx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Regulatory Information	FvFile(478c92a0-2622-42b7-a65d-5894169e4d24)
Boot0015  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0016  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0017  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0018* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0019* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot001A* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001B* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f601)
Boot001C* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001D* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001E  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001F  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0020* USBR BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0021* USBR BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0022* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0023* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0024* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
```

```shell
$ efivar -l | grep fw
no output
```

```shell
$ tree /boot
boot
├── efi
│   └── EFI
│       └── arch_grub
[... 337 lines cut ...]
│       ├── videotest.mod
│       ├── xfs.mod
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── lost+found [error opening dir]
└── vmlinuz-linux

```

Please answer the following questions:

- Operating system and version: arch linux
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? no

I apologize for my naivete -- I'm attempting to update the firmware on my machine, but it doesn't appear to be recognized (see output from `get-devices`). The disk on this laptop was dd'd from a gen5 carbon x1, without an NVMe disk.

hughsie, 2018-09-12:
What does `fwupdmgr get-devices --show-all-devices` show?

stuartnelson3, 2018-09-12:
```bash
$ fwupdmgr get-devices --show-all-devices
UHD Graphics 620
  DeviceId:             8de6c7959053fd5798006dcc63590d33fa5e51cb
  Guid:                 8eb8bd2e-0fca-5aba-9aa8-f341e0aa4482
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Icon:                 audio-card
  Created:              2018-09-12

X1 Carbon Thunderbolt Controller
  DeviceId:             e95521252b392e08d42ac4cb1082bf8fd1df0a1b
  Guid:                 5ff9544e-fce2-5fd9-9cd2-671c95ac91dc
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              30.00
  Icon:                 computer
  Created:              2018-09-12
```

hughsie, 2018-09-12:
What does `sudo /usr/libexec/fwupd/fwupdtool --verbose --plugin-whitelist=uefi get-devices --show-all-devices` display? Can you pastebin it somewhere please.

hughsie, 2018-09-12:
Ohh, and does `/sys/firmware/efi/esrt/entries` exist?

stuartnelson3, 2018-09-12:
`/sys/firmware/efi/esrt/entries` does not exist

stuartnelson3, 2018-09-12:
https://gist.github.com/stuartnelson3/a801fe447d23ff7b257617a98d7e9f30 seems to corroborate that `/sys/firmware/efi/esrt/entries` sounds relatively necessary

stuartnelson3, 2018-09-17:
If anyone stumbles across this issue:

I had turned off UEFI windows updates in the BIOS. Re-enabling this solved the issue immediately.