fwupd/fwupd #746: Lenovo T460s firware update not working in gnome-software or fwupdmgr

askb, 2018-09-13:
Unable to install firware update on Lenovo T460s using gnome-software or fwupdmgr. The update is installed successfully, but after rebooting you are on the same version of the firmware.

```shell
$ fwupdmgr --version
**client version: 1.0.8
daemon version: 1.0.8
compile-time dependency versions
        appstream-glib: 0.7.8
        gusb:   0.2.11
        fwupdate:       10
        efivar: 32
**
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
**UEFI Device Firmware
  DeviceId:             25127f5d4d5dfd6e903cda1a61a67cb092e4bb0f
  Guid:                 63451986-0c42-42db-9a18-342a8455ac68
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              176.12.1008
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-09-13

UEFI Device Firmware
  DeviceId:             33bc73bf0f48eb6177a007e9351e22d5bce2c04d
  Guid:                 ffaf3e12-a6ea-4352-851b-76c6455913c6
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.11
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-09-13

20F9CTO1WW System Firmware
  DeviceId:             444a229da2b21bc0d24faf814d1c58c5bff3315f
  Guid:                 42f29620-3d63-4f09-950b-9b7055570f28
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot                                                                                                                            
  Version:              0.1.25                                                                                                                                                                                     
  VersionLowest:        0.1.22                                                                                                                                                                                     
  Icon:                 computer                                                                                                                                                                                   
  Created:              2018-09-13                                                                                                                                                                                 
                                  **
```

```shell
$ efibootmgr -v
**BootCurrent: 0001                                                                                                                                                                                                  
Timeout: 2 seconds                                                                                                                                                                                                 
BootOrder: 001B,0010,0011,0012,0013,0000,0017,0018,0019,001A,001C,0001                                                                                                                                             
Boot0000* Fedora        HD(1,GPT,d15697c9-00b0-4e50-9f67-6b09e1090853,0x800,0x64000)/File(\EFI\fedora\shim.efi)                                                                                                    
Boot0001* Linux-Firmware-Updater \fwupx64.efi   HD(1,GPT,d15697c9-00b0-4e50-9f67-6b09e1090853,0x800,0x64000)/File(\EFI\fedora\shimx64.efi)\.f.w.u.p.x.6.4...e.f.i...                                               
Boot0010  Setup FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu     FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen      FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics    FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu        FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery   FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key  FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD        VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0 VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PCI LAN       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D* IDER BOOT CDROM       PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,1,0)
Boot001E* IDER BOOT Floppy      PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,0,0)
Boot001F* ATA HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0020* ATAPI CD      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
**
```

```shell
$ efivar -l | grep fw
n/a
```

```shell
$ tree /boot
**/boot
├── config-4.17.14-102.fc27.x86_64
├── config-4.17.6-100.fc27.x86_64
├── config-4.17.9-100.fc27.x86_64
├── efi [error opening dir]
├── elf-memtest86+-5.01
├── extlinux
│   ├── cat.c32
│   ├── chain.c32
│   ├── cmd.c32
│   ├── cmenu.c32
│   ├── config.c32
│   ├── cptime.c32
│   ├── cpu.c32
│   ├── cpuid.c32
│   ├── cpuidtest.c32
│   ├── debug.c32
│   ├── dhcp.c32
│   ├── dir.c32
│   ├── disk.c32
│   ├── dmi.c32
│   ├── dmitest.c32
│   ├── elf.c32
│   ├── ethersel.c32
│   ├── gfxboot.c32
│   ├── gpxecmd.c32
│   ├── hdt.c32
│   ├── hexdump.c32
│   ├── host.c32
│   ├── ifcpu64.c32
│   ├── ifcpu.c32
│   ├── ifmemdsk.c32
│   ├── ifplop.c32
│   ├── kbdmap.c32
│   ├── kontron_wdt.c32
│   ├── ldlinux.c32
│   ├── lfs.c32
│   ├── libcom32.c32
│   ├── libgpl.c32
│   ├── liblua.c32
│   ├── libmenu.c32
│   ├── libutil.c32
│   ├── linux.c32
│   ├── ls.c32
│   ├── lua.c32
│   ├── mboot.c32
│   ├── memdisk
│   ├── meminfo.c32
│   ├── menu.c32
│   ├── pci.c32
│   ├── pcitest.c32
│   ├── pmload.c32
│   ├── poweroff.c32
│   ├── prdhcp.c32
│   ├── pwd.c32
│   ├── pxechn.c32
│   ├── reboot.c32
│   ├── rosh.c32
│   ├── sanboot.c32
│   ├── sdi.c32
│   ├── sysdump.c32
│   ├── syslinux.c32
│   ├── vesa.c32
│   ├── vesainfo.c32
│   ├── vesamenu.c32
│   ├── vpdtest.c32
│   ├── whichsys.c32
│   └── zzjson.c32
├── grub2 [error opening dir]
├── initramfs-0-rescue-61bf28336b574f179d9d82961bb8c6f8.img
├── initramfs-4.17.14-102.fc27.x86_64.img
├── initramfs-4.17.6-100.fc27.x86_64.img
├── initramfs-4.17.9-100.fc27.x86_64.img
├── initrd-plymouth.img
├── lost+found [error opening dir]
├── memtest86+-5.01
├── System.map-4.17.14-102.fc27.x86_64
├── System.map-4.17.6-100.fc27.x86_64
├── System.map-4.17.9-100.fc27.x86_64
├── vmlinuz-0-rescue-61bf28336b574f179d9d82961bb8c6f8
├── vmlinuz-4.17.14-102.fc27.x86_64
├── vmlinuz-4.17.6-100.fc27.x86_64
└── vmlinuz-4.17.9-100.fc27.x86_64

**
```

Please answer the following questions:

- Operating system and version: Fedora 27 x86
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): dnf
- Have you tried rebooting? yes 
- Are you using an NVMe disk? no
- Is secure boot enabled (only for the UEFI plugin)? no

hughsie, 2018-09-13:
> Operating system and version: Fedora 27 x86

Can you try on F28 please? F27 is only just supported and hasn't had any fwupd love for some time. Thanks.

ginkel, 2018-09-16:
@askb Do you have the "Lock boot order" setting enabled in UEFI? That prevents fwupdmgr from booting into the updater...

askb, 2018-11-03:
@ginkel I don't have the lock boot order setting enable in UEFI.  
@hughsie Thanks for the suggestion, upgrading to F28 fixed the issue with `fwupdmgr`.

closed 2018-11-03 by askb