fwupd/fwupd #748: ThinkPad T470p UEFI update cleared custom boot entries

ginkel, 2018-09-16:
```shell
$ fwupdmgr --version
client version: 1.1.1
daemon version: 1.1.1
compile-time dependency versions
        appstream-glib: 0.7.10
        gusb:   0.3.0
        efivar: 35
```

```shell
$ fwupdmgr get-devices
**INSERT OUTPUT HERE**
```

```shell
$ efibootmgr -v
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
```

```shell
$ efivar -l | grep fw
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
```

```shell
$ tree /boot
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version:
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
- Have you tried rebooting?
- Are you using an NVMe disk?
- Is secure boot enabled (only for the UEFI plugin)?

ginkel, 2018-09-16:
Located https://github.com/hughsie/fwupd/issues/709, guess this issue is a duplicate...

hughsie, 2018-09-17:
I think so, lets continue there.

closed 2018-09-17 by hughsie