fwupd/fwupd #749: Failed to download https://cdn.fwupd.org/downloads/firmware.xml.gz: SSL handshake failed

thiagomacieira, 2018-09-17:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version: 1.1.2
daemon version: 1.1.2
compile-time dependency versions
        appstream-glib: 0.7.12
        gusb:   0.2.11
        efivar: 35
```

Please answer the following questions:

- Operating system and version: Clear Linux 25050
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): swupd bundle-add firmware-upgrade
- Have you tried rebooting? Yes

Symptoms:
```shell
# systemctl status fwupd.service
● fwupd.service - Firmware update daemon
   Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static; vendor preset: disabled)
   Active: active (running) since Mon 2018-09-17 12:49:00 PDT; 24s ago
     Docs: https://fwupd.org/
 Main PID: 727 (fwupd)
    Tasks: 5 (limit: 4915)
   Memory: 15.7M
   CGroup: /system.slice/fwupd.service
           └─727 /usr/libexec/fwupd/fwupd

Sep 17 12:48:59 tjmaciei-desk systemd[1]: Starting Firmware update daemon...
Sep 17 12:48:59 tjmaciei-desk fwupd[727]: failed to load remote fwupd: Failed to parse /usr/share/fwupd/remotes.d/fwupd/metadata.xml file: cann>
Sep 17 12:49:00 tjmaciei-desk fwupd[727]: disabling plugin because: failed to startup dell: Firmware updating not supported
Sep 17 12:49:00 tjmaciei-desk fwupd[727]: disabling plugin because: failed to startup redfish: no SMBIOS table
Sep 17 12:49:00 tjmaciei-desk fwupd[727]: disabling plugin because: failed to startup dell_esrt: UEFI firmware already supported
Sep 17 12:49:00 tjmaciei-desk fwupd[727]: disabling plugin because: failed to coldplug amt: ME refused connection
Sep 17 12:49:00 tjmaciei-desk fwupd[727]: Daemon ready for requests
Sep 17 12:49:00 tjmaciei-desk systemd[1]: Started Firmware update daemon.
# fwupdmgr get-updates       
Firmware metadata has not been updated for 30 days and may not be up to date.

Update now? (Requires internet connection) [y|N]: y
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz

Failed to download https://cdn.fwupd.org/downloads/firmware.xml.gz: SSL handshake failed
```

Downloading the file with both curl and wget works without problems. Connecting to cdn.fwupd.org with `openssl s_client` works and presents "Verify return code: 0 (ok)".

Stracing fwupdmgr shows:
```
[pid  1038] socket(AF_INET, SOCK_STREAM|SOCK_CLOEXEC, IPPROTO_IP) = 7<TCP:[17930]>
[pid  1038] fcntl(7<TCP:[17930]>, F_GETFL) = 0x2 (flags O_RDWR)
[pid  1038] fcntl(7<TCP:[17930]>, F_SETFL, O_RDWR|O_NONBLOCK) = 0
[pid  1038] connect(7<TCP:[17930]>, {sa_family=AF_INET, sin_port=htons(443), sin_addr=inet_addr("38.95.109.110")}, 16) = -1 EINPROGRESS (Operation now in progress)
[pid  1038] eventfd2(0, EFD_CLOEXEC|EFD_NONBLOCK) = 8<anon_inode:[eventfd]>
[pid  1038] poll([{fd=7<TCP:[17930]>, events=POLLOUT}, {fd=8<anon_inode:[eventfd]>, events=POLLIN}], 2, 60000) = 1 ([{fd=7, revents=POLLOUT}])
[pid  1038] close(8<anon_inode:[eventfd]>) = 0
[pid  1038] getsockopt(7<TCP:[17930]>, SOL_SOCKET, SO_ERROR, [0], [4]) = 0
[pid  1038] setsockopt(7<TCP:[17930]>, SOL_TCP, TCP_NODELAY, [1], 4) = 0
[pid  1038] futex(0x7f4039e07a78, FUTEX_WAKE_PRIVATE, 2147483647) = 0
[pid  1038] futex(0x7f4039e07a78, FUTEX_WAKE_PRIVATE, 2147483647) = 0
[pid  1038] futex(0x7f4039e07a78, FUTEX_WAKE_PRIVATE, 2147483647) = 0
[pid  1038] futex(0x7f4039e07a78, FUTEX_WAKE_PRIVATE, 2147483647) = 0
[pid  1038] futex(0x7f4039e07a78, FUTEX_WAKE_PRIVATE, 2147483647) = 0
[pid  1038] futex(0x7f4039e07a78, FUTEX_WAKE_PRIVATE, 2147483647) = 0
[pid  1038] futex(0x7f4039e07a78, FUTEX_WAKE_PRIVATE, 2147483647) = 0
[pid  1038] futex(0x7f4039e07a78, FUTEX_WAKE_PRIVATE, 2147483647) = 0
[pid  1038] futex(0x7f4039e07a78, FUTEX_WAKE_PRIVATE, 2147483647) = 0
[pid  1038] futex(0x7f4039e07a78, FUTEX_WAKE_PRIVATE, 2147483647) = 0
[pid  1038] futex(0x7f4039e07a78, FUTEX_WAKE_PRIVATE, 2147483647) = 0
[pid  1038] futex(0x7f4039e07a78, FUTEX_WAKE_PRIVATE, 2147483647) = 0
[pid  1038] futex(0x7f4039e07a78, FUTEX_WAKE_PRIVATE, 2147483647) = 0
[pid  1038] close(7<TCP:[17930]>)       = 0
```

That shows a non-blocking connection attempt, with a poll showing that the socket is ready for writing (i.e. connected). The `getsockopt` for `SO_ERROR` appears to be indicating error 4, which is EINTR (that is not a fatal problem). After that, the connection is closed with no data exchanged.

superm1, 2018-09-17:
@thiagomacieira I would _suspect_ you are hitting the same problem I saw on the snap that you're missing gnutls libraries in your container.

To confirm this you can add libsoup debugging to the client's request to download metadata and recompile the client.  When I did this on my side in the snap I found missing gnutls libraries was the message at least.

On the snap side this is solved by copying over the gio modules to the user's home directory outside of the sandbox.
https://github.com/hughsie/fwupd/blob/master/contrib/snap/fwupd-command#L31

thiagomacieira, 2018-09-17:
This is not a container, this is the minimal root system that runs the containers. The only user on this machine is root and it runs no desktop (no GIO).

GnuTLS 3.6.3 libraries are present:
```
# ls -l /usr/lib64/libgnutls.so.30*
lrwxrwxrwx 1 root root      20 Jul 27 12:58 /usr/lib64/libgnutls.so.30 -> libgnutls.so.30.21.0
-rwxr-xr-x 1 root root 1764752 Jul 27 12:58 /usr/lib64/libgnutls.so.30.21.0
```

`gnutls-cli` is not present on the root system. If I run it from one of the containers, I get:
```
Error setting the x509 trust file
Resolving 'cdn.fwupd.org:443'...
Connecting to '38.95.109.110:443'...
- Certificate type: X.509
- Got a certificate list of 2 certificates.
- Certificate[0] info:
 - subject `CN=cdn.fwupd.org', issuer `CN=Let's Encrypt Authority X3,O=Let's Encrypt,C=US', serial 0x039b62a8976df66720ef34999db415fc8792, RSA key 2048 bits, signed using RSA-SHA256, activated `2018-09-12 01:00:18 UTC', expires `2018-12-11 01:00:18 UTC', pin-sha256="cJ671cGQnTcrU2rY8wHzcfVhB9kp8T9V9DxNH2lWhJU="
        Public Key ID:
                sha1:952df2f4a80f1561f00d4e4b0d2ef825daf30cb8
                sha256:709ebbd5c1909d372b536ad8f301f371f56107d929f13f55f43c4d1f69568495
        Public Key PIN:
                pin-sha256:cJ671cGQnTcrU2rY8wHzcfVhB9kp8T9V9DxNH2lWhJU=

- Certificate[1] info:
 - subject `CN=Let's Encrypt Authority X3,O=Let's Encrypt,C=US', issuer `CN=DST Root CA X3,O=Digital Signature Trust Co.', serial 0x0a0141420000015385736a0b85eca708, RSA key 2048 bits, signed using RSA-SHA256, activated `2016-03-17 16:40:46 UTC', expires `2021-03-17 16:40:46 UTC', pin-sha256="YLh1dUR9y6Kja30RrAn7JKnbQG/uEtLMkBgFF2Fuihg="
- Status: The certificate is NOT trusted. The certificate issuer is unknown. 
*** PKI verification of server certificate failed...
*** Fatal error: Error in the certificate.
*** handshake has failed: Error in the certificate.
```

I'll investigate why it isn't finding any CAs, but at least it's exchanging data.

superm1, 2018-09-17:
So fwupd uses libsoup and libsoup uses glib-networking for SSL/TLS support which does require GIO.  If you don't have all that in place you'll hit the failures you're seeing.

Adding the soup logger is really straightforward, and it will get you a more verbose message on the failure. https://developer.gnome.org/libsoup/stable/SoupLogger.html

superm1, 2018-09-17:
https://github.com/GNOME/libsoup/blob/master/configure.ac#L178

thiagomacieira, 2018-09-17:
I've just confirmed that our GnuTLS is configured without a system CA configuration. But indeed, the glib-networking modules are missing:

```
# ls /usr/lib64/gio/modules/libgiognutls.so
ls: cannot access '/usr/lib64/gio/modules/libgiognutls.so': No such file or directory
```

I'm guessing this is also an issue. This one I can easily fix, though.

thiagomacieira, 2018-09-18:
> https://github.com/GNOME/libsoup/blob/master/configure.ac#L178

libsoup is built with glib-networking support, but there was no dependency in the plugins, so they didn't get deployed. I've added a manual "Requires" now.

superm1, 2018-09-18:
So it's working now with that in place?

thiagomacieira, 2018-09-18:
Changes making through the review process and will make it into the distro soon. I'll test then.

thiagomacieira, 2018-10-09:
Found it: missing gnupg binaries.