fwupd/fwupd #758: 8bitdo SN30 (SF30) Pro upgrade not working on Ubuntu 18.04.1

picandocodigo, 2018-09-22:
I'm using a SN30 Pro controller and tried upgrading the firmware with fwupd with no success. I am able to pair the controller via Bluetooth, and when I press L + R + Start to enter "update mode" and plug it in, fwupdmgr sees the device.

I've ran `refresh`, `get-updates`, but have had no success so far. The GNOME Software frontend doesn't show any possible updates either.

Here's the data for Ubuntu's apt-get version:

```shell
$ fwupdmgr --version
client version: 1.0.6
daemon version: 1.0.6
compile-time dependency versions
        appstream-glib: 0.7.4
        gusb:   0.2.11
        fwupdate:       10
        efivar: 34
```

```shell
$ fwupdmgr get-devices
8Bitdo  
  DeviceId:             17d603463cd8e941ab0f36b885fcd7c8b23a0acd
  Guid:                 b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e
  Guid:                 78acf201-a624-5c0b-a521-e204be14e3d5
  Summary:              A redesigned classic game controller
  Plugin:               ebitdo
  Flags:                updatable|registered
  Vendor:               8Bitdo
  VendorId:             USB:0x2DC8
  Version:              1.03
  Icon:                 input-gaming
  Created:              2018-09-22
```

- Operating system and version: Ubuntu 18.04.1
- How did you install fwupd `apt-get`
- Have you tried rebooting? Yes
- Are you using an NVMe disk? No idea
- Is secure boot enabled (only for the UEFI plugin)? I guess so

I then installed a newer version via snap:
```
$ sudo snap install fwupd --classic                                                                                                                                                                                
fwupd 1.1.2-5-gb41b2d7 from Richard Hughes installed
```

Still being recognized:
```
8Bitdo  
  DeviceId:             71413822e87deae707da948bd4b7e537af463c19
  Guid:                 78acf201-a624-5c0b-a521-e204be14e3d5
  Guid:                 b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e
  Guid:                 5bc60e62-f461-58bd-91b6-7de3392f8bd6
  Summary:              A redesigned classic game controller
  Plugin:               ebitdo
  Flags:                updatable|registered|is-bootloader
  Vendor:               8Bitdo
  VendorId:             USB:0x2DC8
  Version:              1.03
  Icon:                 input-gaming
  Created:              2018-09-22
```
However no updates available:
```
$ fwupd.fwupdmgr refresh
$ fwupd.fwupdmgr get-updates
```

Am I missing something? Thanks!

JohnHouwer, 2018-09-23:
IIRC you need to connect the controller in the normal mode and try to start a firmware update. If you connect the controller in the update mode fwupd can not recognize which controller is connected.

In normal mode with ```get-devices``` should list the specific device and not just 8Bitdo.

fwupd should lead you through the process of disconnecting, booting into update mode and reconnection.

picandocodigo, 2018-09-23:
If I turn on the controller normally, pair it up with bluetooth to my computer, and plug the usb, `get-devices` doesn't list it.

I can use it to play games normally this way, and the bluetooth menu sees it correctly:
[image]
But `fwupdmgr get-devices` doesn't find it. I've only seen it listed when I plug it in "update mode" (L + R + Start).

JohnHouwer, 2018-09-23:
The controller has different modes. I suggest you try them out and see if fwupd recognizes one.

http://download.8bitdo.com/Manual/Controller/SN30pro+SF30pro/SN30pro+SF30pro_Manual.pdf

superm1, 2018-09-24:
Yes that's correct @JohnHouwer fwupd will not recognize all the modes.  We actually originally had it recognizing the VID/PID in more modes but it caused problems with both real Nintendo Switch controllers and Playstation DS4 contollers.
These two commits reverted those modes.
https://github.com/hughsie/fwupd/commit/ead5bf39295ac10589f1e26133f4bc9e955a956c#diff-15bfdb0aa197b6fb4d9f4183e53f43be
https://github.com/hughsie/fwupd/commit/86b3bcf8c9e6c468eca4d64976bc3feca3935eb0#diff-15bfdb0aa197b6fb4d9f4183e53f43be

I believe it should work if you start in Android mode (start +B) or Windows mode (start +X).  Admittedly I haven't personally tested this in a while though on my controller so it's entirely possible there is another bug that has cropped up.

picandocodigo, 2018-09-27:
I eventually got around to updating it. I had to use the Android mode for fwupd to detect it. When trying to update it, it told me to turn it off and restart in update mode. However, when I did this, it didn't find it again. It did show up at some point in GNOME's Software frontend and I tried updating it from there. It failed, so I tried again until eventually it worked and fwupd reported the firmware as being updated.

The controller was proper updated since I'm not having the issue I had when using it with Nintendo Switch where it would go crazy with the rumbling.

Not sure exactly when it got updated, but at least connecting it in Android mode allowed fwupd to download the new firmware. There might be some bug there, would it be possible to reproduce this update now that it's up to date or do I have to wait for a new firmware version?

Just let me know if there's anything else I can do to help with this. Also, the information about connecting it with Android mode would be useful for other users. Would it be worth adding to the wiki or somewhere else?

Thanks!

picandocodigo, 2018-09-27:
Ah, didn't know about the 20s timeout, it could be related to that but I'm not sure. I think it would make sense to bump the delay anyway, since it takes a few seconds to get the controller started with L + R + Start.

Would it be possible to downgrade the firmware in the controller and try again to reproduce this issue?

I'll start a wiki page for 8BitDo controllers and add the information for the model I'm using and if there are specifics for other models, they could be added there. But needing to try different modes and the `L + R + Start` update mode info is a good starting point.

superm1, 2018-09-27:
I'll send a PR to update to a longer timeout.

Yes, you can try to run the "downgrade" command to reproduce the issue if older firmware is published to LVFS.  I don't know off hand if there is any.

Yes, thanks for starting the wiki page, this will be a big help to newcomers.

picandocodigo, 2018-09-27:
@superm1 I tried to downgrade the firmware but apparently it's not available:
```
$ fwupdmgr downgrade 71413822e87deae707da948bd4b7e537af463c19
No downgrades for device, current is 1.26: 1.26=same
```
But if a new firmware is released, I'll make sure to write down everything. I created the wiki page:
https://github.com/hughsie/fwupd/wiki/8BitDo-Controllers
Please let me know if it's ok or if it needs any changes. It didn't look like there was a template or convention for wiki pages, so I'm sorry if it's missing anything or it's just too informal. I'll make sure to add anything new if we get a new upgrade for 8BitDo controller.

picandocodigo, 2018-09-28:
@superm1 Fixed! Thanks, and I'll come back with a new issue or just update the wiki if there's any updates :+1:

brunofisico, 2019-03-23:
Picandocodigo me ajude a instalar esse snes30 da 8bitdo no ubuntu. Uso o Snex para emular mas não consegui de jeito nenhum

commit 559c52ff (2018-10-01): trivial: Update FU_DEVICE_REMOVE_DELAY_USER_REPLUG from 20s to 40s (Fixes: #758)

Sometimes it's difficult to get 8bitdo controllers to turn off, so
give users more time to do it.