fwupd/fwupd #759: X260 fails to update Bios to 1.40

JohnHouwer, 2018-09-23:
```shell
$ fwupdmgr --version
client version:	1.1.2
daemon version:	1.1.2
compile-time dependency versions
	appstream-glib:	0.7.9
	gusb:	0.2.11
	efivar:	35

```
```shell
$ fwupdmgr get-devices
20F6CTO1WW System Firmware
  DeviceId:             10feec6a8f9695002c32cfdfdc9280edd6053635
  Guid:                 29ca1559-e603-4237-ba5c-2c55348da93e
  Guid:                 73d77ea2-d010-5f9d-959c-381f9decebcd
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.34
  VersionLowest:        0.1.26
  Icon:                 computer
  Created:              2018-09-23
  Modified:             2018-09-23
  UpdateState:          needs-reboot

UEFI Device Firmware
  DeviceId:             13b5499ef61d8a23e035c93e9d3bc207b3c217c4
  Guid:                 cda568a1-82ce-41f0-8e83-415a9ab3eb8c
  Guid:                 b56395a2-8414-5b52-b4f3-db6f0fb75b1e
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              176.0.1202
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-09-23

UEFI Device Firmware
  DeviceId:             89e6cace961ff992547647321186cf515bd8e24e
  Guid:                 f77494ca-da28-4789-a35b-fe7515ce8b42
  Guid:                 e299f8bd-9de1-5c82-8149-a180388e0119
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.12
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-09-23

```

```shell
$ efibootmgr -v
BootNext: 0000
BootCurrent: 0027
Timeout: 2 seconds
BootOrder: 0027,0017,0018,0019,001A,001B,001C,0000
Boot0000* Linux Firmware Updater	HD(2,GPT,cca5a441-63d4-4f2d-ab94-49d8a50d2a58,0x1800,0x40000)/File(\EFI\gentoo\fwupdx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
[... 74 lines cut ...]
Update Name:             ThinkPad X260
Update Summary:          Lenovo ThinkPad X260 System Firmware
Update Remote ID:        lvfs
Update Checksum:         SHA1(5e3376e8dc1eea18ad2236a72496be55cdeb7ea4)
Update Location:         https://fwupd.org/downloads/f9abd647920da7d0b9d178ed3bfb93516cc9e390-Lenovo-ThinkPad-X260-SystemFirmware-1.40.cab
Update Description:      Lenovo ThinkPad X260 System Firmware
                         
                         CHANGES IN THIS RELEASE
                         
                         Version 1.40
                         
                         [Important updates]
                          • N/A
                         
                         [New functions or enhancements]
                          • Updated the Diagnostics module to version 04.00.001.
                          • Add boot path for Debian Linux
                         
                         [Problem fixes]
                          • Fixed a mismatch issue between the Intel(rainbow) ME configuration and FPF fuse settings.
                          • Fixed an issue where BIOS POST might display the error message "Boot Manager recover from an error."
```

The System starts the efi image, and prints to above errors, after that it reboots to my normal OS.
I could use the Lenovo ISO to update, but I like to help track this bug down.

grexe, 2019-01-05:
same here with Lenovo V510–15IKB and fwupdate 12, latest suitable BIOS (extracted cap from ISO) from Nov/2018 on Manjaro/Arch x64, Linux kernel 4.20, UEFI else works fine.

lamarios, 2019-03-01:
Same issue with a t480 for firmware 1.21

```
compile-time dependency versions
	gusb:	0.3.0
	efivar:	35
daemon version:	1.2.5
```

joakim-tjernlund, 2019-03-05:
"efivar:	35"
This is a bad release I think, upgrade to efivar 37

superm1, 2019-03-05:
Great point, there are plenty of bugs with the known issues with efivar 35.  Upgrade to 37.  If issues persist, re-open with new logs.

closed 2019-03-05 by superm1