fwupd/fwupd #765: Allow releases to advertise expected update duration?

superm1, 2018-09-25:
I was thinking about how some devices are known to take more than a few seconds to update and we recently introduced support for showing a time estimate (in minutes) per-device while they update.

For devices that are external I think it might be useful to advertise in advance something like `Updates to this device will take approximately X minutes`.  This information seems like it's reasonable to encode either in the metainfo contained in a CAB file or can be stored with the `FwupdDevice` and `FuDevice` and set up by the plugins.

Thoughts?

hughsie, 2018-09-25:
Would the time "expected" depend on the CPU speed, the interface (e.g. USBv2 or USBv3), or anything else not within our control? I think this kind of metadata is useful, but only if it's semi-accurate.

superm1, 2018-09-25:
Given fwupd doesn't have full control of CPU scheduling, bus saturation, etc I think it's impossible to predict an "accurate" number.  What I'm thinking is something that is under normal circumstances semi-accurate.  An average CPU load, correct bus etc.  The frontend can still calculate an estimate as the update is in progress via throughput and if those factors are off then it would give a more accurate representation.

I'd like the unit stored in the metadata to be seconds, but then let the frontend decide how it wants to display it with appropriate rounding.  For example if the device said it was "65 seconds to update" the frontend might round up and say "< 2 minutes to update"

hughsie, 2018-09-25:
Would you include the time to reboot, POST, apply the uefi update, shutdown, POST and get back to the session? For some hardware the POST takes several minutes...

superm1, 2018-09-25:
Right I think this is more useful for "live" updates that fwupd manages the full lifecycle of the device's update.  UEFI updates you start to factor in too much outside stuff and probably shouldn't be included.

hughsie, 2018-09-25:
I'm okay with the idea, and I think a typical number of seconds would be the easiest format -- but I'm unsure where this would be stored in the AppStream file...

superm1, 2018-09-25:
I'm tending to think this should be done in the plugin/quirk rather than in the release itself.  New payloads shouldn't take a variable amount of time for the same device.

YehezkelShB, 2018-09-25:
Unless the payload size changed significantly, of course.

superm1, 2018-09-25:
Right, but so the device should have a concept of how big it's SPI/EEPROM/ flash region is and be able to estimate based on that I would think?

YehezkelShB, 2018-09-25:
Thinking of it again:
Most of the devices will probably just have a single estimation and will never change it (at least not for the specific model). For the special cases, it seems enough to allow the estimation to be per version / range of versions (e.g. 15 seconds for versions until 5.04, 20 seconds since version 5.05, for example).

superm1, 2018-09-25:
> For the special cases, it seems enough to allow the estimation to be per version / range of versions (e.g. 15 seconds for versions until 5.04, 20 seconds since version 5.05, for example).

Right and so that can be done in code (or better; via code that read quirks).

hughsie, 2018-09-25:
> via code that read quirks

100% agree.

commit 0e6a3992 (2018-09-25): Add support for devices to show an estimated flash time (Closes: #765)

This time is provided to frontends to be able to show estimates
before a user would agree to flash the device.
commit 9b972d7d (2018-09-25): Add support for devices to show an estimated flash time (Closes: #765)

This time is provided to frontends to be able to show estimates
before a user would agree to flash the device.
commit 8cecd4c0 (2018-09-25): Add support for devices to show an estimated flash time (Closes: #765)

This time is provided to frontends to be able to show estimates
before a user would agree to flash the device.
commit e02680bf (2018-09-25): Add support for devices to show an estimated flash time (Closes: #765)

This time is provided to frontends to be able to show estimates
before a user would agree to flash the device.
closed 2018-09-26 by hughsie