fwupd/fwupd #767: Can't update logitech unifying receiver

jaresty, 2018-09-27:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version: 1.1.2
daemon version: 1.1.2
compile-time dependency versions
        appstream-glib: 0.7.12
        gusb:   0.3.0
        efivar: 35

```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
fitlet2 System Firmware
  DeviceId:             8c7c8f687a2a7a80b4c50e575d0fd84762edf797
  Guid:                 00000000-0000-0000-0000-000000000000
  Guid:                 a6924b3b-5af6-5f8a-83bc-40100e865b90
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.0.40
  VersionLowest:        0.0.40
  Icon:                 computer
  Created:              2018-09-25

Unifying Receiver
  DeviceId:             dfae5e57c0357cad4c2423cd787bde43114ae1e9
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Guid:                 21e75d9a-5ce6-5da2-b7ab-910c7f3f6836
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR12.09_B0030
  VersionBootloader:    BOT01.04_B0016
  Icon:                 preferences-desktop-keyboard
  Created:              2018-09-25

Jabra SPEAK 410 USB
  DeviceId:             73ef80b60058b4f18549921520bfd94eaf18710a
  Guid:                 f884081f-f58f-5d01-86e8-dc12c88ef073
  Guid:                 537f7800-8529-5656-b2fa-b0901fe91696
  Guid:                 1764c519-4723-5514-baf9-3b42970de487
  Guid:                 93412def-aba7-50ee-9e2e-cf6925047014
  Plugin:               dfu
  Flags:                updatable|registered
  VendorId:             USB:0x0B0E
  Version:              1.12
  Icon:                 drive-harddisk-usb
  Created:              2018-09-25

```

```shell
$ efibootmgr -v

zsh: command not found: efibootmgr

**This is only required if you use the UEFI plugin**
```

[... 351 lines cut ...]
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initramfs-linux-fallback.img
├── initramfs-linux.img
└── vmlinuz-linux

8 directories, 345 files

```

Please answer the following questions:

- Operating system and version: Arch Linux (latest)
- How did you install fwupd? pacman
- Have you tried rebooting? yes
- Are you using an NVMe disk? no
- Is secure boot enabled (only for the UEFI plugin)? no

Every time I run `fwupdmgr update` it tells me `Not compatible with bootloader version: failed predicate [BOT01.0[0-3]_* regex BOT01.04_B0016]
`.

Blackjack92, 2018-10-25:
Same problem here:
```
fwupdmgr --version

client version: 1.0.6
daemon version: 1.0.6
compile-time dependency versions
        appstream-glib: 0.7.4
        gusb:   0.2.11
        fwupdate:       10
        efivar: 34
```

System

```
Operating System: KDE neon 5.14
KDE Plasma Version: 5.14.1
Qt Version: 5.11.2
KDE Frameworks Version: 5.51.0
Kernel Version: 4.15.0-38-generic
OS Type: 64-bit
Processors: 4 × Intel® Core™ i5-7300U CPU @ 2.60GHz
Memory: 11,6 GiB of RAM
```

AnessZurba, 2018-11-21:
same problem here:

```
$ /snap/bin/fwupdmgr --version
client version:	1.1.4
checkout info:	1.1.4
compile-time dependency versions
	appstream-glib:	0.7.9
	gusb:	0.3.0
	efivar:	35
daemon version:	1.1.4

```
System:

Ubuntu 18.04 LTS
Memory: 5.8 GiB
CPU: Intel® Core™ i7-6700HQ CPU @ 2.60GHz × 4 
OS Type: 64-bit
virtual machine through VMware

faced the same issue also on lenovo laptop running the same version of ubuntu on bare hardware (without VMware)

ollie1, 2019-01-12:
Same problem here - I bricked two unifying receivers trying to update using Ubuntu's Software Updater.  This tool fixed one of them (the newer one) when I'd almost given up and was about to buy a new one - thanks so much!  But it refuses to update my older device - I get the following outputs:
```bash
$ fwupdmgr --version
client version:	1.1.2
daemon version:	1.1.2
compile-time dependency versions
	appstream-glib:	0.7.12
	gusb:	0.2.11
	efivar:	34

$ fwupdmgr get-devices
Unifying Receiver
  DeviceId:             654a303a1e04fc691939521f42e5a306fdc28433
  Guid:                 a160f16a-613b-5627-a224-ace42013f9a6
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Guid:                 941f4133-39e5-56a0-9df6-902547436c5c
  Guid:                 21e75d9a-5ce6-5da2-b7ab-910c7f3f6836
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Summary:              A miniaturised USB wireless receiver (bootloader)
  Plugin:               unifying
  Flags:                updatable|supported|registered|is-bootloader
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR00.00_B0000
  VersionBootloader:    BOT01.02_B0014
  Icon:                 preferences-desktop-keyboard
  Created:              2019-01-12

$ fwupdmgr update
Not compatible with bootloader version: failed predicate [BOT03.0[0-1]_* regex BOT01.02_B0014]
```

superm1, 2019-01-12:
Can you please try to install the snap instead?

@hughsie, do we need to get a requirement on minimum version set higher on these?

ollie1, 2019-01-12:
> Can you please try to install the snap instead?
> 
> @hughsie, do we need to get a requirement on minimum version set higher on these?

@superm1 - Thank you so much - installing and using the snap solved my issue - both devices fixed now.

nightsky30, 2019-01-13:
Not that I expected any newer updates for this device in particular, but FWIW I did run across the same error due to my receiver's boot loader version.

$ sudo fwupdmgr get-updates
Not compatible with bootloader version: failed predicate [BOT03.0[0-1]_* regex BOT03.02_B0009]

$ fwupdmgr --version
client version:	1.1.2
daemon version:	1.1.2
compile-time dependency versions
	appstream-glib:	0.7.12
	gusb:	0.2.11
	efivar:	34

$ sudo fwupdmgr get-devices
Unifying Receiver
  DeviceId:             ad3f942734cc8856f47c5ffde16a6d1d01774a41
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Guid:                 21e75d9a-5ce6-5da2-b7ab-910c7f3f6836
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR24.07_B0030
  VersionBootloader:    BOT03.02_B0009
  Icon:                 preferences-desktop-keyboard
  Created:              2019-01-13

daenney, 2019-03-31:
I have a different version of the bootloader:
```
Unifying Receiver
  DeviceId:             2753c80712e2616ba87f53071f843c60374f2a01
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Guid:                 21e75d9a-5ce6-5da2-b7ab-910c7f3f6836
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR12.09_B0030
  VersionBootloader:    BOT01.04_B0016
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      7
  Created:              2019-03-31
```

kacpi2442, 2019-04-07:
I have the same problem on Manjaro linux, except my receiver is stuck in bootloader mode and i cannot use it anymore.
```shell
$ sudo fwupdmgr get-updates
Not compatible with bootloader version: failed predicate [BOT01.0[0-3]_* regex BOT01.04_B0016]
```

```shell
$ fwupdmgr --version
client version: 1.2.2
compile-time dependency versions
        gusb:   0.3.0
        efivar: 35
daemon version: 1.2.6
```

```shell
$ fwupdmgr get-devices
Unifying Receiver
  DeviceId:             708a5b99d0afc737cdfcab5e16716ecbc6dee884
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Guid:                 aa50f902-6989-5b40-ad1a-29955b5f1450
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Guid:                 941f4133-39e5-56a0-9df6-902547436c5c
  Guid:                 afa5518d-7c55-51cb-8e1c-bf3b11b67dfa
  Guid:                 cee9c5bc-aa12-51eb-a220-d0ef588d7cb2
  Guid:                 571229d5-caed-5cbc-91c1-5c216601a194
  Summary:              A miniaturised USB wireless receiver (bootloader)
  Plugin:               unifying
  Flags:                updatable|supported|registered|is-bootloader
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR00.00_B0000
  VersionBootloader:    BOT01.04_B0016
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      7
  Created:              2019-04-07
```

kacpi2442, 2019-04-07:
I tried to install manually correct firmware for this bootloader (which is `RQR12.09_B0030` not `RQR12.08_B0030`) but it gave me following error:
```shell
$ sudo fwupdmgr install Logitech-Unifying-RQR12.09_B0030.cab 
Dekompresowanie…         [***************************************]
Uwierzytelnianie…        [***************************************]
Instalowanie na urządzeniu Unifying Receiver…                    ]
Zapisywanie…             [***************************************]
failed to write @0000: only 1 byte write of 0xff supported
```

hughsie, 2019-04-08:
> failed predicate [BOT03.0[0-1]_* regex BOT03.02_B0009]

I can reproduce this here with devices that need a signed firmware image. Debugging now.

runi95, 2019-06-12:
Same problem here when trying to update the firmware on my Logitech keyboard. Running the upgrade command fails and then leaves the receiver in a locked state until I plug it out and back in again. I'm currently running Manjaro 18.0.4 on kernel version 5.1.8-1

`$ fwupdmgr --version
client version:	1.2.9
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.9`

`$ fwupdmgr get-updates
Unifying Receiver has firmware updates:
Device ID:               8d09844e5ab38eae1e060c97bf2d927b45e153a8
GUID:                    cc4cbfa9-bf9d-540b-b92b-172ce31013c1
GUID:                    279ed287-3607-549e-bacc-f873bb9838c4
ID:                      com.logitech.Unifying.RQR24.firmware
Update Version:          RQR24.06_B0030
Update Name:             Unifying Receiver Device Update Update
Update Summary:          Firmware for the Logitech Unifying receiver
Update Remote ID:        lvfs
Update Duration:         7 seconds
Update Checksum:         SHA1(fe037400f0977a13c215d94e427f6be14ae0be3e)
Update Location:         https://fwupd.org/downloads/0afa79c32ba48db68ccb80670a24980fbf8d8d76-Logitech-Unifying-RQR24.06_B0030.cab
Update Description:      This release addresses an encrypted keystroke injection issue known as Bastille security issue #13. The vulnerability is complex to replicate and would require a hacker to be physically close to a target.
                         
                         A few of Logitech's devices used to send select buttons in an unencrypted way, and in an effort to protect against this vulnerability, Logitech removed the feature. Affected hardware is:
                         
                          • Wireless Mouse M335
                          • Zone Touch Mouse T400
                          • Wireless Mouse M545
                          • Wireless Mouse M560
                          • Touch Mouse M600
                          • Touch Mouse T620
                          • Wireless Rechargeable Touchpad T650
                         
                         Although Logitech does not recommend it, these features may be re-activated by keeping/downgrading the receiver to an older firmware.
                         
ID:                      com.logitech.Unifying.RQR24.firmware
Update Version:          RQR24.05_B0029
Update Name:             Unifying Receiver Device Update Update
Update Summary:          Firmware for the Logitech Unifying receiver
Update Remote ID:        lvfs
Update Duration:         7 seconds
Update Checksum:         SHA1(0e7e9dafeb4dcc144d1434759ebf7bd71ea2a4d7)
Update Location:         https://fwupd.org/downloads/4511b9b0d123bdbe8a2007233318ab215a59dfe6-Logitech-Unifying-RQR24.05_B0029.cab
Update Description:      This release addresses an unencrypted keystroke injection issue known as Bastille security issue #11. The vulnerability is complex to replicate and would require a hacker to be physically close to a target.
                         
No releases found for device: Not compatible with bootloader version: failed predicate [BOT03.0[0-1]_* regex BOT03.02_B0009]`

`$ fwupdmgr get-devices
Unifying Receiver
  DeviceId:             8d09844e5ab38eae1e060c97bf2d927b45e153a8
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR24.01_B0023
  VersionBootloader:    BOT03.01_B0008
  VersionFormat:        plain
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      7
  Created:              2019-06-12`

trulex, 2019-07-24:
Same here:

```shell
$ fwupdmgr --version
client version: 1.2.10
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
daemon version: 1.2.10

```
```shell
$ fwupdmgr get-devices
Walter p50 Thunderbolt Controller
  DeviceId:             c6b3069761700570f962351af5048019dcefebf6
  Guid:                 9414852e-fa88-51e7-96f5-45268b29b86c
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              16.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2019-07-24

20EQS5P400 System Firmware
  DeviceId:             123fd4143619569d8ddb6ea47d1d3911eb5ef07a
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|require-ac|registered|needs-reboot
  Vendor:               LENOVO
  Version:              N1EET81W (1.54 )
  VersionFormat:        plain
  Icon:                 computer
  Created:              2019-07-24
  UpdateError:          Firmware can not be updated in legacy mode, switch to UEFI mode.

SAMSUNG MZNLN512HMJP-000L7
  DeviceId:             5dbeb140337f610d54913c0f43d15fdaa0eafb51
  Guid:                 608662a9-f828-5149-9305-693fe41ca7df
  Guid:                 15b2bd53-1c35-5ac5-9b65-3b220dc30fbb
  Guid:                 42d27a91-8272-507d-be96-7352d07dcab1
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              MAV22L6Q
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-07-24

Unifying Receiver
  DeviceId:             722c847df6997fb18164689eb4c4caf897ffdf1d
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR12.09_B0030
  VersionBootloader:    BOT01.04_B0016
  VersionFormat:        plain
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      7
  Created:              2019-07-24
```

Please answer the following questions:

- Operating system and version:
arch linux
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
pacman
- Have you tried rebooting?
yes
- Are you using an NVMe disk?
no

i-need-to-tell-you-something, 2019-09-24:
I also have this problem.
```
$ fwupdmgr --version
client version: 1.0.9
daemon version: 1.0.9
compile-time dependency versions
        appstream-glib: 0.7.7
        gusb:   0.2.11
        fwupdate:       10
        efivar: 34
```

```
$ fwupdmgr get-devices
Unifying Receiver
  DeviceId:             16324df8d782cbc6d3396541199c8fc397182242
  Guid:                 77d843f7-682c-57e8-8e29-584f5b4f52a1
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR24.07_B0030
  VersionBootloader:    BOT03.02_B0009
  Icon:                 preferences-desktop-keyboard
  Created:              2019-09-23

Latitude 5289 System Firmware
  DeviceId:             e2e09c295efdd32a1c9db191f0e73606ac9f019c
  Guid:                 bc26fcd1-1874-4845-9b96-a06d5397968e
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.17.1
  VersionLowest:        0.1.17.1
  Icon:                 computer
  Created:              2019-09-23
```

```
$ efibootmgr -v
BootCurrent: 0004
Timeout: 2 seconds
BootOrder: 0004,0007,0009,000A,000B,0011,0012
Boot0000* Windows Boot Manager  HD(2,GPT,e342cfad-581c-47e5-b546-7bbdf0461bbe,0xfa000,0x32000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}..._................
Boot0001* antergos_grub HD(1,GPT,7ce26162-992f-4332-a89f-97a8653e9513,0x800,0x100000)/File(\EFI\antergos_grub\grubx64.efi)
Boot0002* ubuntu        HD(1,GPT,4c4a3b9d-fb2a-42b8-be53-2efc1b9d3597,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
Boot0003* Linux Boot Manager    HD(1,GPT,bb980679-b78f-456b-9782-24bd3b4f9b52,0x800,0x100000)/File(\EFI\systemd\systemd-bootx64.efi)
Boot0004* ubuntu        HD(1,GPT,4c4a3b9d-fb2a-42b8-be53-2efc1b9d3597,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
Boot0007* Diskette Drive        BBS(Floppy,Diskette Drive,0x0)..BO
Boot0009* USB Storage Device    BBS(USB,USB Storage Device,0x0)..BO
Boot000A* CD/DVD/CD-RW Drive    BBS(CDROM,CD/DVD/CD-RW Drive,0x0)..BO
Boot000B* Onboard NIC   BBS(Network,Onboard NIC,0x0)..BO
Boot0011* Internal HDD  BBS(HD,Internal HDD,0x0)..BO
Boot0012* UEFI: TOSHIBA THNSNK256GVN8 M.2 2280 256GB, Partition 1       HD(1,GPT,4c4a3b9d-fb2a-42b8-be53-2efc1b9d3597,0x800,0x100000)/File(EFI\boot\bootx64.efi)..BO
```

```
$ efivar -l | grep fw
<no output; empty>
```

```
$ tree /boot
/boot
├── config-5.0.0-27-generic
├── config-5.0.0-29-generic
├── efi [error opening dir]
├── grub
[... 279 lines cut ...]
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initrd.img-5.0.0-27-generic
├── initrd.img-5.0.0-29-generic
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.0.0-27-generic
├── System.map-5.0.0-29-generic
├── vmlinuz-5.0.0-27-generic
└── vmlinuz-5.0.0-29-generic
```
Please answer the following questions:

    Operating system and version: Kubuntu 18.04
    How did you install fwupd? apt
    Have you tried rebooting? yes
    Are you using an NVMe disk? no
    Is secure boot enabled (only for the UEFI plugin)? I don't know/remember

Problem is same as @nightsky30 :
```
$ sudo fwupdmgr get-updates
Not compatible with bootloader version: failed predicate [BOT03.0[0-1]_* regex BOT03.02_B0009]
```

nightsky30, 2019-09-25:
@i-need-to-tell-you-something I went ahead and updated all my receivers using a Windows 10 VM in VirtualBox:

Unifying Receiver
  DeviceId:             ad3f942734cc8856f47c5ffde16a6d1d01774a41
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4 <- HIDRAW\VEN_046D&DEV_C52B
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR24.11_B0036
  VersionBootloader:    BOT03.02_B0009
  VersionFormat:        plain
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      7
  Created:              2019-09-25

Unifying Receiver
  DeviceId:             ad3f942734cc8856f47c5ffde16a6d1d01774a41
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4 <- HIDRAW\VEN_046D&DEV_C52B
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR12.10_B0032
  VersionBootloader:    BOT01.02_B0015
  VersionFormat:        plain
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      7
  Created:              2019-09-25

i-need-to-tell-you-something, 2019-09-25:
@nightsky30 You mean this as a workaround to updating the firmware via fwupd?

nightsky30, 2019-09-25:
> @nightsky30 You mean this as a workaround to updating the firmware via fwupd?

Yes

chriswyatt, 2019-09-27:
It's useful to know it's possible. Updating firmware via VirtualBox is a bit hit-and-miss. Sometimes you have to set up a USB filter and remove everything exception for the manufacturer ID, to get it to work.

FFY00, 2019-09-27:
This error (`Not compatible with bootloader version: failed predicate ...`) means there is no firmware available.

Please do `fwupdmgr refresh` and try again. You probably have an old version of the firmware repo.

i-need-to-tell-you-something, 2019-09-28:
@FFY00 

```
$ fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…           [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

$ sudo fwupdmgr get-updates
[sudo] password for me:
Not compatible with org.freedesktop.fwupd version 1.0.9, requires >= 1.3.1, Not compatible with bootloader version: failed predicate [BOT03.0[0-1]_* regex BOT03.02_B0009]
```
It sounds to me like it _expects_ bootloader version to be either `BOT03.00...` or `BOT03.01...` which doesn't match the `BOT03.02...` that I seem to have.

superm1, 2019-09-28:
The first part of that error is saying you need a newer fwupd too.

Can you install a newer version? You can use snap or flatpak potentially to do this.

superm1, 2019-10-24:
Although this is the "older" issue; I'm going to lump this in with https://github.com/fwupd/fwupd/issues/1261

We need to have better messaging why updates can't occur for some logitech devices.

closed 2019-10-24 by superm1