fwupd/fwupd #769: USB-C charge not working on T470s after UEFI 1.28 upgrade

Anthony25, 2018-09-27:
```shell
$ fwupdmgr --version
client version:	1.1.2
daemon version:	1.1.2
compile-time dependency versions
	appstream-glib:	0.7.12
	gusb:	0.3.0
	efivar:	35
```

```shell
$ fwupdmgr get-devices
20HF004QFR System Firmware
  DeviceId:             84149fb78009cf27cc1dd5520911f46f8792dbfe
  Guid:                 7a176688-0960-47ba-931b-7829849e8347
  Guid:                 399bff3a-e2ad-5947-859c-565bad878b00
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.28
  VersionLowest:        0.1.7
  Icon:                 computer
  Created:              2018-09-27

UEFI Device Firmware
  DeviceId:             df35e203442aa5d529202bea0d91b2f29dfc3d4c
  Guid:                 d95d3ada-eef1-464f-8a2a-a11232b8556b
  Guid:                 38c2e5ab-25e0-56d4-ac0b-23266e9959e1
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              182.29.3287
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-09-27

UEFI Device Firmware
  DeviceId:             03a6930fc4c833443a7502dde379b98b961c8c57
  Guid:                 e2acd8e4-a376-47bb-a316-4f1e62a8ca1f
  Guid:                 118d4b5a-8ef0-5dc6-b320-060dc79f2e36
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.20
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-09-27

SAMSUNG MZVLW512HMJP-000L7
  DeviceId:             e11623b2caa18fee292058a5c09ca4e6152f7ecf
  Guid:                 5b3df2da-f745-5fd0-81de-5dafd7f0bf8c
  Guid:                 8fd4ca73-d0ae-52e8-8977-461435c6f4cf
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Samsung Electronics Co Ltd
  VendorId:             NVME:0x144D
  Version:              6L7QCXY7
  Icon:                 drive-harddisk
  Created:              2018-09-27

T470s Thunderbolt Controller
  DeviceId:             5ad11a283a82eb6d3a6d5a34fcf454011f8ad39d
  Guid:                 96c3f95f-611a-56f8-be95-0f912cb84657
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              04.00
  Icon:                 computer
  Created:              2018-09-27
```

```shell
$ efibootmgr -v
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0017,0018,0019,001A,001B,001C,001D,001E
Boot0000* GRUB	HD(1,GPT,eb1929ae-b904-4cbd-94e6-0b2725eba9fd,0x800,0x82000)/File(\EFI\GRUB\grubx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001E  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot001F* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0020* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0021* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0022* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0023* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
```

```shell
$ efivar -l | grep fw
38243f72-e87f-468f-b19c-478598c46c3f-LenovoTpmFwUpdate
```

Please answer the following questions:

- Operating system and version: ArchLinux
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

After the UEFI upgrade (from 1.10 to 1.28), charging over USB-C does not work. I tested with the included dock and with an LG screen.

espoelstra, 2018-10-05:
Have you tried a Factory Reset from the BIOS and/or plugging in a regular AC adapter once to reset the charging circuit? I found that on a similar UEFI system I had to perform a Factory Reset after upgrading from a fairly old version to the latest, and your jump from 1.10 to 1.28 could be similar.

Anthony25, 2018-10-06:
Thanks @espoelstra, I will try that Tuesday ;)

Anthony25, 2018-10-12:
@espoelstra: you were right, resetting the UEFI fixed the issue. Sorry for that!

closed 2018-10-12 by Anthony25