fwupd/fwupd #770: Dell Latitude 7390 2-in-1 shows firmware updates for Latitude E7x60

ayhc, 2018-09-27:
System is Dell Latitude 7390 2-in-1.

Expected result: fwupdmgr should fetch firmware for Latitude 7390 2-in-1
Actual result: fwupdmgr shows firmware for the Latitude E5X60:

```shell
# fwupdmgr get-updates
Latitude 7390 2-in-1 System Firmware has firmware updates:
GUID:                    6180aaaa-5529-4bbf-b4fd-65b4f788de5b
GUID:                    230c8b18-8d9b-53ec-838b-6cfc0383493a
ID:                      com.dell.uefi6180aaaa.firmware
Update Version:          0.1.4614
Update Name:             Latitude E5X60 System Update
Update Summary:          Firmware for the Dell Latitude E5X60
Update Remote ID:        lvfs
Update Checksum:         SHA1(3463bac4200168c97729db990e85d8887bb14020)
Update Location:         https://fwupd.org/downloads/a8396afb82bdf84e28f2f08af733ad5dbcfaea29-firmware_Latitude_E5x70_Precision_3510_1.18.6.wu.cab
ID:                      com.dell.uefi6180aaaa.firmware
Update Version:          0.1.3588
Update Name:             Latitude E5X60 System Update
Update Summary:          Firmware for the Dell Latitude E5X60
Update Remote ID:        lvfs
Update Checksum:         SHA1(465cc5ad25697cb64a8fd4955161f406e4af6b33)
Update Location:         https://fwupd.org/downloads/358f816d191cda131ca59f74aad3b30d720b885f-firmware_Latitude_E5x70_Precision_3510_1.14.4.wu.cab
ID:                      com.dell.uefi6180aaaa.firmware
Update Version:          0.1.3078
Update Name:             Latitude E5X60 System Update
Update Summary:          Firmware for the Dell Latitude E5X60
Update Remote ID:        lvfs
Update Checksum:         SHA1(06434c070e0068309443aecb562c2d9e684af9d7)
Update Location:         https://fwupd.org/downloads/74fb7637e18cd0ff3155884bd29c36efc5e943a2-firmware_Latitude_E5x70_Precision_3510_1.12.6.wu.cab
Update Description:      None
ID:                      com.dell.uefi6180aaaa.firmware
Update Version:          0.1.2820
Update Name:             Latitude E5X60 System Update
Update Summary:          Firmware for the Dell Latitude E5X60
Update Remote ID:        lvfs
Update Checksum:         SHA1(72f4bc9b0701552e79e945171dbd46b24d7b3b3f)
Update Location:         https://fwupd.org/downloads/754f99d06990155dc9a4dd157823e2a30bb07950-firmware_Latitude_E5x70_Precision_3510_1.11.4.wu.cab
No upgrades for Latitude 7390 2-in-1 Thunderbolt Controller, current is 23.00: 23.00=same
No upgrades for Unifying Receiver, current is RQR12.07_B0029: RQR12.07_B0029=same
```

```shell
$ fwupdmgr --version
client version:	1.1.1
daemon version:	1.1.1
compile-time dependency versions
	appstream-glib:	0.7.10
	gusb:	0.3.0
	efivar:	35
```

```shell
$ fwupdmgr get-devices
Latitude 7390 2-in-1 System Firmware
  DeviceId:             4a91463cbd349e9902ad7c61fcb57e4b41119b9e
  Guid:                 6180aaaa-5529-4bbf-b4fd-65b4f788de5b
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.5.0
  VersionLowest:        0.1.5.0
  Icon:                 computer
  Created:              2018-09-27

Latitude 7390 2-in-1 Thunderbolt Controller
  DeviceId:             70700bfd593f62d200e97cc6e820a5eead825d0e
  Guid:                 0d799b24-3662-5caa-9ef1-c3d72827f964
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|supported|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              23.00
  Icon:                 computer
  Created:              2018-09-27

Unifying Receiver
  DeviceId:             32592a101a595e15f22f13ef5faabe69277c5010
  Guid:                 77d843f7-682c-57e8-8e29-584f5b4f52a1
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR12.07_B0029
  VersionBootloader:    BOT01.02_B0015
  Icon:                 preferences-desktop-keyboard
  Created:              2018-09-27
  UpdateState:          success
```

Please answer the following questions:

- Operating system and version: Manjaro (rolling release; current)
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pamac
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

superm1, 2018-09-27:
Duplicate of https://github.com/hughsie/fwupd/issues/695

I think the proper solution will require a change on the CAB on LVFS side.

closed 2018-09-27 by superm1

robinbye, 2020-05-12:
So for us mere mortals, what exactly is the solution?

superm1, 2020-05-12:
Sorry to hear it's still not fixed...

For now, what I would recommend you do is follow the instructions at https://github.com/fwupd/firmware-dell to install the updated BIOS from Dell Support for your system.  The fix for this to not use the copy and paste GUID will need to happen there.

If the problem persists with the current firmware, I would recommend opening a support case so that they can prioritize rolling out a proper solution with the appropriate teams.

robinbye, 2020-05-12:
Thanks for the extremely quick reply! Will see if I can work it out :)