fwupd/fwupd #774: Thunderbolt Controller not listed anymore on Dell XPS 9370

BjoernDaase, 2018-09-30:
Hi,
I upgraded my Ubuntu 18.04 to the 18.10 beta today which includes an fwupdmgr version upgrade from 1.0.6.
On Ubuntu 18.04 everything worked fine and the thunderbolt controller was listed when running `fwupdmgr get-devices` and as you can see now it's not. But thunderbolt itself works (tested by using the Dell WD15 dock). 
I know that I'm running a beta tree but does anybody have an idea what happend?

```shell
$ fwupdmgr --version
client version:	1.1.2
daemon version:	1.1.2
compile-time dependency versions
	appstream-glib:	0.7.12
	gusb:	0.2.11
	efivar:	34
```

```shell
$ fwupdmgr get-devices
XPS 13 9370 System Firmware
  DeviceId:             8a21cacfb0a8d2b30c5ee9290eb71db021619f8b
  Guid:                 7ceaf7a8-0611-4480-9e30-64d8de420c7c
  Guid:                 43ea5588-d9a4-5031-8ad3-308045302d6b
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.5.1
  VersionLowest:        0.1.5.1
  Icon:                 computer
  Created:              2018-09-30

PM981 NVMe Samsung 512GB
  DeviceId:             f954c7acdf5fab61aeaca1cd71d29ea5ade6992f
  Guid:                 47335265-a509-51f7-841e-1c94911af66b
  Guid:                 8fd4ca73-d0ae-52e8-8977-461435c6f4cf
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Samsung Electronics Co Ltd
  VendorId:             NVME:0x144D
  Version:              EXA73D1Q
  Icon:                 drive-harddisk
  Created:              2018-09-30

```

```shell
$ efibootmgr -v
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000
Boot0000* ubuntu	HD(1,GPT,1d435577-b51a-4491-b78e-b33b18783cfc,0x22,0x77338)/File(\EFI\ubuntu\shimx64.efi)
Boot0002* M.2 PCIe SSD	BBS(HD,P0: PM981 NVMe Samsung 512GB,0x0)..BO
Boot0003* USB Storage Device	BBS(USB,USB Storage Device,0x0)..BO
Boot0004* CD/DVD/CD-RW Drive	BBS(CDROM,CD/DVD/CD-RW Drive,0x0)..BO
Boot0005* USB NIC	BBS(128,Realtek PXE B00 D14,0x0)..BO
Boot0006* UEFI: PM981 NVMe Samsung 512GB, Partition 1	HD(1,GPT,b305adf1-b1dd-415b-89c8-21c9770c5262,0x800,0x177000)/File(EFI\boot\bootx64.efi)..BO

```

```shell
$ efivar -l | grep fw
**It's empty**
```

```shell
$ tree /boot
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version: Ubuntu 18.10
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt-get
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? no

superm1, 2018-09-30:
If I was going to make an educated guess I'd say nothing else changing bolt might have force powered on and off at the wrong time for fwupd (they stomped on each other's toes).
Newer fwupd can communicate with newer bolt to prevent this but it's possible the version in 18.10 doesn't coordinate yet.

Can you please share the following:

1) what version of bolt do you have on the system
2) can you please modify the systemd unit for fwupd to add "--verbose" to ExecStart and power off your system and back on. Then gather the journalctl output for that  unit. I specifically want to see cold boot behavior.
3) kernel dmesg log

Fwiw a few comments too.
1) you can turn off legacy option rom in bios setup. It should speed up your boot process slightly.
2) Wd15 doesn't have thunderbolt. TB16 does.

superm1, 2018-10-01:
1. that version of bolt doesn't have force power.
2. I'll need to see fwupd's verbose log to tell why you're not seeing the device then.  It looks like it was on for the normal amount of time and turned off.

The unit file is here:
`/lib/systemd/system/fwupd.service`

BjoernDaase, 2018-10-01:
Alright, thanks.

That is the output of `journalctl -u fwupd.service -b`
```
-- Logs begin at Thu 2018-08-23 23:57:37 CEST, end at Mon 2018-10-01 19:17:01 CEST. --
Okt 01 19:15:40 XPS-9370 systemd[1]: Starting Firmware update daemon...
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0556 Fu  Verbose debugging enabled (on console 0)
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0557 As  run FuEngine:load
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0557 Fu  loading config values from /etc/fwupd/daemon.conf
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0560 Fwupd looking for os-release at /etc/os-release
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0560 As  run AsStore:load-installed{/usr/share/fwupd/metainfo}
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0561 As  adding existing file: /usr/share/fwupd/metainfo/org.freedesktop.fwupd.remotes.lvfs-testing.metainfo.xml
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0561 As  adding existing file: /usr/share/fwupd/metainfo/org.freedesktop.fwupd.remotes.lvfs.metainfo.xml
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0562 As  Emitting ::changed() [load-installed]
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0562 Fu  using config path of /etc/fwupd
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0562 Fu  loading config from /etc/fwupd/remotes.d/vendor.conf
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0563 Fu  loading config from /etc/fwupd/remotes.d/fwupd.conf
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0563 Fu  loading config from /etc/fwupd/remotes.d/lvfs-testing.conf
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0564 Fu  loading config from /etc/fwupd/remotes.d/lvfs.conf
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0564 Fu  using config path of /usr/lib/fwupd
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0564 Fu  ordering lvfs-testing=lvfs+1
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0564 Fu  ordering lvfs=fwupd+1
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0564 Fu  ordering lvfs-testing=lvfs+1
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0564 As  run FuEngine:load-smbios
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0564 As  run FuEngine:load-hwids
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0564 Fu  smbios property Manufacturer=Dell Inc.
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0564 Fu  smbios property EnclosureKind=a
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0564 Fu  smbios property Family=XPS
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0564 Fu  smbios property ProductName=XPS 13 9370
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0564 Fu  smbios property ProductSku=07E6
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0564 Fu  smbios property BiosVendor=Dell Inc.
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0564 Fu  smbios property BiosVersion=1.5.1
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0564 Fu  smbios property BiosMajorRelease=01
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0564 Fu  smbios property BiosMinorRelease=05
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0564 Fu  smbios property BaseboardManufacturer=Dell Inc.
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0564 Fu  smbios property BaseboardProduct=0F6P3V
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0564 Fu  {70c87b49-62f9-50a7-a051-a5bbc63bb89c}   <- Dell Inc.&XPS&XPS 13 9370&7E6&Dell Inc.&1.5.1&01&05
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0564 Fu  {7ee4fa90-0066-5cd9-9e02-72b0b32fcf72}   <- Dell Inc.&XPS&XPS 13 9370&Dell Inc.&1.5.1&01&05
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0564 Fu  {e951a16d-d5f1-5083-8558-60f680deeb86}   <- Dell Inc.&XPS 13 9370&Dell Inc.&1.5.1&01&05
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0565 Fu  {a00e4d4d-247b-55f9-8869-e83553e74570}   <- Dell Inc.&XPS&XPS 13 9370&7E6&Dell Inc.&F6P3V
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0565 Fu  {d72b8452-80b6-5b66-b95c-8dd75519a30c}   <- Dell Inc.&XPS&XPS 13 9370&7E6
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0565 Fu  {184da599-9986-5212-9dba-226d2d233133}   <- Dell Inc.&XPS&XPS 13 9370
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0565 Fu  {f51ef121-403c-5ebf-8ccc-28b7a3e36b44}   <- Dell Inc.&7E6&Dell Inc.&F6P3V
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0565 Fu  {74bb5080-7e51-5373-aea1-bfc218a2caaf}   <- Dell Inc.&7E6
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0565 Fu  {04840070-fce8-57d3-bec4-2ecebebdecaf}   <- Dell Inc.&XPS 13 9370&Dell Inc.&F6P3V
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0565 Fu  {f33883b9-e996-539e-b35a-73568b903d13}   <- Dell Inc.&XPS 13 9370
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0565 Fu  {b350638b-bc1b-57d9-9946-5a0f555d95e7}   <- Dell Inc.&XPS&Dell Inc.&F6P3V
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0565 Fu  {61178075-a8fd-563c-9045-44227d8c121f}   <- Dell Inc.&XPS
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0565 Fu  {5c163113-9296-5c8d-a92d-ae04eb59a0f7}   <- Dell Inc.&a
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0565 Fu  {7ec53000-14a8-55ad-9834-388b0101c8e3}   <- Dell Inc.&Dell Inc.&F6P3V
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0565 Fu  {85d38fda-fc0e-5c6f-808f-076984ae7978}   <- Dell Inc.
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0565 As  run FuEngine:load-quirks
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0565 Fu  loading quirks from /usr/share/fwupd/quirks.d/altos.quirk
Okt 01 19:15:40 XPS-9370 fwupd[8078]: 17:15:40:0565 Fu  now 2 quirk entries
[... cut ...]
Okt 01 19:16:22 XPS-9370 fwupd[8078]: 17:16:22:0970 ThunderboltPower uevent for 0-0: (/sys/devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:00.0/0000:05:00.0/domain0/0-0) remove
Okt 01 19:16:23 XPS-9370 fwupd[8078]: 17:16:23:0469 Thunderbolt uevent for /sys/devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:00.0/0000:05:00.0/domain0: remove
Okt 01 19:16:23 XPS-9370 fwupd[8078]: 17:16:23:0470 ThunderboltPower uevent for domain0: (/sys/devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:00.0/0000:05:00.0/domain0) remove
```

superm1, 2018-10-02:
Here's the failure.

```
Okt 01 19:16:02 XPS-9370 fwupd[8078]: 17:16:02:0730 Thunderbolt failed to get native mode status: Error reading from file: Input/output error
```
So it was failing to read the native mode status.  Given we've seen errors with reading `nvm_version` requiring multiple reads before the hardware is ready I wonder if we're seeing the same thing now with the nvm image too?

When reading the NVM image we just call `g_file_get_contents`.  The error appears to likely be `G_FILE_ERROR_IO` but that might just be code for glib doesn't know how to return `-EAGAIN`

(see comment in code)
```
		/* glib can't return a properly mapped -ENODATA but the
		 * kernel only returns -ENODATA or -EAGAIN */
```

@westeri @YehezkelShB thoughts on this?  Should we add in something to try again on this error code?

westeri, 2018-10-03:
``nvm_version`` is special because the attribute is there already before NVMem devices get added. Reading NVMem may fail of course in the same way other I/O operations can so the caller should handle those cases properly. In this particular case, I'm not sure what the issue actually is, though. It takes some time for the NVMem devices to appear so could it be that fwupd tries to read them too early and they do not exist? I guess ``errno`` would be ``ENOENT`` in that case, not ``EIO``.

BjoernDaase, 2018-10-03:
I restarted several times (5-6) and the tunderbolt controller did never show up so the problem occurs every time.
I can make a reboot now and give you both related parts if that helps.

BjoernDaase, 2018-10-03:
Alright, here are the related logs:
```

[    0.000000] Linux version 4.18.0-8-generic (buildd@lgw01-amd64-055) (gcc version 8.2.0 (Ubuntu 8.2.0-6ubuntu1)) #9-Ubuntu SMP Mon Sep 10 16:12:10 UTC 2018 (Ubuntu 4.18.0-8.9-generic 4.18.7)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.18.0-8-generic root=UUID=7ea246ac-cdde-4c3f-acc0-40ab63d8dd47 ro quiet splash mem_sleep_default=deep resume=UUID=71013368-e9f9-4063-b176-fd9d4a73e524 vt.handoff=1
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000002dc5ffff] usable
[    0.000000] BIOS-e820: [mem 0x000000002dc60000-0x000000002dc60fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000002dc61000-0x000000002dc61fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000002dc62000-0x000000003ecdafff] usable
[    0.000000] BIOS-e820: [mem 0x000000003ecdb000-0x000000003f09afff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003f09b000-0x000000003f0e7fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000003f0e8000-0x000000003f79ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000003f7a0000-0x000000003ff26fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003ff27000-0x000000003fffefff] type 20
[    0.000000] BIOS-e820: [mem 0x000000003ffff000-0x000000003fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000040000000-0x0000000047ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000048000000-0x0000000048dfffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000048e00000-0x000000004f7fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000004ae7fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.40 by American Megatrends
[    0.000000] efi:  ACPI=0x3f0ab000  ACPI 2.0=0x3f0ab000  SMBIOS=0xf0000  SMBIOS 3.0=0xf0020  ESRT=0x3fdd9018  MEMATTR=0x3c2af018 
[    0.000000] secureboot: Secure boot could not be determined (mode 0)
[    0.000000] SMBIOS 3.0.0 present.
[    0.000000] DMI: Dell Inc. XPS 13 9370/0F6P3V, BIOS 1.5.1 08/09/2018
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] last_pfn = 0x4ae800 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: write-back
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 0080000000 mask 7F80000000 uncachable
[    0.000000]   1 base 0060000000 mask 7FE0000000 uncachable
[    0.000000]   2 base 0050000000 mask 7FF0000000 uncachable
[    0.000000]   3 base 004C000000 mask 7FFC000000 uncachable
[    0.000000]   4 base 004B000000 mask 7FFF000000 uncachable
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000]   8 disabled
[    0.000000]   9 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000000] last_pfn = 0x48e00 max_arch_pfn = 0x400000000
[    0.000000] esrt: Reserving ESRT space from 0x000000003fdd9018 to 0x000000003fdd9050.
[... 1848 lines cut ...]
Okt 03 20:50:17 XPS-9370 fwupd[9860]: 18:50:17:0965 Fu  performing udev_device_added() on udev
Okt 03 20:50:17 XPS-9370 fwupd[9860]: 18:50:17:0965 Fu  performing udev_device_added() on unifying
Okt 03 20:50:17 XPS-9370 fwupd[9860]: 18:50:17:0965 Fu  performing udev_device_added() on nvme
Okt 03 20:50:17 XPS-9370 fwupd[9860]: 18:50:17:0965 Fu  Called GetResults(*)
Okt 03 20:50:17 XPS-9370 fwupd[9860]: 18:50:17:0965 Fu  FuHistory: get devices
Okt 03 20:50:17 XPS-9370 fwupd[9860]: 18:50:17:0965 Fu  FuHistory: got sql result f7d19dece3e6e0cfb963dc7fb7598c82e246f03e
Okt 03 20:50:17 XPS-9370 fwupd[9860]: 18:50:17:0966 Fu  Called GetRemotes()
Okt 03 20:50:18 XPS-9370 fwupd[9860]: 18:50:18:0718 Fu  Called GetDevices()
[... cut ...]
Okt 03 20:50:38 XPS-9370 fwupd[9860]: 18:50:38:0215 ThunderboltPower uevent for 0-0: (/sys/devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:00.0/0000:05:00.0/domain0/0-0) remove
Okt 03 20:50:38 XPS-9370 fwupd[9860]: 18:50:38:0707 Thunderbolt uevent for /sys/devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:00.0/0000:05:00.0/domain0: remove
Okt 03 20:50:38 XPS-9370 fwupd[9860]: 18:50:38:0707 ThunderboltPower uevent for domain0: (/sys/devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:00.0/0000:05:00.0/domain0) remove

```

I I can provide further information please let me know.

superm1, 2018-10-03:
Sorry I should have specified, I wanted to see the exact timescale so grab from `/var/log/syslog` instead (I don't think I can tell from just kernel `[XXXXX.YYYYYY]`

BjoernDaase, 2018-10-03:
Alright, I have the file but its 43MB so it's too large for github to attach and if I try to copy and paste the content the github page dies. How can I give it to you?

superm1, 2018-10-03:
Yeah just needed last entry, thanks I'll take a look.

superm1, 2018-10-03:
CC @gicmo FYI.

@BjoernDaase ah ha.  It _is_ actually bolt causing it but fwupd is implicated because it's actually causing bolt to start. :/

```
Oct  3 20:49:55 XPS-9370 fwupd[9860]: 18:49:55:0735 Fu  performing coldplug() on thunderbolt_power
Oct  3 20:49:55 XPS-9370 dbus-daemon[878]: [system] Activating via systemd: service name='org.freedesktop.bolt' unit='bolt.service' requested by ':1.494' (uid=0 pid=9860 comm="/usr/lib/fwupd/fwupd --verbose " label="unconfined")
Oct  3 20:49:55 XPS-9370 systemd[1]: Starting Thunderbolt system service...
```

Bolt 0.4 is using force power, but doesn't offer support for fwupd and bolt to coordinate the force power request.  The support for coordinating this request is however in fwupd 1.1.2.  I've mentioned this on the bug here https://bugs.launchpad.net/ubuntu/+source/bolt/+bug/1795864 and if you can offer to help test 0.5 I believe this should be fixed.

```
Oct  3 20:49:56 XPS-9370 boltd[9888]: power: setting force_power to OFF
Oct  3 20:49:56 XPS-9370 fwupd[9860]: 18:49:56:0821 ThunderboltPower Bolt force power support: 0
Oct  3 20:49:56 XPS-9370 fwupd[9860]: 18:49:56:0822 ThunderboltPower Bolt force power support: 0
Oct  3 20:49:56 XPS-9370 fwupd[9860]: 18:49:56:0822 ThunderboltPower Setting force power to 1 using kernel
Oct  3 20:49:56 XPS-9370 fwupd[9860]: 18:49:56:0823 ThunderboltPower Setting timeout to 20 seconds
```

superm1, 2018-10-03:
Basically even though bolt doesn't support force power API over dbus it gets started up by fwupd via dbus activation (which fwupd then discovers that bolt doesn't support it) :/

closed 2018-10-03 by superm1

BjoernDaase, 2018-10-03:
@superm1 Thank's for debugging it and leaving the comment over there.
If they provide the upgrade I will help testing.

BjoernDaase, 2018-10-03:
Using Bolt 0.5, which can be found here https://launchpad.net/~ubuntu-desktop/+archive/ubuntu/ppa/+build/15508465, solves the problem and the thunderbolt controller shows up again!
Thanks for debugging it.