fwupd/fwupd #775: Unable to switch TPM mode on XPS 9550

gentoochainsaw, 2018-10-01:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

$ fwupdmgr --version
client version:	1.1.2
daemon version:	1.1.2
compile-time dependency versions
	appstream-glib:	0.7.9
	gusb:	0.2.11
	efivar:	36

XPS 15 9550 TPM 1.2
  DeviceId:             5ab1e5b4c5e2639722437a5b7c5a8391c75fcabe
  Guid:                 9055a777-a92f-55da-8d34-6614d8f9df92
  Guid:                 a4f7987a-5837-5ac6-b579-60ce4e358f84
  Summary:              Platform TPM device
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Vendor:               Dell Inc.
  Version:              5.81.2.1
  Icon:                 computer
  Created:              2018-10-01

XPS15 9550 Thunderbolt Controller
  DeviceId:             84e567453922d3ba97ca4c7261bbf62508ec1727
  Guid:                 c2ca983d-a03d-56c6-80c4-fbb38d829f50
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|supported|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              26.01
  Icon:                 computer
  Created:              2018-10-01

Dell Thunderbolt Cable
  DeviceId:             2e3d37ac318c476635bff9dc8ba53e151d42eabf
  ParentDeviceId:       84edf044a6b53037732cfc7591b65a8d36384cc9
  Guid:                 b4fd3cdf-4e3a-5090-a583-45367cfd6421
  Plugin:               thunderbolt
  Flags:                updatable|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              16.00
  Icon:                 audio-card
  Created:              2018-10-01

Dell Thunderbolt Dock
  DeviceId:             72fd241e50970edb9a150188bb9af8cc8cc535e8
  ParentDeviceId:       84edf044a6b53037732cfc7591b65a8d36384cc9
  Guid:                 76cc74d4-f062-5b93-a11c-8d2a58a25848
  Plugin:               thunderbolt
  Flags:                updatable|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              16.00
  Icon:                 audio-card
  Created:              2018-10-01

Synaptics VMM3320 inside Dell WD15/TB16/TB18 wired Dock
  DeviceId:             7b3cbc344094ace839263dd4711c84ea99bb6dc3
  ParentDeviceId:       84edf044a6b53037732cfc7591b65a8d36384cc9
  Guid:                 152a9833-41e6-5c21-9dde-0ff8984f3e20
  Guid:                 0a52c8c7-26d5-59a0-ae44-6b00e276d775
  Guid:                 51000744-2de6-5a52-8a86-d5f69fd73d34
  Summary:              Multi-Stream Transport Device
  Plugin:               synapticsmst
  Flags:                updatable|registered
  Vendor:               Synaptics
  Version:              3.10.002
[... 144 lines cut ...]
XPS 15 9550 TPM 2.0
  DeviceId:             04d075f2971403e31c1dbdf3b38f709d7c2e749d
  ParentDeviceId:       5ab1e5b4c5e2639722437a5b7c5a8391c75fcabe
  Guid:                 75305a8e-596f-5a3b-9aea-ebee36dc585e
  Guid:                 b2ce3dae-d233-53ce-89e3-4abe40a5b9bf
  Summary:              Alternate mode for platform TPM device
  Plugin:               uefi
  Flags:                internal|require-ac|registered
  Vendor:               Dell Inc.
  Icon:                 computer
  Created:              2018-10-01
  Modified:             2018-10-01

agamemnon ~ # fwupdmgr install https://fwupd.org/downloads/a1a6e10beb96281fa78c62a7d967c8c3a1cf7430-DellTpm2.0_Fw1.3.1.0.cab
Fetching firmware https://fwupd.org/downloads/a1a6e10beb96281fa78c62a7d967c8c3a1cf7430-DellTpm2.0_Fw1.3.1.0.cab
Downloading…           [***************************************] Less than one minute remaining…
Decompressing…         [***************************************]
Device XPS 15 9550 TPM 2.0 [04d075f2971403e31c1dbdf3b38f709d7c2e749d] does not currently allow updates

It is in fact unlocked, as evidenced by trying again:
agamemnon ~ # fwupdmgr unlock 04d075f2971403e31c1dbdf3b38f709d7c2e749d
Authenticating…        [***************************************]
Device 04d075f2971403e31c1dbdf3b38f709d7c2e749d is not locked

Note that restarting the "fwupd" daemon reinstates the "locked" flag, as does rebooting the system.

superm1, 2018-10-01:
(Updated your comment to add back-ticks).

> Note that restarting the "fwupd" daemon reinstates the "locked" flag, as does rebooting the system.

Yes, that's expected behavior.

>   Flags:                internal|require-ac|registered

It looks like after it's unlocked that the updateable flag didn't get put in place on the alternate device.

My guess; you probably need to clear the TPM in BIOS setup before starting.  fwupd won't allow TPM mode switch when the TPM is owned or contains data.

superm1, 2018-10-01:
Hmm, however you have the updateable flag on the 1.2 device.  This might be a bug that was related to the uefi proxy device switchover that happened in 1.1.0.

The flags are "supposed" to be cloned between devices, but since the device is owned by 'uefi' not 'dell' the wrong method probably isn't called anymore.
https://github.com/hughsie/fwupd/blob/master/plugins/dell/fu-plugin-dell.c#L682

superm1, 2018-10-02:
@gentoochainsaw can you please test my branch `wip/superm1/dell-unlock` to see if it helps?  The commit is referenced above.

My current platforms at hand can't switch modes, so I can't check easily myself.

superm1, 2018-10-03:
@gentoochainsaw Do you know if regular BIOS firmware updates are working for you in the current configuration?

That error is peculiar and it makes me wonder if it's related to the way your ESP is arranged.  It doesn't seem like a bug in the TPM update flow now but actually UEFI update flow.

```
/boot
├── backup.efi
├── EFI
│   ├── Dell
│   │   └── Bios
│   │       └── Recovery
│   │           ├── BIOS_CUR.RCV
│   │           └── BIOS_PRE.rcv
│   ├── EFI
│   └── gentoo
│       ├── fw
│       ├── fwupdx64.efi
│       └── fwupx64.efi
└── vmlinuz.efi
```

So a few questions:
1) Where is your ESP actually mounted?  It seems like it's at `/boot`.  It's really odd to me that `vmlinuz.efi` is showing up at the root of `/boot` though`

2) Did you modify `/etc/fwupd/uefi.conf` at all?

3) With that branch in place can you please switch the daemon to verbose and share journal output for `fwupd.service`? (Modify the systemd unit to have `--verbose` in `ExecStart`, reload units, restart `fwupd.service`)

gentoochainsaw, 2018-10-03:
> @gentoochainsaw Do you know if regular BIOS firmware updates are working for you 

Yes, they certainly did. I remember sending the report out for 0.1.8.0 system firmware.

> That error is peculiar and it makes me wonder if it's related to the way your ESP is arranged.

This is my first UEFI installation and it was a bit haphazard, open to improvement suggestions.

> 1. Where is your ESP actually mounted?

```
agamemnon ~ # mount | grep vfat
/dev/nvme0n1p1 on /boot type vfat (rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-15,shortname=mixed,utf8,errors=remount-ro)
```
> It's really odd to me that `vmlinuz.efi` is showing up at the root of `/boot` though

I am using the kernel stub loader and upgrade kernels like this:
```
cd /usr/src/linux && make mrproper && zcat /proc/config.gz > .config && make oldconfig && make menuconfig && make -j8 && cp arch/x86/boot/bzImage /boot/vmlinuz.efi && mount -o remount,rw /sys/firmware/efi/efivars/ && efibootmgr -B -b 0000 && efibootmgr --create --part -1 --disk /dev/nvme0n1 --label Gentoo --loader \vmlinuz.efi && mount -o remount,ro /sys/firmware/efi/efivars
```

> 2. Did you modify `/etc/fwupd/uefi.conf` at all?

Yes.
```
agamemnon ~ # cat /etc/fwupd/uefi.conf 
[uefi]
RequireShimForSecureBoot=true
OverrideESPMountPoint=/boot
```

> 3. With that branch in place

I will be honest with you, I did not use your branch. I applied the patch you attached to this report to fwupd 1.1.2 as a "user patch". It applied cleanly.

superm1, 2018-10-03:
> Yes, they certainly did. I remember sending the report out for 0.1.8.0 system firmware.

OK thanks for confirming.

> This is my first UEFI installation and it was a bit haphazard, open to improvement suggestions.

The part that throws me off is the nested EFI/EFI directories.  If it's working for non TPM updates however though I think it's fine.

> I will be honest with you, I did not use your branch. I applied the patch you attached to this report to fwupd 1.1.2 as a "user patch". It applied cleanly.

Thanks for clarifying.  Let me double check all the other code then to make sure nothing else needs to actually come back with this patch for a backport.  I'd still like to see a verbose daemon log though too.

gentoochainsaw, 2018-10-03:
[fwupd.log](https://github.com/hughsie/fwupd/files/2442356/fwupd.log)

superm1, 2018-10-03:
OK thanks.  I found the problem, it was due to the order of events with Dell plugin running before UEFI plugin.  I've added another patch to that branch that should fix it.

gentoochainsaw, 2018-10-04:
```
agamemnon / # fwupdmgr update
No upgrades for XPS15 9550 Thunderbolt Controller, current is 26.01: 16.00=older
ignoring XPS 15 9550 System Firmware [2da31270d317b076424992de14a0f08ae373c137] as not updatable
ignoring XPS 15 9550 TPM 1.2 [5ab1e5b4c5e2639722437a5b7c5a8391c75fcabe] as not updatable
Downloading 1.3.1.0 for XPS 15 9550 TPM 2.0...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Updating XPS 15 9550 TPM 2.0 from 0.0.0.0 to 1.3.1.0…          ]
Scheduling…            [***************************************]

An update requires a reboot to complete. Restart now? [Y|n]: Y
```
More soon.

gentoochainsaw, 2018-10-04:
[image]
So that didn't work fantastically well; I've taken a photo of the boot entry it created. Used the BIOS menu system to try & launch that file manually and I am not convinced it worked.

```
agamemnon ~ # fwupdmgr get-devices --show-all-devices | grep -C6 "Alternate mode for platform TPM device"

XPS 15 9550 TPM 2.0
  DeviceId:             04d075f2971403e31c1dbdf3b38f709d7c2e749d
  ParentDeviceId:       5ab1e5b4c5e2639722437a5b7c5a8391c75fcabe
  Guid:                 75305a8e-596f-5a3b-9aea-ebee36dc585e
  Guid:                 b2ce3dae-d233-53ce-89e3-4abe40a5b9bf
  Summary:              Alternate mode for platform TPM device
  Plugin:               uefi
  Flags:                internal|require-ac|locked|registered
  Vendor:               Dell Inc.
  Icon:                 computer
  Created:              2018-10-04
  UpdateState:          success
```

Does the "success" refer to the staging process only, or does that suggest the update was in fact applied? I ask because the system came back on TPM 1.2 and I saw none of the usual firmware upgrade progress bars on the Dell logo.

gentoochainsaw, 2018-10-04:
Rerunning it, and just checking the boot entries prior to rebooting:
```
agamemnon / # efibootmgr -v
BootNext: 0001
BootCurrent: 0002
Timeout: 0 seconds
BootOrder: 0000,0001
Boot0000* Gentoo	HD(1,GPT,13a146e5-5ee8-48fc-9928-65e8d873a668,0x800,0x100000)/File(vmlinuz.efi)
Boot0001* Linux Firmware Updater	PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-08-0D-02-00-13-1D-97)/File(\EFI\gentoo\fwupdx64.efi)
```

superm1, 2018-10-04:
Success does only refer to the staging in this context.

Can you describe more about the behavior when you tried to run the boot entry? No Dell logo whatsoever? Or a message came up?
See the typical white text before Dell logo?

It may be useful to see fwupdate debugging, where you run `/usr/libexec/fwupd/fwupdate --debug` first before staging and then use the same tool to view logs when done.

gentoochainsaw, 2018-10-04:
The system reports no bootable device, in white text, offering F-key prompts for setup etc.
I am able, using the BIOS menus, to create a working boot entry for "vmlinuz.efi" and boot my linux kernel. I am not able to boot/use "fwupdx64.efi" in this way.

superm1, 2018-10-04:
Also can you please try efivar master? That boot entry is suspect to me to a similar problem that was seen on Lenovo (https://github.com/rhboot/efivar/issues/121) specifically with efivar 36 or 35 and NVMe disks.

Upstream hasn't cut a new release yet though.

gentoochainsaw, 2018-10-09:
Waiting for efivar release to be cut.

superm1, 2018-10-09:
Other than testing with master, you should also be able to downgrade to 34 to verify

gentoochainsaw, 2018-10-09:
Thank you, it worked flawlessly with your 2 patches on with efivar 34. Now in TPM 2.0 mode.

superm1, 2018-10-10:
Fixed in master with https://github.com/hughsie/fwupd/commit/dc48bec1576449288b05c882cd4af8ae770486f7

closed 2018-10-10 by superm1

ignisf, 2021-03-06:
I am sorry for necrobumping, but I am not able to find the firmware mentioned here (https://fwupd.org/downloads/a1a6e10beb96281fa78c62a7d967c8c3a1cf7430-DellTpm2.0_Fw1.3.1.0.cab) in LVFS' search interface although I was able to use it to change my TPM mode.

Is there some secret stash of firmwares that possibly contains the latest version (which seems to be [1.3.2.8](https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=rf87d&oscode=wb64a&productcode=optiplex-3040m-desktop))?

commit 8151b84b (2018-10-02): uefi: Port the TPM unlocking functionality from Dell plugin (Fixes: #775)