fwupd/fwupd #778: Need to support a version_display

hughsie, 2018-10-02:
Another OEM wants to have a display version (not used for comparison) that's output all the way to the GUI clients. The existing `quad` and `triplet` is insufficient as Intel has defined a somewhat tricky-to-parse ME firmware version of `aaa.bbbbb.cccccccc.dddddddddddddddd` where each letter is a bit, not a byte. Furthermore, `aaa` needs to be offset by decimal `11` -- and that's kinda hard to encode into "LVFS::VersionFormat".

Two ideas here:

* Define a `intel-me` alternative to quad and triplet.

* Create a new version_for_display that gets added to the LVFS, fwupd, libfwupd, gnome-software and fwupdmgr. This would need a boatload of tests obviously as some assumptions subtly change.

* Define a way to encode the weird version formats into the existing `LVFS::VersionFormat` tag, also adding it as a quirk for the daemon devices.

The second option does seem more flexible, but it makes some things a lot harder. e.g. if we do something like

    <requires><firmware version="1.2.3.4" compare="ge">GUID</requires>

then should we compare this as a `uint16_t` by converting it to a quad and then using the raw version number, or do a simple semver comparison? If we're allowing versions like 0x12345678 in the metainfo file do we then set the FuDevice version to `305419896`? I'd be really unhappy showing large decimal numbers in the `fwupdmgr` output, and also changing the expected string format from an API/ABI point of view.

I guess the key issue is that we don't know the encoding of the version number to display the **device** version, without additional metadata from the firmware, which might not exist. We can't even use a quirk, as that would need to be changed with every firmware update that is issued.

The first option does seem a lot simpler, at the expense of hardcoding the weird Intel ME specification into fwupd. This would need a quirk for each device to show the right thing (e.g. HwId would set VersionFormat) -- so we'd combine both `NvmeVersionFormat` and `DaemonVersionFormat` to just `VersionFormat` and set the version format on the FuDevice itself.

Other ideas certainly welcome.

superm1, 2018-10-02:
I think my preference is the second option. Intel ME is probably the first one this is really needed but that's not to say it won't happen for other ones too later.

I think that LVFS can take hints in some cases to automatically do this field creation too for some CABs (such as those with an INF and XML file that detail both ESRT and a display version for Windows).

In doing this to the whole stack I think the important thing will be to make sure that all the tools and libraries can fall back to the old version safely in the various permutations of outdated library/daemon/client.

>The second option does seem more flexible, but it makes some things a lot harder. e.g. if we do something like

If you treat display version only for display then you shouldn't need to worry about comparison. Comparison can still be done the exact same as now.

hughsie, 2018-10-02:
> Comparison can still be done the exact same as now.

So we'd expect people requiring specific versions of the ME firmware to do `<requires><firmware version="0x12345678" compare="ge">GUID</requires>`?

superm1, 2018-10-02:
> I guess the key issue is that we don't know the encoding of the version number to display the device version

Oh hmm..
One idea that comes to mind is to let plugins quirk the device version with some kind of regex that transforms an integer.

hughsie, 2018-10-02:
> let plugins quirk the device version with some kind of regex that transforms an integer.

Like `VersionFormat=intel-me` or `VersionFormat=aaa+11.bbbbb.cccccccc.dddddddddddddddd`?

superm1, 2018-10-02:
I think between those two the latter makes more sense to me.  rather than (a potentially endless) description of intel-format, dell-format, lenovo-format and needing a decoder ring for them.

hughsie, 2018-10-02:
So, I've tried to prototype something like `aaa+11.bbbbb.cccccccc.dddddddddddddddd` -- you end up with a FORTH-style interpreter that's pretty hacky, which I also don't want to implement in a different language for the LVFS... I also don't think something like `aaa+11.bbbbb.cccccccc.dddddddddddddddd` is super easy to read in a metainfo file -- and humans just can't tell that `aaa+11.bbbbb.ccccccc.ddddddddddddddd` is missing a one `c` and a `d` compared to the previous string.

My gut instinct here is to just add the 10 or so lines in the LVFS and libappstream-glib for the `intel-me` format -- on the logic that if you're a tech giant like Intel you get to dictate custom version encodings, but if some small USB reseller tries to pull the same stunt, we can offer just `quad`, `triplet`, `pair` or `intel-me` as the only options allowed.

superm1, 2018-10-02:
Well another relevant question; is there a way to directly get this version string from the device somewhere/somehow without translation in the fwupd daemon?

hughsie, 2018-10-02:
> is there a way to directly get this version string from the device 

Not at the moment, is this something you need? It's really designed as an uninteresting implementation detail that's not really API or ABI.

superm1, 2018-10-02:
Not that I need, but I was going to say it might be another way out of needing to design another format to appstream-glib.  If the device can report it's own "display_version" without translation specifically.

YehezkelShB, 2018-10-02:
(even if the device could report the current one, don't you need a way to display the version of the suggested update?)

superm1, 2018-10-02:
If the device could report it's own current "display_version" then you would still need to plumb something through metadata and LVFS to display version incoming; yes.

hughsie, 2018-10-22:
I think close this one (now that we have intel-me and intel-me version formats) -- I'm quite happy to tell existing vendors that there are now enough defined ways to show a version number, and that they should use one of the existing ones...

closed 2018-10-22 by superm1