fwupd/fwupd #783: 8bitdo update handling flow problems

superm1, 2018-10-04:
I took my SF30 pro and tried to follow the wiki page recently created (https://github.com/hughsie/fwupd/wiki/8BitDo-Controllers) to do the update flow for it.

```
$ fwupdmgr get-devices
<snip> 
8Bitdo SF30 Pro
  DeviceId:             9438a0cff16b538296e104cfe4cef09713d82cf6
  Guid:                 e95bfb47-11f9-5afd-856e-fbc45730456d
  Guid:                 269b3121-097b-50d8-b9ba-d1f64f9cd241
  Guid:                 5bc60e62-f461-58bd-91b6-7de3392f8bd6
  Guid:                 043c7673-bd45-5e2e-b8b9-c22df2dfec2a
  Guid:                 b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e
  Summary:              A redesigned classic game controller
  Plugin:               ebitdo
  Flags:                updatable|supported|needs-bootloader|registered
  Vendor:               8Bitdo
  VendorId:             USB:0x2DC8
  Version:              1.23
  Icon:                 input-gaming
  Created:              2018-10-04

$ fwupdmgr update 
No upgrades for XPS 13 9370 System Firmware, current is 0.1.5.1: 0.1.5.1=same, 0.1.4.0=older, 0.1.3.3=older, 0.1.3.2=older, 0.1.2.1=older, 0.1.2.0=older
No upgrades for XPS 13 9370 Thunderbolt Controller, current is 33.00: 33.00=same, 28.00=older
Downloading 1.26 for 8Bitdo SF30 Pro...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Device 8Bitdo SF30 Pro needs to manually be put in update mode: Unplug the controller, hold down L1 + R1 + START until the red LED on top of the controller is flashing and then reconnect the controller.
```
I unplugged, turned off with start, pressed L1+R1+start and replugged

```
$ fwupdmgr update 
No upgrades for XPS 13 9370 System Firmware, current is 0.1.5.1: 0.1.5.1=same, 0.1.4.0=older, 0.1.3.3=older, 0.1.3.2=older, 0.1.2.1=older, 0.1.2.0=older
No upgrades for XPS 13 9370 Thunderbolt Controller, current is 33.00: 33.00=same, 28.00=older
```

Odd, but OK lets see if the device is actually there.
```
$ fwupdmgr get-devices
<snip>
8Bitdo  
  DeviceId:             9438a0cff16b538296e104cfe4cef09713d82cf6
  Guid:                 78acf201-a624-5c0b-a521-e204be14e3d5
  Guid:                 b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e
  Guid:                 5bc60e62-f461-58bd-91b6-7de3392f8bd6
  Guid:                 e95bfb47-11f9-5afd-856e-fbc45730456d
  Guid:                 269b3121-097b-50d8-b9ba-d1f64f9cd241
  Guid:                 043c7673-bd45-5e2e-b8b9-c22df2dfec2a
  Summary:              A redesigned classic game controller
  Plugin:               ebitdo
  Flags:                updatable|registered|is-bootloader
  Vendor:               8Bitdo
  VendorId:             USB:0x2DC8
  Version:              1.23
  Icon:                 input-gaming
  Created:              2018-10-04
```

Yeah it's there, so lets just try to manually call install.

```
$ fwupdmgr install 3d3a65ee2e8581647fb09d752fa7e21ee1566481-8Bitdo-SF30_Pro-SN30_Pro-1.26.cab 
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Installing on 8Bitdo  …[ -                                     ]
Restarting device…     [***************************************]failed to wait for replug after update: device 9438a0cff16b538296e104cfe4cef09713d82cf6 did not come back
```

That seems to work, except since the controller defaulted to switch mode rather than Android mode (start +b) it didn't show up in devices.

Turning it off and back on in the right mode I can see it did actually update properly.
```
8Bitdo SF30 Pro
  DeviceId:             9438a0cff16b538296e104cfe4cef09713d82cf6
  Guid:                 e95bfb47-11f9-5afd-856e-fbc45730456d
  Guid:                 269b3121-097b-50d8-b9ba-d1f64f9cd241
  Guid:                 5bc60e62-f461-58bd-91b6-7de3392f8bd6
  Guid:                 043c7673-bd45-5e2e-b8b9-c22df2dfec2a
  Guid:                 b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e
  Summary:              A redesigned classic game controller
  Plugin:               ebitdo
  Flags:                updatable|supported|needs-bootloader|registered
  Vendor:               8Bitdo
  VendorId:             USB:0x2DC8
  Version:              1.26
  Icon:                 input-gaming
  Created:              2018-10-04
```
Verbose log of daemon during this whole flow:
[fwupd_verbose.txt](https://github.com/hughsie/fwupd/files/2446771/fwupd_verbose.txt)

superm1, 2018-10-04:
(cc @picandocodigo just FYI since you were working on that wiki page)

picandocodigo, 2018-10-05:
This looks pretty similar to what I remember happening to me. So yeah, there seems to be an issue there, looks like when it's defaulting to Switch mode?

Thanks for keeping me in the loop, let me know if there's anything I can do to help.

amak79, 2018-10-12:
The same update flow also affects the NES30 Pro. I didn't force install the update with `fwupdmgr install` like @superm1 did, so the device still has the previous 4.01 firmware.

Gabitchov, 2019-01-03:
Hello,

I followed the blog post to try to update my gamepad firmware and
I had a weird issue with the update checker. It says that there is no update except there is one:
`No upgrades for 8Bitdo SN30 Pro, current is 1,03: 1.26=older`

There is a comma character in detected version of my gamepad:
```
$fwupdmgr get-devices
8Bitdo SN30 Pro
  DeviceId:             672c087de09848d9e7ee32aa1dea2fbeb8b81e6b
  Guid:                 1bf5ac96-99ce-54f2-abf4-7eabe6802533
  Guid:                 81becceb-f87b-5c66-b30a-b0958043d65d
  Guid:                 5bc60e62-f461-58bd-91b6-7de3392f8bd6
  Guid:                 043c7673-bd45-5e2e-b8b9-c22df2dfec2a
  Guid:                 b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e
  Guid:                 78acf201-a624-5c0b-a521-e204be14e3d5
  Summary:              A redesigned classic game controller
  Plugin:               ebitdo
  Flags:                updatable|supported|needs-bootloader|registered
  Vendor:               8Bitdo
  VendorId:             USB:0x2DC8
  Version:              1,03
  Icon:                 input-gaming
  Created:              2019-01-03
```
My version of fwupd is : 
```
$fwupdmgr --version  
client version:	1.1.4
checkout info:	1.1.4
compile-time dependency versions
	appstream-glib:	0.7.9
	gusb:	0.3.0
	efivar:	35
daemon version:	1.1.4
```
I am on an Ubuntu 16.04.10 distrib.

Have you an idea from where this issue could come ?

Regards,

Gabriel

hughsie, 2019-01-03:
This was fixed in https://github.com/hughsie/fwupd/pull/904

superm1, 2019-01-07:
Also, just backported it to 1_1_X branch so it will go into the 1.1.5 release when that happens if you can't adopt 1.2.x.

In terms of where to easily test the fix, fwupd 1.2.3 is released in the snap store now.  You can switch to that most easily.
https://snapcraft.io/fwupd

Gabitchov, 2019-01-08:
Hi @superm1,

I tried with the last version on snap, 1.2.3. This solves my problem withe the parsed version of the gamepad. But now, after dowloading of the version file, I have an other issue:
```
$fwupdmgr update
Downloading 1.26 for 8Bitdo SN30 Pro...
Decompressing?           [***************************************]
Authenticating?          [***************************************]
Could not check for auth: GDBus.Error:org.freedesktop.PolicyKit1.Error.Failed: Action org.freedesktop.fwupd.update-hotplug-trusted is not registered
```
Last time, I read similar issue in open ones. Maybe, it's related to another ticket.

Regards

superm1, 2019-01-08:
Do you by chance also still have the deb version installed before you installed the snap?

That seems to be a snap issue with policykit.  You might be able to try with sudo instead to avoid it.
I'll try to reproduce though.

Gabitchov, 2019-01-09:
Hi @superm1,

Indeed, install the deb version of fwupd solves the policykit issue. The only remaining minor issue is that the gamepad doesn't restart in last used mode so, at the end of installation, the gamepad is not redetected after the restart. But it seems also that the update changes the bluetooth id of the gamepad for the pairing settings(I needed to pair a new time after the update).

Thanks a lot.

Regards

Below my related payload report:
```
 {
                           "ReportVersion" : 2,
                           "MachineId" : "2b358082636154926fa96e7333020df01d7697882c4af0da5ce38c3ebd2aeac8",
                           "Metadata" : {
                             "DistroId" : "ubuntu",
                             "DistroVersion" : "18.04"
                           },
                           "Reports" : [
                             {
                               "Checksum" : "c33b8487ae9eb332061245a3986e29669f8e82b3",
                               "UpdateState" : 3,
                               "Guid" : "78acf201-a624-5c0b-a521-e204be14e3d5",
                               "Plugin" : "ebitdo",
                               "VersionOld" : "1.03",
                               "VersionNew" : "1.26",
                               "Flags" : 8226,
                               "Created" : 1547027407,
                               "Modified" : 1547027421,
                               "Metadata" : {
                                 "CpuArchitecture" : "x86_64",
                                 "RuntimeVersion(org.freedesktop.fwupd)" : "1.2.4",
                                 "DistroId" : "ubuntu",
                                 "CompileVersion(org.freedesktop.fwupd)" : "1.2.4",
                                 "CompileVersion(com.redhat.fwupdate)" : "12",
                                 "CompileVersion(org.freedesktop.gusb)" : "0.3.0",
                                 "RuntimeVersion(org.freedesktop.appstream-glib)" : "0.7.14",
                                 "KernelVersion" : "4.15.0-43-generic",
                                 "DistroVersion" : "18.04",
                                 "CompileVersion(com.redhat.efivar)" : "37",
                                 "RuntimeVersion(com.dell.libsmbios)" : "2.4",
                                 "BootTime" : "1546957646",
                                 "RuntimeVersion(com.redhat.fwupdate)" : "12"
                               }
                             }
                           ]
                         }
```

superm1, 2019-09-23:
This issue got a few different people commenting in it with a hodge podge of stuff, but the original issue from fwupd perspective the relevant thing to do is to add a quirk for these devices indicating they're "not going to come back" and to not show an error when they fail to replug.

commit 8a9f9d84 (2019-10-12): 8bitdo: use will-disappear flag for sf30/sn30 controllers (Fixes: #783)

These controllers reboot back into Nintendo Switch mode typically
and fwupd will not be able to probe them.
commit f37af548 (2019-10-12): 8bitdo: use will-disappear flag for sf30/sn30 controllers (Fixes: #783)

These controllers reboot back into Nintendo Switch mode typically
and fwupd will not be able to probe them.
closed 2019-10-12 by superm1