fwupd/fwupd #789: Not sure how to merge the xmlb branch

hughsie, 2018-10-11:
The new libxmlb branch drops the mandatory dep on appstream-glib and adds a hard dep on libxmlb, which is a super new library. I've asked people package libxmlb for gnome-software (which will also switch soon) but so far it's only packaged for Fedora.

Options we have:

 * ship a submodule copy and always build that
 * just disable CI targets until each distro packages the new library
 * build libxmlb locally if not available in the distro
 * support both the submodule and distro version

All have upsides and downsides. Some people hate the submodule -- distro people get quite upset. I'm open for ideas here. Building it in the CI makes it harder to get started locally, and also means it breaks building packages. Ideas welcome!

hughsie, 2018-10-11:
@superm1 -- any ideas?

superm1, 2018-10-11:
My opinions
**1st option**
Would that be part of tarball releases too then?  If it was a submodule, can it check out a tagged version of the library or it always gets mater?

**2nd option**
I don't think that is a smart one.  I really feel that ever since we introduced CI for so many different distros we often find things in one or the other that have made the project much higher quality.

**3rd and 4th**
My opinion would lean on these.
I made some changes to your `libxmlb-distro` branch to try to help get that working in the Debian cases.  There's a few patches to the build system that I think should be reverted when stuff does "land" in distro though.

I think if you do support a submodule option you should 'prefer' the system library if available in the build system, and only build submodule if not or too old.

superm1, 2018-10-11:
I've got your `libxmlb-distro` branch working on all the various distros that were failing now.  I think my preference is option 4 as long as the submodule is configured by default to checks out tagged releases and ends up in tarballs.

Whenever libxmlb is in all the distros that we care about, drop the submodule for the next tagged release.

hughsie, 2018-10-11:
I really appreciate that. I'll rebase and squash those down, do a final check and then do a PR.

superm1, 2018-10-11:
Sure.  If you keep the submodule approach can you make sure to update top level `README.md` to explain how to get the submodule into the right state?  I know that they support putting branches in `.gitmodules` but beyond that I've never worked with 'em before.

closed 2018-10-11 by hughsie