fwupd/fwupd #794: Dell dock code exploded on update

hughsie, 2018-10-12:
08:50:57:0061 FuMain               Emitting PropertyChanged('Status'='device-write')
    08:50:57:0061 FuPluginDellDock     Writing firmware version 05.03.05
    08:50:57:0061 FuEngine             no plugin specified for udev device /sys/devices/pci0000:00/0000:00:1c.4/0000:05:00.0/0000:06:02.0/0000:3d:00.0/usb3/3-1/3-1.5/3-1.5:1.0/0003:413C:B06E.0010/hidraw/hidraw4
    08:50:57:0061 FuPlugin             performing udev_device_added() on nvme
    08:50:57:0061 FuPlugin             performing udev_device_added() on udev
    08:50:57:0061 FuPlugin             performing udev_device_added() on unifying
    08:50:57:0063 FuPluginDellDock     attempt 1/5: set control transfer failed: endpoint stalled or request not supported
    08:50:58:0066 FuPluginDellDock     attempt 2/5: set control transfer failed: endpoint stalled or request not supported
    08:50:59:0069 FuPluginDellDock     attempt 3/5: set control transfer failed: endpoint stalled or request not supported
    08:51:00:0072 FuPluginDellDock     attempt 4/5: set control transfer failed: endpoint stalled or request not supported

    Thread 1 "fwupd" received signal SIGSEGV, Segmentation fault.
    0x0000000000436d68 in fu_plugin_runner_update (self=0x4bb6d0, device=0x82b790, blob_cab=0x536640, blob_fw=0x538190, flags=FWUPD_INSTALL_FLAG_NONE, error=0x7fffffffdc58)
        at ../src/fu-plugin.c:1412
    1412			fu_device_set_update_error (device, error_update->message);
    Missing separate debuginfos, use: dnf debuginfo-install bzip2-libs-1.0.6-28.fc29.x86_64 elfutils-libelf-0.174-1.fc29.x86_64 gdk-pixbuf2-2.38.0-4.fc29.x86_64 glib2-2.58.1-1.fc29.x86_64 gmp-6.1.2-8.fc29.x86_64 gnutls-3.6.4-1.fc29.x86_64 gpgme-1.11.1-3.fc29.x86_64 gvfs-client-1.38.1-1.fc29.x86_64 json-glib-1.4.4-1.fc29.x86_64 keyutils-libs-1.5.10-8.fc29.x86_64 krb5-libs-1.16.1-20.fc29.x86_64 libacl-2.2.53-2.fc29.x86_64 libarchive-3.3.2-2.fc29.x86_64 libattr-2.4.48-3.fc29.x86_64 libblkid-2.32.1-1.fc29.x86_64 libcap-2.25-12.fc29.x86_64 libffi-3.1-18.fc29.x86_64 libgcab1-1.1-3.fc29.x86_64 libgcc-8.2.1-2.fc29.x86_64 libgcrypt-1.8.3-3.fc29.x86_64 libgpg-error-1.31-2.fc29.x86_64 libgudev-232-4.fc29.x86_64 libgusb-0.3.0-2.fc29.x86_64 libpng-1.6.34-6.fc29.x86_64 libselinux-2.8-4.fc29.x86_64 libsoup-2.64.1-1.fc29.x86_64 libstemmer-0-11.585svn.fc29.x86_64 libtasn1-4.13-4.fc29.x86_64 libunistring-0.9.10-4.fc29.x86_64 libxml2-2.9.8-4.fc29.x86_64 libyaml-0.2.1-2.fc29.x86_64 nettle-3.4-5.fc29.x86_64 openssl-libs-1.1.1-3.fc29.x86_64 pcre-8.42-4.fc29.x86_64 pcre2-10.32-3.fc29.x86_64 polkit-libs-0.115-2.fc29.x86_64 sqlite-libs-3.24.0-2.fc29.x86_64 systemd-libs-239-3.fc29.x86_64 zlib-1.2.11-14.fc29.x86_64
    (gdb) bt
    #0  0x0000000000436d68 in fu_plugin_runner_update (self=0x4bb6d0, device=0x82b790, blob_cab=0x536640, blob_fw=0x538190, flags=FWUPD_INSTALL_FLAG_NONE, error=0x7fffffffdc58)
        at ../src/fu-plugin.c:1412
    #1  0x000000000041d16a in fu_engine_install_blob
        (self=0x482820, device_orig=0x82b790, blob_cab=0x536640, blob_fw2=0x538190, version=0x4beab4 "05.03.05", flags=FWUPD_INSTALL_FLAG_NONE, error=0x7fffffffdec8) at ../src/fu-engine.c:1526
    #2  0x000000000041c93a in fu_engine_install (self=0x482820, task=0x53c760, blob_cab=0x536640, flags=FWUPD_INSTALL_FLAG_NONE, error=0x7fffffffdec8) at ../src/fu-engine.c:1381

I'll add a workaround in the daemon to not explode when a plugin returns FALSE from the `FuPluginUpdateFunc` and doesn't set an error -- I couldn't see the bug in the Dell plugin. When I retried the update it worked.

superm1, 2018-10-12:
By chance do you have a coredump to go with that backtrace?

hughsie, 2018-10-12:
Unfortunately not, sorry. I'd redirected core dumps so I could use AFL.

hughsie, 2018-10-12:
Alas not, sorry. I know that's somewhat unhelpful, but I've been working 100.0% on the new 1.1.3 in RHEL.

PR #796 (2018-10-14): dell-dock: correct a scenario where an error message might not be set
This should fix #794 I think.
closed 2018-10-14 by hughsie