fwupd/fwupd #795: Package level of Dell dock has firmware updates, always

hughsie, 2018-10-12:
I've just done a dock upgrade and power cycled:

    [hughsie@hughsie-work build (1_1_X %)]$ fwupdmgr get-updates -v
    No upgrades for RTS5487 in Dell dock, current is 01.27: 01.27=same, 01.26=older
    No upgrades for RTS5413 in Dell dock, current is 01.10: 01.10=same, 01.09=older
    No upgrades for XXXXXXX 130W DP, current is 00.00.00.12: 00.00.00.12=same, 00.00.00.11=older
    Package level of Dell dock has firmware updates:
    GUID:                    8ceeeffd-51b6-580c-9b75-69143227aff8
    ID:                      com.dell.wd198ceeeffd.firmware
    Update Version:          00.00.06.01
    Update Name:             Dell WD19/WD19DC/WDTB Update
    Update Summary:          Update package for the WD19 family
    Update Remote ID:        dell-embargo
    Update Checksum:         SHA1(4b38d6b20d574e80b70748584ea326e0c0e624ef)
    Update Location:         https://fwupd.org/downloads/11a1b4cc6758e926cbe0f5c8c80730bd87754653-WD19FirmwareUpdateLinux_00.00.06.cab
    No upgrades for VMM5331 in Dell dock, current is 05.03.05: 05.03.05=same

So I do:

[hughsie@hughsie-work build (1_1_X %)]$ fwupdmgr update
No upgrades for RTS5487 in Dell dock, current is 01.27: 01.27=same, 01.26=older
No upgrades for RTS5413 in Dell dock, current is 01.10: 01.10=same, 01.09=older
No upgrades for xxxxxxxxxxxxx 130W DP, current is 00.00.00.12: 00.00.00.12=same, 00.00.00.11=older
Downloading 00.00.06.01 for Package level of Dell dock...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Updating Package level of Dell dock from 00.00.06.00 to 00.00.06.01…
Restarting device…     [***************************************]
No upgrades for VMM5331 in Dell dock, current is 05.03.05: 05.03.05=same

Then I do `get-devices`, and get:

    Package level of Dell dock
      DeviceId:             7bf361a43bd87062755675ce6285472728a4ed50
      ParentDeviceId:       ef3b3397619993975d045fa2cd00f379f823d0b8
      Guid:                 8ceeeffd-51b6-580c-9b75-69143227aff8
      Summary:              A representation of dock update status
      Plugin:               dell_dock
      Flags:                updatable|supported|registered
      Vendor:               Dell Inc
      Version:              00.00.06.00
      Created:              2018-10-12

so get-updates reports an update again, etc.

superm1, 2018-10-12:
Yes; known bug with EC. Should be fixed in next EC release. Will leave this open to track.

superm1, 2018-10-31:
The separate fix I alluded to is targeted for package "00.00.09.01"

superm1, 2018-11-26:
All the corner cases I mentioned are fixed in package 00.00.13.01.

closed 2018-11-26 by superm1