fwupd/fwupd #798: Please support elogind, a standalone version of logind

a17r, 2018-10-13:
It seems like all we would need is a patch to the build system to alternatively accept elogind. `HAVE_SYSTEMD` should probably be renamed to the more general `HAVE_LOGIND` in case the future brings further systemd support.

superm1, 2018-10-13:
No conceptual opposition; patches welcome

superm1, 2018-10-15:
If you (or someone else chooses to do this), please do note that `HAVE_SYSTEMD` is also going to be used by PR #787.  If a patch is created to the build system to accept elogind, please create an additional `HAVE_LOGIND` and the place that it makes sense in `src/fu-util.c` do something like `defined(HAVE_SYSTEMD) || defined(HAVE_LOGIND)`

PR #1040 (2019-02-25): Add elogind support
Closed #798

Type of pull request:
- [ X] Feature

I have tested this changeset on top of 1.2.4 release on Gentoo system with OpenRC. In the PR changes are rebased on top of master.
closed 2019-02-25 by hughsie