fwupd/fwupd #800: Need to fix SYSCONFDIR constant with default sysconfdir value

jtojnar, 2018-10-15:
On NixOS, packages can only occupy their own prefix in `/nix/store` directory and cannot install directly into `/etc`. Instead, we rely on modules, that describe the construction of `/etc` directory tree declaratively.

We still want to install the certificates somewhere in the prefix, so that we can link them from the module. Using the Meson default `sysconfdir` of `etc/` will install them to `$prefix/etc`, which works for us, but then, fwupd will also try to load the configuration from there at runtime. That is less fine, since `$prefix/etc` is immutable and we would like to use `/etc` since it is what the module generates.

Currently, we use the following patch but we would like to come up with something more upstreamable.

```diff
diff --git a/meson.build b/meson.build
index 96d752ae..fc5cb4dd 100644
--- a/meson.build
+++ b/meson.build
@@ -294,11 +294,17 @@
 
 plugin_dir = join_paths(libdir, 'fwupd-plugins-3')
 
+default_sysconfdir = get_option('sysconfdir')
+if default_sysconfdir == 'etc'
+  message('sysconfdir of etc makes no sense, using /etc')
+  default_sysconfdir = '/etc'
+endif
+
 conf.set_quoted('BINDIR', bindir)
 conf.set_quoted('LIBEXECDIR', libexecdir)
 conf.set_quoted('DATADIR', datadir)
 conf.set_quoted('LOCALSTATEDIR', localstatedir)
-conf.set_quoted('SYSCONFDIR', sysconfdir)
+conf.set_quoted('SYSCONFDIR', default_sysconfdir)
 conf.set_quoted('PLUGINDIR', plugin_dir)
 
 conf.set_quoted('GETTEXT_PACKAGE', meson.project_name())
@@ -310,12 +316,6 @@
   configuration : conf
 )
 
-default_sysconfdir = get_option('sysconfdir')
-if default_sysconfdir == 'etc'
-  message('sysconfdir of etc makes no sense, using /etc')
-  default_sysconfdir = '/etc'
-endif
-
 plugin_deps = []
 plugin_deps += appstream_glib
 plugin_deps += gio
```

I guess what we really want is separate flags for configuration/building and installation like autotools have.

superm1, 2018-10-15:
So I've got another potential solution to offer.  Fwupd has support to declare the directory to use for conf files at runtime using an environment variable: https://github.com/hughsie/fwupd/blob/master/src/fu-common.c#L937

Think that might work better for you guys?  That's what is used in the `SNAP` for similar reasons.

jtojnar, 2018-10-15:
We could pass the environment variable to the systemd service quite easily; I wonder if it would be enough, though. Making sure every program using libfwupd has the environment variable set would be hard, if we want to avoid setting the variable globally. For that reason, compile time option might be a better choice in this case.

What do you think about the patch in the OP? Do you think the printed message applies to `SYSCONFDIR` constant as well, or do you expect someone to use `$prefix/etc` default for storing configuration?

superm1, 2018-10-15:
You actually only need that variable set in the daemon.  So I think plumbing it at the systemd unit may very well be enough.
The client shouldn't be pulling anything from `/etc` at all.  

> What do you think about the patch in the OP? Do you think the printed message applies to SYSCONFDIR constant as well, or do you expect someone to use $prefix/etc default for storing configuration?

I do think that `$prefix/etc` can get used by `/usr/local` installation when you don't want to dirty your systemd-wide `/etc`.  For example I have fwupd 1.0.x with conf in `/etc` and I install 1.1.x into `/usr/local` and use conf directory `/usr/local/etc` for it.

The whole reason behind the way that b0aa3793 behaves is that meson used to pick pretty bad defaults for sysconfdir.  This was sorted in meson 0.44.0 ( https://github.com/mesonbuild/meson/issues/1637)  Currently this project targets meson 0.43.0, but if bumping to 0.44.0 I'd say we can probably drop that clause all-together.  I'll open a pull request for this.  https://github.com/hughsie/fwupd/pull/801.

superm1, 2018-10-16:
Between #801 and using environment variables I think you should be in good shape now.  Reopen and let me know if you think anything else is missing.

jtojnar, 2018-10-22:
The environment variable is not enough because the value of sysconfdir is also used in the service file: https://github.com/hughsie/fwupd/blob/7e5c6eebfebd1f69a87a43a1cf987f465d5800b6/data/meson.build#L41

jtojnar, 2018-10-22:
The environment variable is not enough because the sysconfdir is used in the service file: https://github.com/hughsie/fwupd/blob/7e5c6eebfebd1f69a87a43a1cf987f465d5800b6/data/meson.build#L41

superm1, 2018-10-22:
Hmm and I suppose it's not possible to

jtojnar, 2018-10-30:
Having separate `install_sysconfdir` which would default to `sysconfdir` and would be used just as install location would be quite simple solution. In the long term I need to promote implementing something like that in meson.

superm1, 2018-10-30:
Sounds fine to me, go ahead and open a PR and we'll see how it looks.

closed 2019-09-30 by stale[bot]