fwupd/fwupd #805: Support also traditional BIOSes (aka legacy/non-UEFI)?

petterreinholdtsen, 2018-10-17:
Is there any chance to have fwupd support also traditional BIOSes, for example the ones supported by flashrom?  I flashed my Acer Power M8 machine the other day using flashrom, and it would be great if it could have been done with fwupd.  Likewise for my Thinkpad X230.

What is the background for the UEFI limitation?  How hard would it be to handle non-UEFI BIOS-es?

-- 
Happy hacking
Petter Reinholdtsen

superm1, 2018-10-17:
I'll let @hughsie take a stab at the background.  I'd mention that most things in fwupd exist because manufactures cooperate, develop, test and publish firmware to LVFS.

Regarding `flashrom` though;
There actually is a `flashrom` plugin for fwupd.
https://github.com/hughsie/fwupd/tree/master/plugins/flashrom

If you add your `HwId` for your system to the quirk list, it will generate a fwupd device that you can try to use.
https://github.com/hughsie/fwupd/blob/master/plugins/flashrom/flashrom.quirk

hughsie, 2018-10-17:
We actually have a flashrom plugin, but alas no vendors are using it yet. Very few vendors care about hardware more than a couple of years old.

petterreinholdtsen, 2018-10-17:
[Mario Limonciello]
> Regarding `flashrom` though;
> There actually is a `flashrom` plugin for fwupd.
> https://github.com/hughsie/fwupd/tree/master/plugins/flashrom

Ah, cool.

> If you add your `HwId` for your system to the quirk list, it will
> generate a fwupd device that you can try to use.
> https://github.com/hughsie/fwupd/blob/master/plugins/flashrom/flashrom.quirk

How can I find my HwId?  'fwupdmgr get-devices' show nothing on the Acer
I successfully firmware upgraded the other day.

-- 
Happy hacking
Petter Reinholdtsen

petterreinholdtsen, 2018-10-18:
Well, the closest thing I can find is 'fwupdmgr get-devices --show-all-devices', and it is not showing the system/bios at all.  Where is the way to find hardware IDs using fwupd documented?

superm1, 2018-10-18:
What version of fwupd? You might need to check fwupdtool.

superm1, 2018-10-18:
`/usr/lib/fwupd/fwupdtool` or `/usr/libexec/fwupd/fwupdtool` with the argument `hwids`

superm1, 2018-10-18:
Then use fwupdtool

petterreinholdtsen, 2018-10-20:
Right, I got this output for the machine in question, which I
successfully firmware upgraded using flashrom.

# /usr/lib/fwupd/fwupdtool hwids
Computer Information
--------------------
BiosVendor: Phoenix Technologies, LTD
BiosVersion: R01-C0
Manufacturer: Acer
ProductName: ASE380/AST180/APM8
EnclosureKind: 3
BaseboardManufacturer: Acer
BaseboardProduct: EM61SM/EM61PM

Hardware IDs
------------
not available as 'Family' unknown
not available as 'Family' unknown
not available as 'BiosMajorRelease' unknown
not available as 'Family' unknown
not available as 'Family' unknown
not available as 'Family' unknown
not available as 'ProductSku' unknown
not available as 'ProductSku' unknown
{bece8aa6-1f3f-5ce5-991c-1b5ba5a50669}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{3fd875b7-91d3-508e-837d-423699b79ad5}   <- Manufacturer + ProductName
not available as 'Family' unknown
not available as 'Family' unknown
{4abf1d25-cde6-52ec-97f7-23c7d028a22b}   <- Manufacturer + EnclosureKind
{1b31f021-015f-52d4-897e-fa6d714bda2c}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{1e301734-5d49-5df4-9ed2-aa1c0a9dddda}   <- Manufacturer
#

Would it be possible for me to register such model for automatic
firmware upgrades?  If so, how?

And if I can not, who can and how would they do it?  Is the flashrom
lvfs procedure documented somewhere?

-- 
Happy hacking
Petter Reinholdtsen

superm1, 2018-10-21:
> Would it be possible for me to register such model for automatic firmware upgrades?

Let me make it clear how all this stuff normally works. OEMs test the new firmware using the Linux tools, upload it to LVFS and then customers pull it down from LVFS using fwupd.

If you don't have your OEM testing the upgrade and uploading to LVFS you have to do those steps manually *yourself*. Keep in mind all the inherent risks with flashing an untested firmware.

So registering the device with fwupd will let fwupd call flashrom for a properly registered device.

So the flow for you would be:
1. (One time) add your device to the flashrom quirk file.
2. (Every time) Download new BIOS from your OEM
3. (Every time) Create new XML metadata to describe the BIOS update
4. (Every time) Package the XML and BIOS payload into a CAB file
5. (Every time) Install the cab file

Does that make sense?

petterreinholdtsen, 2018-10-21:
Thank you for the explanation.  Where can I find documentation on the
steps 3-5, with details on XML format and paths used by fwupd?

[Mario Limonciello]
> If you don't have your OEM testing the upgrade and uploading to LVFS
> you have to do those steps manually *yourself*. Keep in mind all the
> inherent risks with flashing an untested firmware.

Given that the vendors of my hardware seem to ignore the models I got
(or worse, ignore LVFS completely), how hard is it to set up my own
private alternative source (in addition to LVFW), so I can test once and
automatically spread to the rest of the machines on site?  I would
rather not have to do the updates manually on each machine individually.

Can fwupd handle several sources of its firmware information?

-- 
Happy hacking
Petter Reinholdtsen

hughsie, 2018-10-22:
> so I can test once and automatically spread to the rest of the machines on site?

You can fairly easily spin up your own private LVFS instance, although see below. It might be easier to create the XML metadata file manually from a directory of .cab files with https://github.com/hughsie/lvfs-website/blob/master/local.py

>  I would rather not have to do the updates manually on each machine individually.

Do you have legal authority to distribute the firmware? Normally an OEM has permission to redistribute original or modified firmware, but various copyright, licensing and trademark issues stop people or customers doing this themselves. This is the reason we don't allow users to upload firmwares themselves to the LVFS, and only accept them with the legal blessing of the original OEM.

> Can fwupd handle several sources of its firmware information?

Sure, you can drop a new remote in `/etc/fwupd/remotes.d/` and it just works with all the existing fwupd tools.

superm1, 2018-10-22:
> Sure, you can drop a new remote in /etc/fwupd/remotes.d/ and it just works with all the existing fwupd tools.

Something to mention however is that fwupd is currently only configured to accept the signing key for LVFS.  If you create a private LVFS instance you'll need to also enroll the public portion of the GPG key that you use for signing with fwupd.

> steps 3-5, with details on XML format and paths used by fwupd?

The format is documented here: https://fwupd.org/metainfo
You can use `makecab.exe` on Windows or `gcab` on Linux to build the CAB itself.

iamohtep, 2018-11-07:
Hi,
i would also like to update a couple of Notebooks in our company which has Legacy BIOS since we installed it using our deployment system via network (pxe). We have some Thinkpad T470s series which already has a .cab file in https://fwupd.org/lvfs/device/7a176688-0960-47ba-931b-7829849e8347 . Can I use fwupdmgr with the flashrom plugin to update firmware? Can you guys help me how? I was trying fwupdmgr -v install {the-new-firmware.cab} without success. 

Regards,

hughsie, 2018-11-07:
The cabinet files contain UEFI capsules, and so I don't think that's going to work with Flashrom. I'd really advise against using flashrom unless you're 100% sure it's supported by the vendor. At least one laptop I own hard-freezes in the middle of writing flash using flashrom (interaction with the embedded controller). 

I really think switching to the UEFI boot option is going to be an easier process.

closed 2019-09-23 by hughsie