fwupd/fwupd #809: expose virtual device for end-to-end testing in gnome-software

mkrajnak, 2018-10-18:
The goal is to be able to test firmware update feature in gnome-software using the virtual device.

superm1, 2018-10-18:
This actually exists already, there is a plugin called `test`.  It's blacklisted by default, but for the purpose of end to end tests it's a simple `s/BlacklistPlugins=test/BlacklistPlugins=/` of `/etc/fwupd/daemon.conf`

superm1, 2018-10-18:
Well I guess depends on what you have in mind for an end to end test?  I don't think end to end tests need to rely on LVFS actually being up.  That's inviting CI to have a failure rate dependent upon a webservice.  Use a local remote and the metadata we use for the `fwupdmgr` installed-tests.

But that said I think if what it has isn't sufficient it can certainly be expanded upon.

closed 2018-11-29 by superm1