fwupd/fwupd #815: Set proxy settings?

gcorsini, 2018-10-23:
When opening GNOME software behind a corporate proxy (Ubuntu 18.04 LTS), it displays the following error message:
```
Unable to download firmware updates from "fwupd":
[*/cabinet/*source/fwupd/*] failed to download https://cdn.fwupd.org/downloads/firmware.xml.gz.asc:
Connection terminated unexpectedly
```

When the link is opened in the browser, it asks if the file should be downloaded. When using `wget https://cdn.fwupd.org/downloads/firmware.xml.gz.asc` the file is downloaded as expected. Hence, the server as well as the configured proxy settings seem to work fine.

Then, I used verbose mode to update the firmware in the terminal and this command failed again:
```shell
$ fwupdmgr -v refresh
(fwupdmgr:1565): Fu-DEBUG: 16:38:16.314: creating path /home/agsadmin/.cache/fwupdmgr
(fwupdmgr:1565): Fwupd-DEBUG: 16:38:16.314: looking for os-release at /etc/os-release
(fwupdmgr:1565): Fu-DEBUG: 16:38:16.314: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz to /home/agsadmin/.cache/fwupdmgr/lvfs-metadata.xml.gz
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz

Failed to download https://cdn.fwupd.org/downloads/firmware.xml.gz: SSL handshake failed
```

My best guess is that the default proxy settings are **not** taken into consideration by fwupd. If this is the case, **where can I specify the proxy settings for fwupd**?

Below are the requested outputs:

```shell
$ fwupdmgr --version
client version:	1.0.6
daemon version:	1.0.6
compile-time dependency versions
	appstream-glib:	0.7.4
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34
```

```shell
$ fwupdmgr get-devices
SVGA II Adapter
  DeviceId:             be516e1c4ee3695b820e40b1919c218265d54649
  Guid:                 8960d6bc-2120-5452-9bcf-cbc8a34b6fc6
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               VMware
  VendorId:             PCI:0x15AD
  Icon:                 audio-card
  Created:              2018-10-23

53c1030 PCI-X Fusion-MPT Dual Ultra320 SCSI (LSI Logic Parallel SCSI Controller)
  DeviceId:             3f812b7d0ddfc21b186b50690e58e373668169cc
  Guid:                 7c63a664-ee8b-58f3-8f82-65ef64ffc3b0
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               LSI Logic / Symbios Logic
  VendorId:             PCI:0x1000
  Icon:                 audio-card
  Created:              2018-10-23

VMXNET3 Ethernet Controller
  DeviceId:             7c39e72d0dd85c692f317a171df1f40da68914b5
  Guid:                 42035d2a-7b60-5f02-920b-b8556dbab7f9
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               VMware
  VendorId:             PCI:0x15AD
  Icon:                 audio-card
  Created:              2018-10-23
```

Answers to the questions:

- Operating system and version: 
    Ubuntu 18.04.1 LTS
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): 
    None, it was preinstalled and is used by GNOME Software showing the issue described at the beginning. 
- Have you tried rebooting? 
   Yes, several times while playing around with the proxy settings (in case some specific settings would be used).
- Are you using an NVMe disk? 
   I have no information about this. If needed, I could try to find out.
- Is secure boot enabled (only for the UEFI plugin)? 
   The UEFI plugin is not used. (At least I don't think so.)

superm1, 2018-10-23:
Did you configure the proxy settings via the GUI?  I think that puts them into `/etc/environment` right?

We might be missing this in our systemd unit:

```
diff --git a/data/fwupd.service.in b/data/fwupd.service.in
index 996ff4c..33601ea 100644
--- a/data/fwupd.service.in
+++ b/data/fwupd.service.in
@@ -8,6 +8,7 @@ Before=gdm.service
 Type=dbus
 BusName=org.freedesktop.fwupd
 ExecStart=@libexecdir@/fwupd/fwupd
+EnvironmentFile=-/etc/environment
 PrivateTmp=yes
 ProtectHome=yes
 ProtectSystem=full
```

You can try manually adding that to the systemd unit in `/lib/systemd/system/fwupd.service`, reloading the units (or rebooting) and see if that helps.

gcorsini, 2018-10-24:
Thanks a lot for the quick reply!

### Proxy:
I configured the proxy settings via GUI indeed. Additionally, I've put them in a script inside profile.d and made sure that these settings are kept when using sudo. The proxy settings are not stored in `/etc/environment`, at least they were not shown, when doing `cat /etc/environment`. Therefore, I added the two entries regarding proxies manually (after trying your proposed solution without success):
```
HTTP_PROXY="http://myproxy.com:proxyPort"
HTTPS_PROXY="https://myproxy.com:proxyPort"
```
Since Ubuntu has moved back to GNOME, the proxy settings are usually stored as gsettings. You can list them with the command `gsettings list-recursively org.gnome.system.proxy`.

### Testing proposed solution (no success):
In order to try your solution, I changed the file `/lib/systemd/system/fwupd.service` by adding the line `EnvironmentFile=/etc/environment` at the bottom of the unit part like:
```
[Unit]
Description=Firmware update daemon
Documentation=http://www.fwupd.org/
After=dbus.service
Before=gdm.service
EnvironmentFile=/etc/environment
```

As this didn't work, I looked more closely at your output and thought that I should have added the line to the service part like so:
```
[Service]
Type=dbus
BusName=org.freedesktop.fwupd
ExecStart=/usr/lib/fwupd/fwupd
EnvironmentFile=/etc/environment
MemoryDenyWriteExecute=yes
PrivateTmp=yes
ProtectControlGroups=yes
ProtectHome=yes
ProtectKernelModules=yes
ProtectSystem=full
RestrictAddressFamilies=AF_NETLINK AF_UNIX
RestrictRealtime=yes
```

As this kept producing the same failure, I added the hyphen in front of /etc like so `EnvironmentFile=-/etc/environment` and tested again once in the service and once in the unit part. Unfortunately, none of it worked.

### Code I looked at:
When I searched the code, I found that the error message is produced by the file `fwupd/fu-util.c` in the function [fu_util_download_file](https://github.com/hughsie/fwupd/blob/62f8486ffe8b6b61ad6be4f39c96aa064a05f198/src/fu-util.c#L1254), I think. In my oppinion, one of the function calls in there is not taking into consideration. It might be in the function fu_util_setup_networking (line 1275) or soup_session_send_message (line 1306).
Currently, I can't pull the code and therefore can't follow these leads. If I get around to it, I will let you know.

hughsie, 2018-10-24:
The daemon doesn't care about proxies, as it's not the thing doing internet access. All the client tools download on behalf of the daemon, and *squirt* them into the daemon using a file descriptor. The proxy settings are obtained from the `https_proxy` and `http_proxy` environment variables (lower case!) that should be set in your /session/.

Reading around, it appears some apps use the upper case version, some the lower case version -- e.g. https://superuser.com/questions/876100/https-proxy-vs-https-proxy -- maybe we should fall back too?

closed 2018-10-24 by hughsie

gcorsini, 2018-10-24:
Thanks for your reply. In my system, I've set all four: http_proxy, https_proxy, HTTP_PROXY and HTTPS_PROXY. Using `printenv` lists them (also with sudo). Additionally, I've set the proxies in a file inside `/etc/apt/apt.conf.d/` as well.
I can download software without issues for the moment. But every time, it's opened the following image is displayed:
[image]

As of now, I still believe that there is some proxy issue. Either I set up something incorrectly, GNOME software ignores the proxies in combination with fwupd or fwupd ignores the proxy.

hughsie, 2018-10-24:
Does it work in fwupdmgr? I didn't test gnome-software at all yet.

gcorsini, 2018-10-24:
> 
> 
> Does it work in fwupdmgr? I didn't test gnome-software at all yet.

No, I don't think so:
```
$ fwupdmgr -v refresh
(fwupdmgr:2037): Fu-DEBUG: 17:47:56.626: creating path /home/agsadmin/.cache/fwupdmgr
(fwupdmgr:2037): Fwupd-DEBUG: 17:47:56.628: looking for os-release at /etc/os-release
(fwupdmgr:2037): Fu-DEBUG: 17:47:56.629: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz to /home/agsadmin/.cache/fwupdmgr/lvfs-metadata.xml.gz
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz

Failed to download https://cdn.fwupd.org/downloads/firmware.xml.gz: SSL handshake failed
```

> 
> 
> With gnome-software you might need to log out and back in because gnome-software starts at session startup and if you configure the proxy after that the environment for it won't contain the proxy information.

I restarted the machine and the command `printenv | grep http` displays all the aforementioned proxies (http_proxy, https_proxy, HTTP_PROXY and HTTPS_PROXY) with the correct values.

hughsie, 2018-10-24:
> SSL handshake failed

I wonder if glib-networking or ca-certificates are the problem. It sounds a lot like https://bugzilla.redhat.com/show_bug.cgi?id=1623471

hughsie, 2018-10-29:
> Sorry!

No worries, no need to apologize. Feel free to re-open when you have time.

sei-vsarvepalli, 2021-04-24:
I know this issue is closed, but it happened to come up in one of our customer's environment.  All their internal servers in their lan do not have direct Internet connection.  They have found two ways to solve this issue and shared it with me, for anyone reading this issue again, can perhaps use it:

1. Use systemd's global environment (Manager/DefaultEnvironment)so all the systemd child processes that start many service can use these including in /etc/systemd/system.conf
```
[Manager]
DefaultEnvironment=http_proxy=http://proxy.internal.corp:3128/ https_proxy=http://proxy.internal.corp:3128/
```
2. Use fwupd.service systemd's environment (service/PassEnvironment) to set proxy environment variables that will be used by fwupd.service in /etc/systemd/system/fwupd.service (don't forget to "systemctl daemon-reload" after this settings has changed)
```
[Service]
Environment=http_proxy=http://proxy.internal.corp:3128/ https_proxy=http://proxy.internal.corp:3128/
PassEnvironment=http_proxy https_proxy
```

Hope that works in your environments as well
Vijay

superm1, 2021-04-24:
The daemon doesn't talk to the web though. All of that is client communication. So it should need to be in the client's environment you setup proxy.

hughsie, 2021-04-25:
I think Vijay means fwupd-refresh.service, not fwupd.service -- I'm not immediately sure of the best way to inherit proxy settings for a service file. Will tweet.

superm1, 2021-04-25:
Ah I see... There was an explicit mention to fwupd.service which is why I thought daemon.

For fwupd-refresh.service I think that first option of making it global for all systemd services is probably the best and most scalable solution.

sei-vsarvepalli, 2021-04-26:
> I think Vijay means fwupd-refresh.service, not fwupd.service -- I'm not immediately sure of the best way to inherit proxy settings for a service file. Will tweet.

It makes sense for this setting to be in fwupd-refresh.service. I will check to see in the test environment and update the ticket.

hughsie, 2021-04-26:
@sei-vsarvepalli I think https://github.com/fwupd/fwupd/pull/3189 might make you happy

sei-vsarvepalli, 2021-04-27:
Just confirming that the /etc/systemd/system/fwupd-refresh.service is the service file to be updated in Debian and Ubuntu.   This will route fwupd-refresh traffic through the proxy using the environment.  Looks like @hughsie has better plans with #3189 with lib proxy as well.  

Thank you
Vijay

```
*** /lib/systemd/system/fwupd-refresh.service	2020-06-18 16:04:18.000000000 +0000
--- fwupd-refresh.service	2021-04-26 15:44:26.787620498 +0000
***************
*** 15,17 ****
--- 15,19 ----
  RestrictRealtime=yes
  SuccessExitStatus=2
  ExecStart=/usr/bin/fwupdmgr refresh --no-metadata-check
+ Environment=http_proxy=http://proxy.internal.corp:3128/ https_proxy=http://proxy.internal.corp:3128/
+ PassEnvironment=http_proxy https_proxy
```

hughsie, 2021-04-27:
@sei-vsarvepalli -- would you be able to ask the customer if libproxy support is sufficient? I think basically everything uses libproxy, and I'd be amazed if it didn't do what they need.

sei-vsarvepalli, 2021-04-27:
> @sei-vsarvepalli -- would you be able to ask the customer if libproxy support is sufficient? I think basically everything uses libproxy, and I'd be amazed if it didn't do what they need.

Yes - We had a chat about it and a look at the current PR.  The libproxy support will be plus for sure.  Current setup of environment in systems is not easy to manage.

aFineBoi, 2021-12-23:
I'm in a working environment, and I've also got similar network problems.

Config files for proxy were created for a workaround, one is /etc/systemd/system/fwupd.service.d/http-proxy.conf, another is /etc/systemd/system/fwupd-refresh.service.d/http-proxy.conf
```
[Service]
Environment="http_proxy=http://1.2.3.4:5678/"
```

After reboot, I verified the proxy with these:
```
# systemctl status fwupd
● fwupd.service - Firmware update daemon
     Loaded: loaded (/lib/systemd/system/fwupd.service; static; vendor preset: enabled)
    Drop-In: /etc/systemd/system/fwupd.service.d
             └─http-proxy.conf
     Active: active (running) since Thu 2021-12-23 20:56:18 AWST; 21min ago
       Docs: https://fwupd.org/
   Main PID: 2879 (fwupd)
      Tasks: 5 (limit: 18943)
     Memory: 15.9M
     CGroup: /system.slice/fwupd.service
             └─2879 /usr/libexec/fwupd/fwupd

Dec 23 20:56:17 dell-g3-3590 systemd[1]: Starting Firmware update daemon...
Dec 23 20:56:18 dell-g3-3590 systemd[1]: Started Firmware update daemon.
```
```
# systemctl status fwupd-refresh
● fwupd-refresh.service - Refresh fwupd metadata and update motd
     Loaded: loaded (/lib/systemd/system/fwupd-refresh.service; static; vendor preset: disabled)
    Drop-In: /etc/systemd/system/fwupd-refresh.service.d
             └─http-proxy.conf
     Active: inactive (dead)
TriggeredBy: ● fwupd-refresh.timer
       Docs: man:fwupdmgr(1)
```
```
# systemctl show fwupd-refresh | grep proxy
Environment=http_proxy=http://1.2.3.4:5678/
DropInPaths=/etc/systemd/system/fwupd-refresh.service.d/http-proxy.conf
```

I believe I've got this right. My Snap Store has similar proxy settings and it works just fine. However, fwupd is just not working:
```
# fwupdmgr refresh --force
Updating lvfs
Downloading…             [ -                                     ]failed to download file: Operation timed out after 60001 milliseconds with 0 out of 0 bytes received
```
Got any idea? Thanks in advance.

marc-hb, 2023-07-08:
As of Ubuntu 22.04 LTS, @sei-vsarvepalli's `/lib/systemd/system/fwupd-refresh.service` solution above seems still required and... it works! Thanks.

Before:
```
journalctl -b / systemctl status fwupd-refresh

Jul 08 02:54:16  fwupdmgr[3199]: Downloading…: 0%
Jul 08 02:55:12  fwupdmgr[3199]: Idle…: 0%
Jul 08 02:55:12  systemd[1]: fwupd-refresh.service: Main process exited, code=exited, status=1/FAILURE
Jul 08 02:55:12  systemd[1]: fwupd-refresh.service: Failed with result 'exit-code'.
Jul 08 02:55:12  systemd[1]: Failed to start Refresh fwupd metadata and update motd.
```

After:

```
journalctl -f 

systemctl daemon-reload            # <= don't forget this!
systemctl restart fwupd-refresh

Jul 08 02:56:21  fwupdmgr[3357]: Downloading…: 0%
Jul 08 02:56:21  fwupdmgr[3357]: Downloading…: 1%
Jul 08 02:56:21  fwupdmgr[3357]: Downloading…: 3%
...

```

hughsie, 2023-07-08:
I wondered if someone could be kind enough to write a man page for `fwupd-refresh.service` please? It would be something like https://github.com/fwupd/fwupd/blob/main/docs/fwupd-remotes.d.md and mention the proxy setting there. Anything basic would be better than nothing and we can add more content after that.

hughsie, 2023-07-10:
I had a bit of time this morning and came up with ^^