fwupd/fwupd #818: fwupdmgr update doesn't always grab latest CAB file for composite updates

superm1, 2018-10-24:
I've seen this occur on a Dell dock from the factory (IE original firmware on this board prototype).  There are two updates available on LVFS.

**Update 1**
Package 00.00.00.06
RTS5487 1.27
EC 12
RTS5413 1.10
VMM5331 5.3.5

**Update 2**
Package 00.00.00.07
RTS5487 1.27
EC 14
RTS5413 1.10
VMM5331 5.3.6

Running `# fwupdmgr update` will fetch the list of devices, and then fetch the CAB for the first one and daemon will process the update on all components in that CAB.

In my test run it fetched "Update 1" since both updates had `RTS5487 1.27` in them.  This meant that it installed all the stuff (it could) in that update and then proceeded to try to install the stuff from update 2.
This failed because the dock was actually still rebooting after finishing the stuff in update 1 so those devices had gone away and not come back yet.

PR #816 will in general help this since EC is the thing that changes most frequently and it will bump to the front of the list but I don't think it's a proper fix.

What I was thinking is a real solution is to store a timestamp associated with the release and let the engine sort by that.  Then if a bunch of releases (which come from composite CAB files) provide the same firmware and that device is flashed first then it will at least grab the newest one.

hughsie, 2018-10-24:
>  store a timestamp associated with the release

Isn't that exactly what `date="2018-10-09" ` lets you do?

> then it will at least grab the newest one

Sounds sane to me. I'm about to go on PTO for 2 days, but I'm happy to review if I have internet access at the hotel.

superm1, 2018-10-25:
I dug a little further into this, and it seems like it's actually appstream-glib on the LVFS side that subsumes the releases when the version number is the same.

So I think this should fix it (at least locally generated metadata splits into two releases now).
https://github.com/hughsie/appstream-glib/pull/267

hughsie, 2018-10-29:
I maybe think just not using libappstream-glib in metadata.py might be the best plan -- we're not using the dedupe feature of if anymore, so I'm struggling to see why we're just not writing the XML using something like `xml.etree.ElementTree` -- if you agree I can spend some time porting it.

closed 2018-11-01 by hughsie