fwupd/fwupd #819: Please provide updated Thunderbolt NVM (36) for XPS 9575

rstrube, 2018-10-24:
Hi there!

I noticed that Dell has updated their Thunderbolt firmware for the XPS 9575 to version 36.  See:
https://www.dell.com/support/home/us/en/04/drivers/driversdetails?driverId=PR8NR&osCode=WT64A&productCode=xps-15-9575-2-in-1-laptop

Unfortunately this new firmware does not appear in the lvfs *or* lvfs-testing repositories.

Is there any chance you could add this firmware to lvfs-testing? I'm having problems with getting my eGPU working over TB3, and I'd like to be running the fully updated firmware to rule out any firmware issues that might be causing problems.

- Operating system and version: Ubuntu 18.10
- How did you install fwupd: came with OS
- Have you tried rebooting? N/A
- Are you using an NVMe disk? No
- Is secure boot enabled? No

```shell
$ fwupdmgr --version
client version:	1.1.2
daemon version:	1.1.2
compile-time dependency versions
	appstream-glib:	0.7.12
	gusb:	0.2.11
	efivar:	34
```

```shell
XPS 9575 Thunderbolt Controller
  DeviceId:             069ac71f347e92d158f2c211cca10d52a19e2d41
  Guid:                 8926f505-8219-5d6c-969a-e927534113fb
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              30.00
  Icon:                 computer
  Created:              2018-10-24
```

Thanks!
Rob

Please note I have another issue created in the old dell *thunderbolt-nvm-linux** repository here: https://github.com/dell/thunderbolt-nvm-linux/issues/15, apologies for the duplication but it looks like this repo is much more active.

superm1, 2018-10-24:
That's the right place to ask, closing this issue.

closed 2018-10-24 by superm1

rstrube, 2018-10-24:
OK apologies.  Thanks!