fwupd/fwupd #820: system UEFI firmware fails to update on Thinkpad P52 (20M90019RT)

rugubara, 2018-10-28:
The process of updating firmware stops when UEFI program for update starts (after a reboot). A blue screen appears (design resembling Windows 10 Boot manager, I was frightened as I removed any traces of Windows from my disks) 

> Your PC/Device needs to bе repaired
> A required device isn't connected or can't be accessed
> Error code: 0xc0000225
> 
> You'll need to use recovery tools. If you don't have any installation media (like a disc or USB device), contact your PC administrator or PC/Device manufacturer.
> 

After seeing it for the 1st time, I did a manual reboot and entered UEFI boot menu and chose Linux Firmware Updater. It took me to the same screen. 
I rebooted into Recovery Environment by pressing F1 and it took me to my default GRUB menu removing UEFI boot entry for Linux Firmware Updater.
 
Please direct me how I can collect more information about this error so you could fix it.

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version: 1.1.3
compile-time dependency versions
        appstream-glib: 0.7.9
        gusb:   0.2.11
        efivar: 36
daemon version: 1.1.3

```
```shell
$ fwupdmgr get-devices
20M90019RT System Firmware
  DeviceId:             65b6a9dc7b7df18bdff003584b51bf21373e3aa6
  Guid:                 1e1fe415-74e8-49e1-9508-106b3d13d50d
  Guid:                 171800c9-1a51-5fd9-a32b-7b3999cb1c4e
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.12
  VersionLowest:        0.1.0
  Icon:                 computer
  Created:              2018-10-28
  UpdateState:          success

UEFI Device Firmware
  DeviceId:             619057f1d0da67b3fa4e9f74e09fc9218ce32643
  Guid:                 971682af-21a0-46e3-91c0-825702c2ba70
  Guid:                 b892b4fd-841b-5ec5-a700-437299fb233f
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              192.7.1122
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-10-28

UEFI Device Firmware
  DeviceId:             0f3c64df11304fc5711ef77a93e3e042e2738907
  Guid:                 88b8ba7e-296b-4f9e-929f-dcc19318dbcf
  Guid:                 629122db-a18d-5770-a3f3-1e3d42907f7c
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.10
  VersionLowest:        0.1.10
  Icon:                 audio-card
  Created:              2018-10-28

Samsung SSD 970 EVO 2TB
  DeviceId:             03281da317dccd2b18de2bd1cc70a782df40ed7e
  Guid:                 47335265-a509-51f7-841e-1c94911af66b
  Guid:                 8fd4ca73-d0ae-52e8-8977-461435c6f4cf
  Serial:               S464NB0K704432V
  Summary:              NVM Express Solid State Drive
[... 620 lines cut ...]
├── initramfs-4.18.15-gentoo-zfs.img
├── initramfs-4.18.16-gentoo-zfs.img
├── System.map-4.18.14-gentoo-zfs
├── System.map-4.18.14-gentoo-zfs.old
├── System.map-4.18.15-gentoo-zfs
├── System.map-4.18.16-gentoo-zfs
├── System.map-4.18.16-gentoo-zfs.old
├── System.map-4.19.0-gentoo-zfs
├── vmlinuz-4.18.14-gentoo-zfs
├── vmlinuz-4.18.14-gentoo-zfs.old
├── vmlinuz-4.18.15-gentoo-zfs
├── vmlinuz-4.18.16-gentoo-zfs
├── vmlinuz-4.18.16-gentoo-zfs.old
└── vmlinuz-4.19.0-gentoo-zfs

57 directories, 509 files
```

Please answer the following questions:

- Operating system and version: Gentoo linux, latest unstable amd64
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): fwupd is installed from main distribution repository gentoo
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

superm1, 2019-02-28:
It sounds to me like the boot entries might have just been removed.  There was some discussion about this type of behavior with Lenovo lately and some changes landed in fwupd 1.2.5 that affect them.  Can you please upgrade to fwupd 1.2.5 before doing another upgrade in the future?  If it happens again you can re-open the bug or open a new one.

rugubara, 2019-02-28:
I happened to upgrade to 1.2.5 recently and I did another upgrade. I didn't observe the behaviour described in this issue. There was a reset of my boot entries to a default list instead. 

Please have a look at https://bugs.gentoo.org/678862

superm1, 2019-02-28:
It sounds like there is still some lenovo bug then even with the fixed entries from 1.2.5.  @hughsie can you talk to them?

superm1, 2019-03-01:
#709 looks like duplicate too.

@hughsie at least until lenovo gets to the bottom of this, I wonder if it's worth taking a big hammer approach - require all lenovo systems to have a `\efi\boot\bootx64.efi` in place before updating?

rugubara, 2019-03-01:
>
> @hughsie at least until lenovo gets to the bottom of this, I wonder if it's worth taking a big hammer approach - require all lenovo systems to have a `\efi\boot\bootx64.efi` in place before updating?

That's what I ended up with. It seems that grub stores the information about location of grub.cfg to this file - so it needs updating after every grub-install.