fwupd/fwupd #822: Lenovo T480s firmware update not working in gnome-software or fwupdmgr

bliemli, 2018-10-30:
I have a very similar issue to #746, my setup is slightly different though (see below).

Symptoms are: Clicking "Update" in Gnome Software doesn't do anything except showing a progress bar for a short time and then showing the exact same update entry again.
`fwupdmgr update` does its thing and then asks to reboot the machine but doesn't boot into the firmware flashing mode.

```shell
$ fwupdmgr --version
client version:	1.0.9
daemon version:	1.0.9
compile-time dependency versions
	appstream-glib:	0.7.11
	gusb:	0.3.0
	fwupdate:	11
	efivar:	35
```

```shell
$ fwupdmgr get-devices
Thinkpad T480s Thunderbolt Controller
  DeviceId:             3efd1102feb65c9496b59ce3521bdc5e7f47247f
  Guid:                 061f2847-061b-5cfb-9c39-3dbfe2b49d69
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              14.00
  Icon:                 computer
  Created:              2018-10-30

ThinkPad Thunderbolt 3 Dock
  DeviceId:             2912fd14f88799eb4d5d79d2ee8656b79f9e43b6
  Guid:                 df675f10-53ac-59ca-bddd-0a86ee492920
  Plugin:               thunderbolt
  Flags:                updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0108
  Version:              15.00
  Icon:                 audio-card
  Created:              2018-10-30

UEFI Device Firmware
  DeviceId:             cf3ad1e5e6838b09cba64bb5ccb1142dd44995d2
  Guid:                 676af093-2a5c-4238-9c29-db8063a33532
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              184.50.3460
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-10-30

UEFI Device Firmware
  DeviceId:             323bda90831520867f8c02544efb9711dfca9fb6
  Guid:                 4d254d6e-cd67-477b-97d5-bc3048af45c4
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.8
  VersionLowest:        0.1.8
  Icon:                 audio-card
  Created:              2018-10-30

20L7001LMZ System Firmware
  DeviceId:             e7cc234e82d08b6a14e246beb6b5f8142091e570
  Guid:                 ebfe8df8-dee7-4692-a721-cbcf5095c5cf
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.23
  VersionLowest:        0.0.1
  Icon:                 computer
  Created:              2018-10-30
  Modified:             2018-10-30
  UpdateState:          needs-reboot
```

```shell
$ efibootmgr -v
BootNext: 0000
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0017,0018,0019,001A,001B,001C,001D,001E,0023,0000
Boot0000* Linux-Firmware-Updater \fwupx64.efi	HD(1,GPT,51aa2291-6925-45f6-a790-1b025fa0f829,0x800,0x64000)/File(\EFI\fedora\shimx64.efi)\.f.w.u.p.x.6.4...e.f.i...
Boot0001* Fedora	HD(1,GPT,51aa2291-6925-45f6-a790-1b025fa0f829,0x800,0x64000)/File(\EFI\fedora\shimx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001E  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot001F* USBR BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0020* USBR BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0021* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0022* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0023* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupdate-ebfe8df8-dee7-4692-a721-cbcf5095c5cf-0
```

```shell
$ sudo tree /boot
/boot
├── config-4.18.13-200.fc28.x86_64
├── config-4.18.14-200.fc28.x86_64
├── config-4.18.16-200.fc28.x86_64
├── efi
│   ├── EFI
│   │   ├── BOOT
│   │   │   ├── BOOTIA32.EFI
│   │   │   ├── BOOTX64.EFI
│   │   │   ├── fallback.efi
│   │   │   ├── fbia32.efi
│   │   │   └── fbx64.efi
│   │   └── fedora
│   │       ├── BOOT.CSV
│   │       ├── BOOTIA32.CSV
│   │       ├── BOOTX64.CSV
│   │       ├── fonts
│   │       │   └── unicode.pf2
│   │       ├── fw
│   │       │   └── fwupdate-p35Frr.cap
│   │       ├── fwupia32.efi
│   │       ├── fwupx64.efi
│   │       ├── gcdia32.efi
│   │       ├── gcdx64.efi
│   │       ├── grub.cfg
│   │       ├── grubenv
│   │       ├── grubia32.efi
│   │       ├── grubx64.efi
│   │       ├── loader
│   │       │   └── entries
│   │       ├── mmia32.efi
│   │       ├── mmx64.efi
│   │       ├── MokManager.efi
│   │       ├── shim.efi
│   │       ├── shimia32.efi
│   │       ├── shimia32-fedora.efi
│   │       ├── shimx64.efi
│   │       └── shimx64-fedora.efi
│   ├── mach_kernel
│   └── System
│       └── Library
│           └── CoreServices
│               └── SystemVersion.plist
├── elf-memtest86+-5.01
├── extlinux
│   ├── cat.c32
│   ├── chain.c32
│   ├── cmd.c32
│   ├── cmenu.c32
│   ├── config.c32
│   ├── cptime.c32
│   ├── cpu.c32
│   ├── cpuid.c32
│   ├── cpuidtest.c32
│   ├── debug.c32
│   ├── dhcp.c32
│   ├── dir.c32
│   ├── disk.c32
│   ├── dmi.c32
│   ├── dmitest.c32
│   ├── elf.c32
│   ├── ethersel.c32
│   ├── gfxboot.c32
│   ├── gpxecmd.c32
│   ├── hdt.c32
│   ├── hexdump.c32
│   ├── host.c32
│   ├── ifcpu64.c32
│   ├── ifcpu.c32
│   ├── ifmemdsk.c32
│   ├── ifplop.c32
│   ├── kbdmap.c32
│   ├── kontron_wdt.c32
│   ├── ldlinux.c32
│   ├── lfs.c32
│   ├── libcom32.c32
│   ├── libgpl.c32
│   ├── liblua.c32
│   ├── libmenu.c32
│   ├── libutil.c32
│   ├── linux.c32
│   ├── ls.c32
│   ├── lua.c32
│   ├── mboot.c32
│   ├── memdisk
│   ├── meminfo.c32
│   ├── menu.c32
│   ├── pci.c32
│   ├── pcitest.c32
│   ├── pmload.c32
│   ├── poweroff.c32
│   ├── prdhcp.c32
│   ├── pwd.c32
│   ├── pxechn.c32
│   ├── reboot.c32
│   ├── rosh.c32
│   ├── sanboot.c32
│   ├── sdi.c32
│   ├── sysdump.c32
│   ├── syslinux.c32
│   ├── vesa.c32
│   ├── vesainfo.c32
│   ├── vesamenu.c32
│   ├── vpdtest.c32
│   ├── whichsys.c32
│   └── zzjson.c32
├── grub2
│   ├── grubenv -> ../efi/EFI/fedora/grubenv
│   └── themes
│       └── system
│           ├── background.png
│           └── fireworks.png
├── initramfs-0-rescue-86cc09b2641a4c6cac2d475eab21c6aa.img
├── initramfs-0-rescue-d1043b50b87f4e5289641433ef6aa716.img
├── initramfs-4.18.13-200.fc28.x86_64.img
├── initramfs-4.18.14-200.fc28.x86_64.img
├── initramfs-4.18.16-200.fc28.x86_64.img
├── loader
│   └── entries
├── lost+found
├── memtest86+-5.01
├── System.map-4.16.3-301.fc28.x86_64
├── System.map-4.18.13-200.fc28.x86_64
├── System.map-4.18.14-200.fc28.x86_64
├── System.map-4.18.16-200.fc28.x86_64
├── vmlinuz-0-rescue-86cc09b2641a4c6cac2d475eab21c6aa
├── vmlinuz-0-rescue-d1043b50b87f4e5289641433ef6aa716
├── vmlinuz-4.18.13-200.fc28.x86_64
├── vmlinuz-4.18.14-200.fc28.x86_64
└── vmlinuz-4.18.16-200.fc28.x86_64

18 directories, 111 files
```

Please answer the following questions:

- Operating system and version: Fedora 28
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): `dnf`
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? yes

Thanks in advance.

bliemli, 2018-10-31:
Just upgraded to Fedora 29. Gnome Software still didn't work but `fwupdmgr update` did.

joakim-tjernlund, 2018-11-21:
I got the same problem
fwupdmgr --version
client version:	1.1.4
compile-time dependency versions
	appstream-glib:	0.7.9
	gusb:	0.2.11
	efivar:	36
daemon version:	1.1.4
se-jocke-lx ~ # fwupdmgr get-devices
20L7001UMX System Firmware
  DeviceId:             e7cc234e82d08b6a14e246beb6b5f8142091e570
  Guid:                 ebfe8df8-dee7-4692-a721-cbcf5095c5cf
  Guid:                 2b79e0a0-0801-5fc9-a7a5-90374e924a9b
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.25
  VersionLowest:        0.0.1
  Icon:                 computer
  Created:              2018-11-21
  UpdateState:          success

UEFI Device Firmware
  DeviceId:             cf3ad1e5e6838b09cba64bb5ccb1142dd44995d2
  Guid:                 676af093-2a5c-4238-9c29-db8063a33532
  Guid:                 039bd398-f61a-5c46-b02c-b660d1cf4926
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              184.50.3460
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-11-21

UEFI Device Firmware
  DeviceId:             323bda90831520867f8c02544efb9711dfca9fb6
  Guid:                 4d254d6e-cd67-477b-97d5-bc3048af45c4
  Guid:                 cc9cf2fe-ba41-532b-bc58-17ca9cc3c63d
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.8
  VersionLowest:        0.1.8
  Icon:                 audio-card
  Created:              2018-11-21

fwupd wants to install 0.1.25 aand fwupd update does somethings, reboot and the
machine just does a plian reboot, not update takes place

joakim-tjernlund, 2018-11-21:
efivar -l  >bef
se-jocke-lx ~ # fwupdmgr update
Downloading 0.1.27 for 20L7001UMX System Firmware...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Updating 20L7001UMX System Firmware…                           ]
Scheduling…            [***************************************]

An update requires a reboot to complete. Restart now? [Y|n]: n
se-jocke-lx ~ # efivar -l  > aft
se-jocke-lx ~ # diff -u bef aft
--- bef	2018-11-21 18:49:14.677531596 +0100
+++ aft	2018-11-21 18:49:47.557794954 +0100
@@ -1,3 +1,5 @@
+8be4df61-93ca-11d2-aa0d-00e098032b8c-BootNext
+0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-ebfe8df8-dee7-4692-a721-cbcf5095c5cf-0
 eb704011-1402-11d3-8e77-00a0c969723b-MTC
 0b7646a4-6b44-4332-8588-c8998117f2ef-LastBootCurrent
 8be4df61-93ca-11d2-aa0d-00e098032b8c-BootOrder

joakim-tjernlund, 2018-11-21:
efibootmgr -v
BootNext: 0000
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0017,0018,0019,001A,001B,001C,001D,001E,0000
Boot0000* Linux Firmware Updater	PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-25-38-B3-71-BA-F3-6A)/File(\EFI\gentoo\fwupdx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001E  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot001F* USBR BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0020* USBR BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0021* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0022* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0023* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)

joakim-tjernlund, 2018-11-23:
discovered the the F12 BIOS menu displays a Linux Firmware Update entry but
using this just makes the screen go black and then GRUB appears.
I have tried fwupd 1.1.3, 1.1.4  and 1.2.0

joakim-tjernlund, 2018-11-23:
and master fails as well.
@hughsie ?

superm1, 2019-01-14:
@joakim-tjernlund this was fixed in efivar 37.  If you can please retry.  If you still have issues, please open a new issue with fresh information.

closed 2019-01-14 by superm1

joakim-tjernlund, 2019-01-14:
Yes, I known. Already tried and it worked.