fwupd/fwupd #828: fwupd prevents Microsoft Xbox 360 pad from working if it's plugged in after boot

switchwitch, 2018-11-01:
So I've upgraded to Ubuntu 18.10, and now my Switch Pro pad works by default (thanks again), but my Xbox 360 pad (usb) doesn't. It's a normal official Microsoft Xbox 360 pad.

It does work if I plug in the 360 pad before Ubuntu boots and keep it plugged in, but if I unplug it and plug it in again later it won't work. If I don't have it plugged in at boot and plug it in later it also won't work. The indicator ring lights up for a few seconds and then shuts off again and apps won't detect the pad.

If I kill fwupd and then plug in the Xbox 360 pad, it works. If I start fwupd again after that, the pad stops working.

dmesg output:
```shell
[  328.678459] usb 3-1: new full-speed USB device number 9 using xhci_hcd
[  328.831364] usb 3-1: New USB device found, idVendor=045e, idProduct=028e, bcdDevice= 1.14
[  328.831369] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  328.831371] usb 3-1: Product: Controller
[  328.831374] usb 3-1: Manufacturer: ©Microsoft Corporation
[  328.831377] usb 3-1: SerialNumber: 03094FD
[  328.834476] input: Microsoft X-Box 360 pad as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/input/input26
[  329.386951] xpad 3-1:1.0: xpad_try_sending_next_out_packet - usb_submit_urb failed with result -2
[  329.454613] usb 3-1: usbfs: process 3588 (fwupd) did not claim interface 1 before use
```

udevadm monitor output:
```shell
KERNEL[328.553284] add      /devices/pci0000:00/0000:00:14.0/usb3/3-1 (usb)
KERNEL[328.555920] add      /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0 (usb)
KERNEL[328.555967] add      /devices/pci0000:00/0000:00:14.0/usb3/3-1/leds/xpad0 (leds)
KERNEL[328.556127] add      /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/input/input26 (input)
KERNEL[328.556243] add      /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/input/input26/event21 (input)
KERNEL[328.556381] add      /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/input/input26/js0 (input)
KERNEL[328.556446] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0 (usb)
KERNEL[328.556496] add      /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.1 (usb)
KERNEL[328.556641] add      /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.2 (usb)
KERNEL[328.558436] add      /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.3 (usb)
KERNEL[328.558506] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-1 (usb)
UDEV  [329.106604] add      /devices/pci0000:00/0000:00:14.0/usb3/3-1 (usb)
KERNEL[329.108640] remove   /devices/pci0000:00/0000:00:14.0/usb3/3-1/leds/xpad0 (leds)
KERNEL[329.108902] remove   /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/input/input26/event21 (input)
UDEV  [329.109269] add      /devices/pci0000:00/0000:00:14.0/usb3/3-1/leds/xpad0 (leds)
KERNEL[329.144139] remove   /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/input/input26/js0 (input)
KERNEL[329.176159] remove   /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/input/input26 (input)
KERNEL[329.176198] unbind   /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0 (usb)
KERNEL[329.176234] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0 (usb)
KERNEL[329.176273] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.1 (usb)
UDEV  [329.176604] add      /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.3 (usb)
UDEV  [329.176657] add      /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0 (usb)
UDEV  [329.176697] add      /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.1 (usb)
UDEV  [329.176761] add      /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.2 (usb)
UDEV  [329.178949] add      /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/input/input26 (input)
UDEV  [329.180553] add      /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/input/input26/js0 (input)
UDEV  [329.181538] add      /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/input/input26/event21 (input)
UDEV  [329.182709] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0 (usb)
UDEV  [329.183966] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-1 (usb)
UDEV  [329.184945] remove   /devices/pci0000:00/0000:00:14.0/usb3/3-1/leds/xpad0 (leds)
UDEV  [329.185189] remove   /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/input/input26/event21 (input)
UDEV  [329.185266] remove   /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/input/input26/js0 (input)
UDEV  [329.185641] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.1 (usb)
UDEV  [329.186277] remove   /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/input/input26 (input)
UDEV  [329.187049] unbind   /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0 (usb)
UDEV  [329.188032] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0 (usb)
KERNEL[334.176446] unbind   /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0 (usb)
KERNEL[334.176467] unbind   /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.1 (usb)
UDEV  [334.177305] unbind   /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0 (usb)
UDEV  [334.177628] unbind   /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.1 (usb)
```

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.1.2
daemon version:	1.1.2
compile-time dependency versions
	appstream-glib:	0.7.12
	gusb:	0.2.11
	efivar:	34
```

- Operating system and version:
Ubuntu 18.10
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
It was installed by default
- Have you tried rebooting?
Yes
- Are you using an NVMe disk?
No
- Is secure boot enabled (only for the UEFI plugin)?
No

vianpl, 2018-11-07:
Hi same issue here,
It seems that _ebitdo.quirk_ contains the xbox controller VID/PID:
```
# SF30 PRO/SN30 PRO
## Dinput mode (Start + B)
[DeviceInstanceId=USB\VID_2DC8&PID_6000]
Plugin = ebitdo
Flags = none
[DeviceInstanceId=USB\VID_2DC8&PID_6001]
Plugin = ebitdo
Flags = none
## Xinput mode (Start + X)
[DeviceInstanceId=USB\VID_045E&PID_028E] << this are the Xbox controller ids
Plugin = ebitdo
Flags = none
```
Does it even make sense? 8bitdo should have it's own VID isn't it? maybe it's using the Xbox controller chips.
I'd suggest removing that quirk for now since it breaks Xbox controllers.

superm1, 2018-11-07:
The approach i'm thinking is:
1. Add back VID/PID
2. Add a custom flag or quirk that can match to the correct iManufacturer strings for 8bitdo.
3. In the plugin if that flag/quirk found only initialize device with correct string.

vianpl, 2018-11-07:
Sounds good :)

brianredbeard, 2019-01-22:
FYI, because I (independently) stumbled across a bunch of the VID/PID goofiness, I updated the [wiki page](https://github.com/hughsie/fwupd/wiki/8BitDo-Controllers) with a bunch of the various values from both the older NES30/SNES30 and the newer SN30 Pro.

superm1, 2019-10-21:
Thanks for sharing that wiki.  From looking through it there really doesn't seem to be a string that we can match the fake controllers on.  So unfortunately the only option will be to set it to a mode that matches the real 8bitdo VID to do the updates.

closed 2019-10-21 by superm1