fwupd/fwupd #830: Firmware version incompatible to run program.

jparr721, 2018-11-02:
I am having issues getting the program to work for my machine. Every time I run fwupdmgr it always says "Not compatible with firmware version 0.1.22, requires >= 0.1.27" and so I downloaded the latest firmware, but this tool still throws this error even when I am trying to use the file to get the necessary updates. I am not sure how to proceed here.

```shell
$ fwupdmgr --version
client version:	1.1.3
compile-time dependency versions
	appstream-glib:	0.7.13
	gusb:	0.3.0
	efivar:	35
daemon version:	1.1.3

```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
X1 Carbon Thunderbolt Controller
  DeviceId:             693982bd64816a98d01a571fd1b579877b2dc930
  Guid:                 2da42a33-cd30-5ef5-a8fb-2c800a4b760f
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              33.00
  Icon:                 computer
  Created:              2018-11-02

20KHCTO1WW System Firmware
  DeviceId:             1c53551e7da69d896138fac1ae131c83ad46d923
  Guid:                 a4b51dca-8f97-4310-8821-3330f83c9135
  Guid:                 6b3879ff-921f-5bc5-a0c2-5cec28f31d2f
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.22
  VersionLowest:        0.0.1
  Icon:                 computer
  Created:              2018-11-02

UEFI Device Firmware
  DeviceId:             e563ad307df81c99f0de8c26292afd71cf409673
  Guid:                 42a0a96e-c9f3-438f-9687-7826be33e4ce
  Guid:                 375afb87-ea51-5a6e-96d5-c35cc342dc65
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              184.50.3460
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-11-02

UEFI Device Firmware
  DeviceId:             9698faabddf0d7b18925cfbbda95f8b0d0dacc53
  Guid:                 3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
  Guid:                 7090c94a-ee40-593e-91bc-8cfd5a8c0a5d
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.8
  VersionLowest:        0.1.8
  Icon:                 audio-card
  Created:              2018-11-02

SAMSUNG MZVLW256HEHP-000L7
  DeviceId:             04e17fcf7d3de91da49a163ffe4907855c3648be
  Guid:                 5b3df2da-f745-5fd0-81de-5dafd7f0bf8c
  Guid:                 8fd4ca73-d0ae-52e8-8977-461435c6f4cf
  Summary:              NVM Express Solid State Drive
[... 397 lines cut ...]
│       ├── videoinfo.mod
│       ├── video.lst
│       ├── video.mod
│       ├── videotest_checksum.mod
│       ├── videotest.mod
│       ├── xfs.mod
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initramfs-linux-fallback.img
├── initramfs-linux.img
└── vmlinuz-linux 
```

Please answer the following questions:

- Operating system and version: Arch Linux version 4.18.16
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): yay -S fwupd
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

closed 2018-11-14 by jparr721

hughsie, 2018-11-14:
`Not compatible with firmware version 0.1.22, requires >= 0.1.27` means that the firmware Lenovo has uploaded can only be deployed when your firmware is already at a version greater or equal to 0.1.27 -- you have 0.1.22. This is because some old firmware has a bug that causes the flash of the new firmware to fail, and for the laptop to be turned into an expensive **brick** -- so you really don't want to *force* the upgrade. I guess you have one of the early-release X1's with the original factory firmware.

I believe there is a Windows tool that lets you update from 0.1.22 to 0.1.27 -- which then lets you upgrade to newer versions using fwupd automatically. I know using Windows to update the firmware to let you update the firmware using Linux seems slightly odd, and the irony isn't lost on me either.

jparr721, 2018-11-17:
@hughsie I actually managed to flash the new version. Thanks for your comment!

daya, 2019-06-14:
the solution suggested here is based on the assumption that everyone has access to MS Windows which is not true. So how does one get over this hump of reaching a usable version 0.1.27 ? I am stuck at 0.1.25 ... :-1:

hughsie, 2019-06-25:
> Any updates????

No. You need to flah your hardware with something like winflash before we can flash with Linux successfully. This isn't a software bug, this is a firmware limitation.

jparr721, 2019-09-02:
@daya You don't need windows to flash it. You can download an update tool and run it through the bios.

vcavallo, 2021-06-28:
> @daya You don't need windows to flash it. You can download an update tool and run it through the bios.

This is the second vague comment where you say "it's possible" but don't provide any details on how to do it.