fwupd/fwupd #831: Failed to download https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz.asc: Not Found

kaufholdr, 2018-11-04:
Hi,

```
# fwupdmgr refresh
Failed to download https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz.asc: Not Found
```
Seems to be related to #391. Will this be fixed in https://github.com/hughsie/lvfs-website/issues/138 ?

Is it possible to provide a workaround for now?
Please let me know if you need further information from my side.

fwupd version 0.7.4-2.

hughsie, 2018-11-04:
0.7.4 is super old, what distro is this. We stopped mirroring to S3 a few weeks ago, as we've since moved CDNs.

kaufholdr, 2018-11-04:
It's Debian Stretch.

# cat /etc/issue
Debian GNU/Linux 9 \n \l

fwupdate 10-3~bpo9+1 (stretch-backports)

hughsie, 2018-11-04:
I think Debian needs to backport https://github.com/hughsie/fwupd/commit/c1cc0aa35319bd7488690a22ef9482392860549e

kaufholdr, 2018-11-04:
You are right. Thank you for providing the solution. :-)

kaufholdr, 2018-11-04:
The Debian package bug report for this Issue:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=912414

closed 2018-11-04 by kaufholdr

tuxayo, 2019-03-04:
Thanks for report here and on the Debian tracker. For someone having the same issue:
> It's possible to change the config file "/etc/fwupd.conf" and replace:
> DownloadURI=https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz
> with:
> DownloadURI=https://cdn.fwupd.org/downloads/firmware.xml.gz
> as a workaround.

tuxayo, 2019-03-04:
Aditional info: The next Debian (9/Buster) will have version 1.2.4 which isn't affected.

bestouff, 2019-07-29:
FWIW, just changing the configuration file as described still doesn't work:
```
failed to update metadata: Failed to parse XML: Mismatched XML
```

hughsie, 2019-07-29:
> Failed to parse XML: Mismatched XML

You're using a very old and unsupported fwupd version.

bestouff, 2019-07-29:
I know that. But this issue is the first result after a google of the problem, so I'm leaving this comment so that people with servers stuck on Debian stretch know this solution isn't working.

superm1, 2019-07-29:
The cause reported above by @bestouff here is that LVFS grew quite a bit since the ancient fwupd release.  The same thing happened in Ubuntu.  You can reference this bug here: https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1829813

A workaround is to backport this fix if it's impossible to move to a newer fwupd release:
https://github.com/hughsie/fwupd/commit/a1e6e7ff254f3409e0bced0fb400073d6e2b96e8

But I would highly encourage anyone affected by this to move to a newer fwupd release.