fwupd/fwupd #838: Report metadata can contain wrong distribution version

exploide, 2018-11-08:
Today, I upgraded from Fedora 28 to 29. Afterwards, I wanted to see if and how fwupdmgr evolved, so I ran `fwupdmgr get-devices`. Then, it asked me to report one or two successful firmware updates it performed several days or weeks ago. In the printed details, I saw that it lists Fedora 29 for distribution version. However, it certainly was a Fedora 28 when the firmware update happened.

Without knowing much about the internals of fwupd, I guess such details should be written to the prepared report when the update happens, not when the report is sent.

However, I understand that this is only partially possible because an update might reboot the machine. So whether it was successful is only known the next time fwupdmgr is run. But I guess details like the update timestamp are also written on update and not on reporting time.. So maybe it's possible.

Another thing I spotted: This only happened because `fwupdmgr update` (which I run daily) does not ask to submit the report from its last run. It only does when I occasionally run `fwupdmgr get-devices`. Don't know if this is intended behavior, just wanted to mention this.

```shell
$ fwupdmgr --version
client version:	1.1.3
compile-time dependency versions
	appstream-glib:	0.7.12
	gusb:	0.3.0
	efivar:	35
daemon version:	1.1.3
```

hughsie, 2018-11-09:
You're completely right, the `DistroId`, `DistroVersion` and `DistroVariant` are all retrieved at upload time, rather than when the update was scheduled. I'll have a think about this now.

hughsie, 2018-11-09:
Looking more into this, the "Metadata" section is on the root, but I do agree the Distro is a property of the attempt, rather than the upload. I'll work on a fix now, although it'll need an LVFS change too I think.

closed 2018-11-09 by hughsie