fwupd/fwupd #844: Should show the install duration in fwupd.efi

hughsie, 2018-11-12:
Before we start the upgrade, we should show a message like:

   Starting firmware upgrade... This process will take approximately 12 minutes and your computer may restart several times. Please do not power down the computer during the upgrade or permanent damage may occur.

    <wait perhaps 5 seconds...>

This will need us to put the install duration in an EFI variable, or squeeze it in `update_info` somehow.

hughsie, 2018-11-12:
I've not seen a system use the display capsule yet :/

superm1, 2019-10-24:
I'm leaning on we're probably never going to do this.  

At least on some capsule implementations it's not possible to tie a time value to how long the capsule will take to install because fwupd won't know which parts of the capsule actually need to be installed.  Part of installing the capsule involves examining the system and then queuing up the updates.

So I think if there is a desire to show a duration of time to improve the CE of a capsule install - it's better to move this into the OEM's `UpdateCapsule` implementation rather than fwupd.

closed 2019-10-24 by superm1