fwupd/fwupd #845: systemd independed fwupd-1.2.0/1.2.1

ohquait, 2018-11-13:
i will build as rpm's for a systemd-free distribution
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

can't answer they questions, i will build and have not installed fwupd

Please answer the following questions:

- Operating system and version: PCLinuxOS Mate, x86_64
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt-get
- Have you tried rebooting? yes
- Are you using an NVMe disk? no
- Is secure boot enabled (only for the UEFI plugin)? no

What am I doing wrong?
I have a distribution without systemd and can not find the right switch to disable the dependency of systemd in meson-build (even with ninja).
Please is possible to become a link to right place for search?
Thank you!

hughsie, 2018-11-13:
You need to use the `-Dsystemd=false` configure flag. Building and using fwupd without systemd is almost entirely untested and unsupported by upstream.

closed 2018-11-13 by hughsie

ohquait, 2018-11-13:
**Thank you!** i had a syntax-error: -Denable-systemd=false <- can not go good
now with yours is good enough!