fwupd/fwupd #848: fwupd reads 504kB of Thunderbolt firmware in 4300ms

hughsie, 2018-11-14:
At startup fwupd reads back the Thunderbolt firmware from `/sys/*/domain0/0-0/nvm_active0/nvmem` so it can tell if the controller is in native mode. This takes over 4 seconds on my machine -- to transfer just 504Kb of data. Of that data we need just one byte at offset 0x7B, to find out if the controller is in native mode or not. We're using half a meg of heap at startup just for checking one bit, and this represents over a third of the total heap used by fwupd at boot.

I guess I'm asking a few things:

 * Why does it take 4 seconds to read so little data?
 * If it's reading over i2c or SPI, can the kernel cache this? It's just as slow for every start of fwupd.
 * Can we seek to the file stream at the right offsets, rather than just slurping in the entire blob and jumping about reading uint32's and things? Is this going to be faster or is the hardware just going to read the entire blob anyway?
 * Could we just have a trivial sysfs attribute we could read instead?

@westeri @gicmo @superm1 -- is there anyone better suited to answer these questions? Thanks.

YehezkelShB, 2018-11-14:
1. The time it takes is because the initial read from Thunderbolt controller flash memory the kernel does for populating the virtual file in sysfs is done over the control channel to the Thunderbolt host controller and each read operation is of 16 bytes max.
1. The kernel caches it indeed. But it can't cache it when the controller is powered off (as the flash content may have changed in between). If your controller is usually powered off (nothing is connected and the platform config is such that it powers it down when nothing is connected) and fwupd forces it up on start (by the force power plugin), then it explains why you see it each time you start fwupd. If the controller is powered up, I bet you wouldn't see this delay.
1. As this isn't a normal file (it's only a sysfs attribute with behavior similar to a file), it doesn't support seek operations, as much as I remember. So seek will not help, we must read it sequentially from the start, but maybe we can stop the read earlier.
1. I don't think @westeri tends to add such attributes to the kernel, as user-space can get it from parsing the FW (as done today) and kernel APIs are basically to be maintained forever.

hughsie, 2018-11-14:
> then it explains why you see it each time you start fwupd

Ahh! That explains why sometimes it's instant, e.g. if I'm docked v.s. not-docked and using a TBT->DVI dongle.

> it doesn't support seek 

I believe seek() is implemented by the driver and not the generic subsystem, but I can certainly test this and see what happens.

> and kernel APIs are basically to be maintained forever

I think I'd rather keep a kernel sysfs attribute stable than internal firmware address layout :)

YehezkelShB, 2018-11-14:
> I think I'd rather keep a kernel sysfs attribute stable than internal firmware address layout :)

The kernel would have to keep this address layout to provide the attribute...
But I'll let Mika commenting on that.

westeri, 2018-11-14:
In this case since fwupd and especially the image validator already have to know all the details about the image format, I don't think adding a new sysfs attribute is good idea.

hughsie, 2018-11-14:
> I don't think adding a new sysfs attribute is good idea.

Fair enough. In that case, would it be possible to seek to the right place in the input stream instead? IIUC it's a property of the driver not the generic sysfs subsystem.

westeri, 2018-11-14:
I don't think seek is supported in sysfs attributes. But you can read say first 128 bytes and stop there. That would take only two 64B reads over the Thunderbolt fabric.

YehezkelShB, 2018-11-14:
(Oops, yes, it's not 16 bytes each time, it's 16 DWORDs == 64 bytes.)