fwupd/fwupd #854: Shows non-existent update for Dell XPS 15 9570

gtklocker, 2018-11-19:
```shell
$ fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz                                     Downloading…           [***************************************] ] Less than one minute remaining…g…
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

$ fwupdmgr update
Downloading 0.1.1280 for XPS 15 9570 System Firmware...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
failed to run update_prepare() on upower: Cannot install update when not on AC power unless forced
```
Thankfully I was on battery and the update didn't go forward. The version number seemed a little odd so I checked on the website and sure, no such update there. https://fwupd.org/lvfs/device/e4d7bdc4-cc95-4aa0-a982-5e915fc04bf0

Checking on `~/.cache/fwupd` reveals only one update, the one of the actual latest version (1.5.0).

**I have already applied the 1.5.0 update in the past.**

```shell
[master][~/.cache/fwupd]$ ls -1
69dace181f469ac7d2c15868b1f998892888db5d-XPS_9570_BIOS_1.5.0.wu.cab
lvfs-metadata.xml.gz
lvfs-metadata.xml.gz.asc
```

```shell
$ fwupdmgr --version
client version:	1.2.0
compile-time dependency versions
	gusb:	0.3.0
	efivar:	35
daemon version:	1.2.0
```

```shell
$ fwupdmgr get-devices
XPS 15 9570 System Firmware
  DeviceId:             66352e92895dc6d018071ad4ecfae249b0b8af0d
  Guid:                 e4d7bdc4-cc95-4aa0-a982-5e915fc04bf0
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Guid:                 127676e3-b872-5f45-83d2-de29b06d8dfd
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.5.0
  VersionLowest:        0.1.5.0
  Icon:                 computer
  Created:              2018-11-18

KXG50ZNV1T02 NVMe TOSHIBA 1024GB
  DeviceId:             f2759da7fe8e0388c5f3601cb072f837b1070b03
  Guid:                 4d0aed03-a30c-52c6-99e7-a8977797c3d9
  Guid:                 ad9fe8f7-cdc4-52c9-9fea-31b6f4988ffa
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Toshiba America Info Systems
  VendorId:             NVME:0x1179
  Version:              AADA4105
  Icon:                 drive-harddisk
  Created:              2018-11-18

XPS 15 9570 Thunderbolt Controller
  DeviceId:             c4f608cfb780b824ab07304e758612d5979083f0
  Guid:                 560774f3-821b-5ebb-9f1a-97cf34ce266f
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              33.00
  Icon:                 computer
  Created:              2018-11-18
```

Please answer the following questions:

- Operating system and version: Arch Linux, 1.2.0-2
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman
- Have you tried rebooting? No
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

hughsie, 2018-11-19:
0.1.1280 is a very odd number indeed. Can you get the daemon log `fwupd --verbose` when you get the random version shown please?

gtklocker, 2018-11-19:
Upon restarting I can't get it to show me the incorrect version, I stupidly refreshed it by accident and this may have interfered. I'm providing the log, but there's probably nothing useful there https://gist.github.com/gtklocker/a80373b280ae0e794e655e335520bef3.

I'm also uploading a couple of screenshots I managed to get showing the error:
[image]
[image]

I'll update the issue if this reappears, feel free to close it if you don't feel like pursuing it further.

hughsie, 2018-11-19:
Aha! Those are the release notes for the `00.01.05.00` BIOS, and `0500` in hex is `1280` decimal. The get-devices output has the correct `0.1.5.0` In the metadata it's listed as version="66816", which is indeed decimal for `0x10500` -- I was expecting to see `0x00010500` listed there however, no biggie.

When we get a firmware with an unspecified version encoding, we're supposed to fall back to a few heuristics. The firmware doesn't tell us, and so we use the deprecated component ID quirks, which in this case fall back to `com.dell.uefi*` which this firmware matches. I'd be interested if you could try fwupd git master and see if you can reproduce the issue there.

@superm1 I think the better solution to this deprectated fallback is to just set the `LVFS::VersionFormat` to `quad` for all firmwares on the LVFS that match `com.dell.uefi*.firmware` -- then we can remove the quirk in master and make everything a little simpler.

superm1, 2018-11-20:
I thought that was already the policy on the Dell account at LVFS? But yeah otherwise I think that solution makes perfect sense.

closed 2018-11-21 by hughsie