fwupd/fwupd #856: get-devices does not list System Firmware

danbondd, 2018-11-21:
I recently noticed that my machine (Lenovo ThinkPad T480s) was [supported](https://fwupd.org/lvfs/component/1480/all) by fwupd and I attempted to try and update my BIOS to the latest version.

After following the Arch Linux [docs](https://wiki.archlinux.org/index.php/Fwupd), I do not to see an entry for "System Firmware" when running `fwupdmgr get-devices`.

```shell
$ fwupdmgr --version
client version:	1.2.0
compile-time dependency versions
	gusb:	0.3.0
	efivar:	35
daemon version:	1.2.0
```

```shell
$ fwupdmgr get-devices
Thinkpad T480s Thunderbolt Controller
  DeviceId:             44ccf50982c47753b46a9f4311e61e88af98d902
  Guid:                 061f2847-061b-5cfb-9c39-3dbfe2b49d69
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              12.02
  Icon:                 computer
  Created:              2018-11-21

Samsung SSD 960 EVO 250GB
  DeviceId:             c82980cd3fe66f7d0a8bb83b353cbc271177efd4
  Guid:                 5b3df2da-f745-5fd0-81de-5dafd7f0bf8c
  Guid:                 8fd4ca73-d0ae-52e8-8977-461435c6f4cf
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Samsung Electronics Co Ltd
  VendorId:             NVME:0x144D
  Version:              2B7QCXE7
  Icon:                 drive-harddisk
  Created:              2018-11-21
```

Please answer the following questions:

- Operating system and version: `Linux t480s 4.19.2-arch1-1-ARCH #1 SMP PREEMPT Tue Nov 13 21:16:19 UTC 2018 x86_64 GNU/Linux`
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): `pacman`
- Have you tried rebooting? Yes.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled (only for the UEFI plugin)? N/A

danbondd, 2018-11-21:
Apologies in advance if this is a misconfiguration problem rather than an issue!

hughsie, 2018-11-21:
What does `fwupdmgr get-devices --show-all-devices` say?

danbondd, 2018-11-21:
```
$ fwupdmgr get-devices --show-all-devices
Thinkpad T480s Thunderbolt Controller
  DeviceId:             44ccf50982c47753b46a9f4311e61e88af98d902
  Guid:                 061f2847-061b-5cfb-9c39-3dbfe2b49d69
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              12.02
  Icon:                 computer
  Created:              2018-11-21

UHD Graphics 620
  DeviceId:             bbbf1ce3d1cf15550c3760b354592040292415bb
  Guid:                 fed2efa4-7045-55a1-b4fc-b29283d59fe5
  Guid:                 8b72a10c-1279-5f8e-a28a-34fb11a58240
  Guid:                 afb8c4ec-b9a5-5cd3-aaef-cc976ce37eb8
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Version:              07
  Icon:                 audio-card
  Created:              2018-11-21

Samsung SSD 960 EVO 250GB
  DeviceId:             c82980cd3fe66f7d0a8bb83b353cbc271177efd4
  Guid:                 5b3df2da-f745-5fd0-81de-5dafd7f0bf8c
  Guid:                 8fd4ca73-d0ae-52e8-8977-461435c6f4cf
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Samsung Electronics Co Ltd
  VendorId:             NVME:0x144D
  Version:              2B7QCXE7
  Icon:                 drive-harddisk
  Created:              2018-11-21
```

hughsie, 2018-11-22:
Are you booting with UEFI boot, or BIOS legacy boot? Is anything to do with capsule (UEFI) updates turned off in the BIOS?

danbondd, 2018-11-22:
I am using UEFI boot and I noticed that the first an last options in my BIOS settings were disabled.

[image]

I enabled them and I now see System and Device Firmware listed:

```
$ fwupdmgr get-devices
Thinkpad T480s Thunderbolt Controller
  DeviceId:             44ccf50982c47753b46a9f4311e61e88af98d902
  Guid:                 061f2847-061b-5cfb-9c39-3dbfe2b49d69
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              12.02
  Icon:                 computer
  Created:              2018-11-22

20L7CTO1WW System Firmware
  DeviceId:             e7cc234e82d08b6a14e246beb6b5f8142091e570
  Guid:                 ebfe8df8-dee7-4692-a721-cbcf5095c5cf
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Guid:                 2b79e0a0-0801-5fc9-a7a5-90374e924a9b
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.25
  VersionLowest:        0.0.1
  Icon:                 computer
  Created:              2018-11-22

UEFI Device Firmware
  DeviceId:             cf3ad1e5e6838b09cba64bb5ccb1142dd44995d2
  Guid:                 676af093-2a5c-4238-9c29-db8063a33532
  Guid:                 039bd398-f61a-5c46-b02c-b660d1cf4926
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              184.50.3425
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-11-22

UEFI Device Firmware
  DeviceId:             323bda90831520867f8c02544efb9711dfca9fb6
  Guid:                 4d254d6e-cd67-477b-97d5-bc3048af45c4
  Guid:                 cc9cf2fe-ba41-532b-bc58-17ca9cc3c63d
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.8
  VersionLowest:        0.1.8
  Icon:                 audio-card
  Created:              2018-11-22

Samsung SSD 960 EVO 250GB
  DeviceId:             c82980cd3fe66f7d0a8bb83b353cbc271177efd4
  Guid:                 5b3df2da-f745-5fd0-81de-5dafd7f0bf8c
  Guid:                 8fd4ca73-d0ae-52e8-8977-461435c6f4cf
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Samsung Electronics Co Ltd
  VendorId:             NVME:0x144D
  Version:              2B7QCXE7
  Icon:                 drive-harddisk
  Created:              2018-11-22
```

Thank you!

closed 2018-11-22 by danbondd

danbondd, 2018-11-22:
Just updated System Firmware to the latest version without any problems - thanks again for the help!