fwupd/fwupd #858: XPS 9360 device list problem

tapir, 2018-11-26:
```shell
$ fwupdmgr --version
client version:	1.2.0
compile-time dependency versions
	gusb:	0.3.0
	efivar:	35
daemon version:	1.2.0
```

```shell
$ fwupdmgr get-devices
XPS 13 9360 System Firmware
  DeviceId:             3f0ecc7823ee99c718693e8ecca38f54e8738dc5
  Guid:                 5ffdbc0d-f340-441c-a803-8439c8c0ae10
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Guid:                 66d5d415-835d-561c-b687-dcfe4c8b2558
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.2.10.0
  VersionLowest:        0.2.10.0
  Icon:                 computer
  Created:              2018-11-26

PC300 NVMe SK hynix 256GB
  DeviceId:             e11623b2caa18fee292058a5c09ca4e6152f7ecf
  Guid:                 1fc4a944-6ee5-528d-83f2-9c4d1a240f1d
  Guid:                 8c1c105b-8aad-5ad2-9d67-68d143164128
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               SK hynix
  VendorId:             NVME:0x1C5C
  Version:              20005A00
  Icon:                 drive-harddisk
  Created:              2018-11-26

XPS13 9360 Thunderbolt Controller
  DeviceId:             b742f739005ea7435cc0e3a6edf01615680a1527
  Guid:                 7488415f-8db5-5993-a48e-5a43914afb69
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              21.00
  Icon:                 computer
  Created:              2018-11-26
```

- Operating system and version: antergos (arch) latest
- How did you install fwupd: through pacman
- Have you tried rebooting: yes
- Are you using an NVMe disk: yes

I'm not sure this is a bug it just came up because I was looking at some old logs.

I'm using a Dell XPS 9360. In the past I couldn't get fwupdate enabled so I opened up this bug report: https://github.com/rhboot/fwupdate/issues/115

Which turned out it was due to me using a 'msdos' partition table. Fast forward couple months I've switched to a GPT partition table and now everything works OK. I've just updated my system firmware with success.

The problem is, in the old report, I see that my device list has other things that I don't see right now (GPU, TPM etc). It's the same computer so shouldn't I see the same devices?

Here is the old device list for reference:
```shell
Intel AMT [unprovisioned]
  DeviceId:             088df415cdee883ec89563e41e6d495924250174
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              11.8.50
  VersionBootloader:    11.8.50
  Icon:                 computer
  Created:              2018-06-04

XPS 13 9360 TPM 2.0
  DeviceId:             3368246a16d77193e42a2fea8d95439b38f01670
  Guid:                 2e513587-0d9b-50e6-8e15-1845d44e88eb
  Summary:              Platform TPM device
  Plugin:               dell
  Flags:                internal|require-ac|registered
  Vendor:               Dell Inc.
  Version:              1.3.2.8
  Icon:                 computer
  Created:              2018-06-04

UHD Graphics 620
  DeviceId:             8de6c7959053fd5798006dcc63590d33fa5e51cb
  Guid:                 8eb8bd2e-0fca-5aba-9aa8-f341e0aa4482
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Icon:                 audio-card
  Created:              2018-06-04
```

Thanks

superm1, 2018-11-26:
Newer versions of fwupdmgr don't show all the devices if they're not upgradable by default.

You can just append `--show-all-devices` to the command line to see all those other devices.

tapir, 2018-11-26:
:+1:  thanks

closed 2018-11-26 by tapir