fwupd/fwupd #862: Provide fake firmware updates to test UI such as GNOME Software

8none1, 2018-11-28:
On Ubuntu, and this is likely to be a common occurance cross-distro, we want to be able to test updates to fwupd from the perspective of the end-user.  To that end when we have bug fixes to SRU we need to perform some basic smoke testing to see that the fixes are reflected in the way we expected.

For example:  https://bugs.launchpad.net/ubuntu/+source/gnome-software/+bug/1719797
Comment 16 is what I'm trying to test again with a new version.  

That is:

After a fw update had been "applied" (from G-S's POV anyway) clicking refresh would cause the fw update to appear again in the list of available updates, giving the impression that it hadn't been applied/prepared for being applied.
This was fixed, but now I can't test it without there being a firmware update available and showing in the G-S UI.

The CI tests don't make this easy for me as person-in-the-street to test.

It would be great if fwupd could provide a fake fw update which reacts in the same was as a real one, but without all that tedious mucking about writing to flash ;)

superm1, 2018-11-28:
So I think this is a duplicate of https://github.com/hughsie/fwupd/issues/809

I guess the biggest piece missing from existing test plugin is having the payloads that match it available at LVFS or an easy way to enable the local source.

hughsie, 2018-11-28:
Ohh, I suppose it is a dupe. Red Hat QA were also asking for approximately the same functionality, so I might have a play and see what I can do.

closed 2018-11-29 by superm1