fwupd/fwupd #874: After update, get-updates says there is still an update

kuviman, 2018-12-01:
Using Arch, tried to update firmware with fwupd. It looks like it is updated, but `fwupdmgr get-updates` still shows that updates are available.

```shell
$ fwupdmgr --version
client version:	1.2.1
compile-time dependency versions
	gusb:	0.3.0
	efivar:	35
daemon version:	1.2.1

$ fwupdmgr get-devices
Inspiron 15 7000 Gaming System Firmware
  DeviceId:             befb2b3b58f507bd11e848ddb5f48ecbe1766265
  Guid:                 1610b70e-2db1-41d0-bcaf-19ad38fff695
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Guid:                 cb8cd134-67ef-5366-a78a-8dd3bb171cf2
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.8.1
  VersionLowest:        0.1.8.1
  Icon:                 computer
  Created:              2018-12-01
  UpdateState:          success

$ fwupdmgr get-updates
Inspiron 15 7000 Gaming System Firmware has firmware updates:
GUID:                    1610b70e-2db1-41d0-bcaf-19ad38fff695
GUID:                    230c8b18-8d9b-53ec-838b-6cfc0383493a
GUID:                    cb8cd134-67ef-5366-a78a-8dd3bb171cf2
ID:                      com.dell.uefi1610b70e.firmware
Update Version:          0.1.2049
Update Name:             Inspiron 14 7000 Gaming/Inspiron 15 7000 Gaming System Update
Update Summary:          Firmware for the Dell Inspiron 14 7000 Gaming/Inspiron 15 7000 Gaming
Update Remote ID:        lvfs
Update Duration:         49710 days
Update Checksum:         SHA1(746e1aceffe43573f4a99ce5d8ca223682289791)
Update Location:         https://fwupd.org/downloads/34fa885b4e9f300b0ba6d0db70936947901444b7-firmware_Inspiron_14_15_7000_Gaming_1.8.1.cab
Update Description:      This stable release fixes the following issues:
                         
                         Some new functionality has also been added:
                         
ID:                      com.dell.uefi1610b70e.firmware
Update Version:          0.1.1792
Update Name:             Inspiron 14 7000 Gaming/Inspiron 15 7000 Gaming System Update
Update Summary:          Firmware for the Dell Inspiron 14 7000 Gaming/Inspiron 15 7000 Gaming
Update Remote ID:        lvfs
Update Duration:         49710 days
Update Checksum:         SHA1(493d8a993a5f752168704441adf460b299982969)
Update Location:         https://fwupd.org/downloads/5651ce42b6bdafff3125e597fbad7293c876dce2-firmware_Inspiron_14_15_7000_Gaming_Firelord%20BIOS%201.7.0.cab
ID:                      com.dell.uefi1610b70e.firmware
Update Version:          0.1.1536
Update Name:             Inspiron 14 7000 Gaming/Inspiron 15 7000 Gaming System Update
Update Summary:          Firmware for the Dell Inspiron 14 7000 Gaming/Inspiron 15 7000 Gaming
Update Remote ID:        lvfs
Update Duration:         49710 days
Update Checksum:         SHA1(ad1d6bca530296742a4044fd85bc4c7be8d4ff14)
Update Location:         https://fwupd.org/downloads/d67a65a8d2e09575afb75f11a064814c0a27d921-firmware_Inspiron_14_15_7000_Gaming_1.6.0.wu.cab
ID:                      com.dell.uefi1610b70e.firmware
Update Version:          0.1.256
Update Name:             Inspiron 14 7000 Gaming/Inspiron 15 7000 Gaming System Update
Update Summary:          Firmware for the Dell Inspiron 14 7000 Gaming/Inspiron 15 7000 Gaming
Update Remote ID:        lvfs
Update Duration:         49710 days
Update Checksum:         SHA1(5858977978c66676ba32e8351164cd5d19dc570b)
Update Location:         https://fwupd.org/downloads/9c0a882fd36ebd3ac8a8ab4253c9ffcc2111ba00-firmware.cab
```

superm1, 2018-12-01:
Sounds like a duplicate of https://github.com/hughsie/fwupd/issues/854 

Have you rerun refresh?

hahnjo, 2018-12-01:
I'm also seeing the very same thing after refreshing twice on my XPS 13 9350:
```
XPS 13 9350 System Firmware has firmware updates:
GUID:                    33773727-8ee7-4d81-9fa0-57e8d889e1fa
GUID:                    230c8b18-8d9b-53ec-838b-6cfc0383493a
GUID:                    583b55f5-1f4f-5991-aace-f82163f7babd
ID:                      com.dell.uefi33773727.firmware
Update Version:          0.1.1792
Update Name:             XPS 13 9350 System Update
Update Summary:          Firmware for the Dell XPS 13 9350
Update Remote ID:        lvfs
Update Duration:         49710 days
Update Checksum:         SHA1(d3deafd3651ed257de671f9aa16bd1d650c0e6d5)
Update Location:         https://fwupd.org/downloads/73e11e9f713523e26f79a00ea2588dc9d7e64c16-firmware.cab
ID:                      com.dell.uefi33773727.firmware
Update Version:          0.1.1537
Update Name:             XPS 13 9350 System Update
Update Summary:          Firmware for the Dell XPS 13 9350
Update Remote ID:        lvfs
Update Duration:         49710 days
Update Checksum:         SHA1(6440be13470c639fa553de7deb5d939f5f561409)
Update Location:         https://fwupd.org/downloads/657f113f6f026135caa140c3d766dc828924d26f-Signed_1152921504627265278.cab
ID:                      com.dell.uefi33773727.firmware
Update Version:          0.1.1281
Update Name:             XPS 13 9350 System Update
Update Summary:          Firmware for the Dell XPS 13 9350
Update Remote ID:        lvfs
Update Duration:         49710 days
Update Checksum:         SHA1(1bb95787c07612137e6cd00ea0bfde61b9a7b7a9)
Update Location:         https://fwupd.org/downloads/8f8215fdf6bb569f98c6ff76b5a48c168b158df8-firmware_XPS_9350_Bios_1.5.1.wu.cab
ID:                      com.dell.uefi33773727.firmware
Update Version:          0.1.1042
Update Name:             XPS 13 9350 System Update
Update Summary:          Firmware for the Dell XPS 13 9350
Update Remote ID:        lvfs
Update Duration:         49710 days
Update Checksum:         SHA1(4bd36e439f854416f4716f3a5ab75166d8c408f1)
Update Location:         https://fwupd.org/downloads/01be31ce4749ea73bcd3bf6405b117f7da4d77f6-firmware_XPS_9350_Bios_1.4.18.cab
ID:                      com.dell.uefi33773727.firmware
Update Version:          0.1.1041
Update Name:             XPS 13 9350 System Update
Update Summary:          Firmware for the Dell XPS 13 9350
Update Remote ID:        lvfs
Update Duration:         49710 days
Update Checksum:         SHA1(871a20d049ad744f601f682705ae3f7986990874)
Update Location:         https://fwupd.org/downloads/ea2aa51a9db18a3b980d6fa8f4472aca6b3757eb-XPS_9350_1.4.17.cab
ID:                      com.dell.uefi33773727.firmware
Update Version:          0.1.1037
Update Name:             XPS 13 9350 System Update
Update Summary:          Firmware for the Dell XPS 13 9350
Update Remote ID:        lvfs
Update Duration:         49710 days
Update Checksum:         SHA1(7ac3d5b84fba286c5096fc295a34cb2b1eac52e3)
Update Location:         https://fwupd.org/downloads/7adcf7f3d0857d6c0de0c567653ec0a70b225666-XPS_9350_BIOS_1.4.13.wu.cab
ID:                      com.dell.uefi33773727.firmware
Update Version:          0.1.1034
Update Name:             XPS 13 9350 System Update
Update Summary:          Firmware for the Dell XPS 13 9350
Update Remote ID:        lvfs
Update Duration:         49710 days
Update Checksum:         SHA1(8d1a2cf95d061371f93bc789d7eb14fd830b0171)
Update Location:         https://fwupd.org/downloads/f443e68f2130cca6d13f70c31078f3877fbd75ab-XPS_9350_1.4.10_BIOS.wu.cab
ID:                      com.dell.uefi33773727.firmware
Update Version:          0.1.1028
Update Name:             XPS 13 9350 System Update
Update Summary:          Firmware for the Dell XPS 13 9350
Update Remote ID:        lvfs
Update Duration:         49710 days
Update Checksum:         SHA1(7992d453f35844ca392ee0218bab8b29fc9e2f05)
Update Location:         https://fwupd.org/downloads/d6d3a1e2101044a5ac02a7a691e7c4b732d4311e-XPS_9350_1.4.4_BIOS.wu.cab
[... 12 lines cut ...]
Update Name:             XPS 13 9350 System Update
Update Summary:          Firmware for the Dell XPS 13 9350
Update Remote ID:        lvfs
Update Duration:         49710 days
Update Checksum:         SHA1(cb38891e7eccd5290070e375f96ad233836e4aa1)
Update Location:         https://fwupd.org/downloads/2b6e0ae88720a33e36c3ff75adc95511202b5971-firmware.cab
ID:                      com.dell.uefi33773727.firmware
Update Version:          0.1.265
Update Name:             XPS 13 9350 System Update
Update Summary:          Firmware for the Dell XPS 13 9350
Update Remote ID:        lvfs
Update Duration:         49710 days
Update Checksum:         SHA1(a56430fae2f21f4ff0fd6081f74217b34ed56fd5)
Update Location:         https://fwupd.org/downloads/430eeb74eb7b682afe75d89b141cb66cd7aca466-firmware.cab
ID:                      com.dell.uefi33773727.firmware
Update Version:          0.1.263
Update Name:             XPS 13 9350 System Update
Update Summary:          Firmware for the Dell XPS 13 9350
Update Remote ID:        lvfs
Update Duration:         49710 days
Update Checksum:         SHA1(3c4c6615f64e3c50fd1940bad1f57439c696d96a)
Update Location:         https://fwupd.org/downloads/3ef300e6eb15d8673cdad2703020203f8184aba6-firmware.cab
```

Interestingly if I manually download the file for `0.1.1792` (the "newest" version), it's indeed 1.7.0 that I already have installed.

kuviman, 2018-12-01:
Refresh does not help, the update can be installed but it is the same thing as already installed

Maybe there is wrong version specified in the update meta, different from the one in the update file?

MicroJoe, 2018-12-01:
Same issue on Dell XPS 9360. It took me 6 reboots and firmware updates to realize that I was in a loop. The "Update Version" displays 0.2.2560 but the installed version is "2.10.0" which is latest. The difference in the version numbers is confusing.

xcaptain, 2018-12-02:
Same issue here rebooted several times and can't install version `0.1.1280`

sparky005, 2018-12-02:
Similar issue here on the XPS 9370 using Manjaro. It's looping trying to install `0.1.1539`, which I believe it successfully installed once as `0.1.6.3`. If I'm reading the linked issue correctly, the issue is that fwupd is not correctly parsing the super weird version numbers that Dell is providing. Is that right?

superm1, 2018-12-03:
The version number string is just a large integer.  How it's interpreted is OEM specific.  You can see that fwupd is properly interpreted as AA.BB.CC.DD.

In the newer fwupd version a quirk was removed to automatically format metadata this way, and by default it's showing up as AA.BB.CCDD (leading to this behavior).  The linked bug report indicated that this was supposed to be fixed on the LVFS side to explicitly set the metadata format properly, but it seems to not be working correct.

For now any versions with only 2 dots (eg 0.1.1234) mean that this failure is still happening.

superm1, 2018-12-03:
The update duration is also a little bit comical with default metadata right now, that's a separate bug..
```
Update Duration:         49710 days
```

sparky005, 2018-12-03:
I noticed that too, I thought I might just be a bit behind on updates ;) haha. Thanks for clarifying though. I guess I can rest assured that I am not crazy!

hughsie, 2018-12-03:
> Update Duration:         49710 days

Odd. I can't reproduce that here.

> SHA1(d3deafd3651ed257de671f9aa16bd1d650c0e6d5)

I also see this in the metadata:

    <metadata>
      <value key="LVFS::VersionFormat">quad</value>
    </metadata>

@kuviman -- can you try 1.2.2 please?

hughsie, 2018-12-03:
> can you try 1.2.2 please?

Ignore me, I've not tagged 1.2.2 yet -- Monday pre-coffee. I'll try on an XPS13 now and see if I can reproduce either of these. Can anyone supply a `fwupd --verbose` daemon log in case not?

closed 2018-12-03 by hughsie

gentoochainsaw, 2018-12-03:
Got the wrong version numbers on an XPS9550 with fwupd 1.2.1 (Gentoo). I accepted the upgrades and let them run; they completed without error (not making any real changes) and the version numbers have been okay since.
Ran a fwupdmgr refresh and no phantom upgrades resulted. Did some on-IRC troubleshooting with @hughsie on this. Seems fine now.

hahnjo, 2018-12-03:
I can confirm that `fwupd` isn't showing wrong upgrades anymore (even though it's a bit sad that Dell hasn't uploaded 1.9.0 for my XPS 13 9350 yet...)