fwupd/fwupd #879: Failed to run update_attach() on thunderbolt: failed to read nvm_authenticate: Invalid argument

steevee, 2018-12-03:
```shell
$ fwupdmgr --version
client version:    1.1.4
checkout info:    1.1.4
compile-time dependency versions
    appstream-glib:    0.7.9
    gusb:    0.3.0
    efivar:    35
daemon version:    1.1.4
```

```shell
$ fwupdmgr get-devices
PM951 NVMe SAMSUNG 256GB
  DeviceId:             e11623b2caa18fee292058a5c09ca4e6152f7ecf
  Guid:                 cce2386f-6d68-5f85-9f3f-ebead9288696
  Guid:                 77bb4022-9503-586c-99b7-c8a64502da30
  Guid:                 8fd4ca73-d0ae-52e8-8977-461435c6f4cf
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  VendorId:             NVME:0x144D
  Version:              BXV77D0Q
  Icon:                 drive-harddisk
  Created:              2018-12-03

XPS 13 9350 Thunderbolt Controller
  DeviceId:             210d24afec6b138edd9cd49bbd545bde1194ddd7
  Guid:                 dbbe0ed9-df91-529c-ac86-8976b78f726d
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|supported|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              16.00
  Icon:                 computer
  Created:              2018-12-03
```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001
Boot0000* Windows Boot Manager    HD(1,GPT,bae71ac8-d6f9-4f1c-b186-d4f5eff56d1a,0x800,0xfa000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* ubuntu    HD(1,GPT,725626e4-d9f1-4c50-87c1-ca9d956d69c4,0x800,0x5f000)/File(\EFI\ubuntu\shimx64.efi)
```

```shell
$ efivar -l | grep fw
N/A
```

```shell
$ tree /boot
N/A
```

Please answer the following questions:

- Operating system and version: Ubuntu 17.10
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): snap
- Have you tried rebooting? Yes
- Are you using an NVMe disk? No
- Is secure boot enabled (only for the UEFI plugin)? n/a

# Problem
(copied from https://github.com/dell/thunderbolt-nvm-linux/issues/18)

Received error when trying to update FW on an XPS 13 9350 running Ubuntu 17.10 (Kernel 4.15.8) that originally came with Windows installed (system id: 0x0704). 

I'm yet to restart and wondering how to rectify (downgrading FW is an option!) (Given issues in #17 I'm worried about getting another `could not start ICM firmware` if I restart!)

### Steps taken:

1. Noticed older thunderbolt firmware version:
  ```
  XPS 13 9350 Thunderbolt Controller
    DeviceID:             cf030000-0000-8f08-a208-32ca02c4d016
    Guid:                 dbbe0ed9-df91-529c-ac86-8976b78f726d
    Description:          <p>Updating the thunderbolt NVM improves performance and stability.</p>
    Plugin:               thunderbolt
    Flags:                internal|updatable|supported|registered
    DeviceVendor:         DELL
    DeviceVendorId:       TBT:0x00D4
    Version:              12.0
    Created:              2018-12-03
  ```

2. `fwdupdmgr refresh` was giving me errors `Failed to download https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz: Not Found` so I updated `fwupdate` from 0.97 to 1.1.4 via snap.

3. Refreshed fwupdate metadata, ran update, and received a similar error to reported above:
  ```bash
  $ sudo fwupd.fwupdmgr refresh 
  Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
  Downloading…           [*                                      ] Less than one minute remaDownloading…           
  [***************************************]
  Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

  $ sudo fwupd.fwupdmgr update
  Downloading 16.00 for XPS 13 9350 Thunderbolt Controller...
  Fetching firmware https://fwupd.org/downloads/75984aeb95d33ed7fab857b18b28ffe317cb870d-8FDJ2_NVM16.00.cab
  Downloading…           [***************************************]
  Decompressing…         [***************************************]
  Authenticating…        [***************************************]
  Updating XPS 13 9350 Thunderbolt Controller…                   ]
  Restarting device…     [***************************************]failed to run update_attach() on thunderbolt: failed to read 'nvm_authenticate: Invalid argument
  ```
4. `dmesg -w` [output during FW update](https://gist.github.com/steevee/97770b8d2946f824e807fde1b8b20f41)

5. `cat /sys/bus/thunderbolt/devices/0-0/nvm_authenticate` reads `0x0`

6. Device still appears in `fwupdate` and works with displayport, but no longer registers when unplugging a thunderbolt-network
```bash
  sudo fwupd.fwupdmgr get-devices
  XPS 13 9350 Thunderbolt Controller
    DeviceId:             210d24afec6b138edd9cd49bbd545bde1194ddd7
    Guid:                 dbbe0ed9-df91-529c-ac86-8976b78f726d
    Summary:              Unmatched performance for high-speed I/O
    Plugin:               thunderbolt
    Flags:                internal|updatable|supported|registered
    Vendor:               Dell
    VendorId:             TBT:0x00D4
    Version:              16.00
    Icon:                 computer
    Created:              2018-12-03
    UpdateError:          failed to run update_attach() on thunderbolt: failed to read 'nvm_authenticate: Invalid argument
```

7. Have tried manually reinstalling, but always get same error `fwupd.fwupdmgr install ~/Downloads/d09142be32326f9aa588bff45cd1f5fe7136b369-ThunderboltFirmwareUpdateLinux_4.26.13.002.cab` (with/without TBT device ID)

## Additional info
Nothing unusual in `journalctl -u snap.fwupd.fwupd.service`

Was resolved upon reboot (no error message and new version of thunderbolt FW apparently working), but error message was cause for concern.

superm1, 2018-12-03:
Well so that error is happening because of a failure to parse the output of reading `nvm_authenticate` in base 16.  It looks like the kernel isn't doing anything particularly to sanitize the value from the DMA read but it does take a mutex whenever it's being written and read.

I haven't seen this locally myself in tests.  So my best guess here is that this read of `nvm_authenticate` happened too quickly.  Perhaps in `update_attach` it would be best to try to read a few times in this situation.
@gicmo @YehezkelShB @westeri you guys agree?

westeri, 2018-12-03:
Yes, the kernel does not do anything else than pass the raw error value in that attribute when it is read. It is always prefixed with `0x` and anything `!= 0x0` is an error code from the authentication process. Not sure where the "Invalid argument" error is coming, though

superm1, 2018-12-03:
Looking closer though that error is supposed to be for outside of the valid range for the **base**.  16 is certainly valid.  So maybe it is just that `errno` wasn't cleared from an earlier call.

I think we should just remove that particular check since it's hardcoded to 16.

westeri, 2018-12-03:
Indeed, strtoull() returns EINVAL if the base is not correct so my vote goes to removing that particular check since 16 is as you say a valid base :-)