fwupd/fwupd #884: Versioning firmware in metadata.xml vs fwupd

artur-rs, 2018-12-04:
Hello, I came across some problems with versioning in metadata.xml. I have added firmware version w/o leading "v" (e.g. 4.8.0.3) to the LVFS, but `fwupdmgr get-devices` outputs version v4.8.0.3, so automatically `fwupdmgr get-updates` outputs all uploaded firmware as possible updates.

```shell
$ fwupdmgr get-devices
LT1000
  DeviceId:             ebd74a0d326a36f55ee97436cc967187c66819dc
  Guid:                 52b68c34-6b31-5ecc-8a5c-de37e666ccd5
  Plugin:               flashrom
  Flags:                internal|updatable|supported|registered
  Vendor:               Libretrend
  Version:              v4.8.0.3
  Created:              2018-12-04
```
```shell
$ fwupdmgr get-updates
LT1000 has firmware updates:
GUID:                    52b68c34-6b31-5ecc-8a5c-de37e666ccd5
ID:                      com.Libretrend.LT1000.firmware
Update Version:          4.8.0.3
Update Name:             LT1000 Coreboot Firmware
Update Summary:          Firmware for the Libretrend LT1000 platform
Update Remote ID:        3mdeb-embargo
Update Checksum:         SHA1(c6d33477cd2c7d97a936efbb6b33dcc21b52431c)
Update Location:         https://fwupd.org/downloads/59c0738d765dea9e3e275ff8e27e1c1650bde8db-com.Lib
Update Description:      This release features:
                         
ID:                      com.Libretrend.LT1000.firmware
Update Version:          4.8.0.2
Update Name:             LT1000 Coreboot Firmware
Update Summary:          Firmware for the Libretrend LT1000 platform
Update Remote ID:        3mdeb-embargo
Update Checksum:         SHA1(cfa6e954206c58625db684b53212ee557ed058a5)
Update Location:         https://fwupd.org/downloads/058413d4f4935f070d2f23ac9f04a1e0e96474aa-com.Lib
Update Description:      This release adds integration with new Intel FSP.
```
Additional info:
- system info: Ubuntu 18.04 LTS bionic
- fwupd installed from source
- debug output (--verbose) from get-updates: https://hastebin.com/liwiluyaca.sql

```shell
$ fwupdmgr --version
client version:	1.2.2
checkout info:	1.0.8-559-gc62f5be0
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.2.2
```

superm1, 2018-12-04:
flashrom plugin sets the version using SMBIOS strings: https://github.com/hughsie/fwupd/blob/master/plugins/flashrom/fu-plugin-flashrom.c#L78

So I suspect that if you read the string from  `/sys/class/dmi/id/bios_version` you'll get the same result with the leading "v".

Do you have an opportunity to fix this in BIOS?  If so, that's the best way to me.  If not, my take on how to solve this is to introduce a quirk for forcing a numeric version (strip letters) and when you add your system to `flashrom.quirks` apply that quirk to it.

In the plugin, recognize that quirk and sanitize the string.

Btw - while you are introducing a quirk for this characteristic you should probably also quirk a few other things:
1) the extra arguments that were needed for https://github.com/3mdeb/fwupd/pull/2/commits/935b28f529c298c7beaa1fd38f59b1bfeec0a01a too.
2) Does that require a minimum version of `flashrom` to support that new flag?  If so you should quirk making sure that minimum version is installed (and remove `updatable` flag and set `UpdateError` if it's not)

hughsie, 2018-12-04:
My initial thought was just "_fix it automatically in fu-smbios.c_" but that's going to change the HWIDs in `fu-hwids.c` and that would be a Bad Thing™. I'm not sure if we should just remove the leading `v` in `fu-plugin-flashrom.c` as a per-plugin quirk, or even in `fu-device.c` (in `fu_device_set_version()`) if we want to support this quirk in all plugins. I'm also not sure if we want to remove `ver`, `V`, `VER` or just anything that's not numeric and dots. I'm erring on a "global" `FU_QUIRK_VERSION_FIXUP_TO_SEMVER` or something.

> minimum version of flashrom to support that new flag

I think the short-to-medium term plan is for 3mdeb to contribute the missing libflashrom, and we can make everything a lot more sane. Part of that might be adding more quirks to control using the new API, but I'm happy to help there. Porting to libflashrom would be a good first step.

superm1, 2018-12-04:
> My initial thought was just "fix it automatically in fu-smbios.c" but that's going to change the HWIDs in fu-hwids.c and that would be a Bad Thing™.

Yeah that was my first thought too and the reason I didn't think it would work either.

> I'm also not sure if we want to remove ver, V, VER or just anything that's not numeric and dots.

Some devices do use alphanumeric versions (like NVME storage).  

> I'm erring on a "global" FU_QUIRK_VERSION_FIXUP_TO_SEMVER or something.

OK if you think it should be global that's fine to me.

> Porting to libflashrom would be a good first step

Ah I see, wasn't aware that was the eventual plan.  Sounds good to me.  Not worth over-engineering fwupd fixes before that.

hughsie, 2018-12-04:
> if you think it should be global that's fine to me

I think it's something that's not flashrom-specific. What about a good name? `FIXUP_TO_SEMVER` or `REPAIR_VERSION` or `VERSION_FORCE_NUMERIC` or something else? The idea is we'd filter out any chars that are not `[0-9]` and `.` and use that instead.

superm1, 2018-12-04:
I like `FORCE_NUMERIC_VERSION`.  It leaves the possibility to introduce other version fixups later without changing the meaning of this quirk.

hughsie, 2018-12-05:
I began to implement this just now, and if we want to support future version fixups we probably want something like `VersionFixups`, much like we have `VersionFormat`. That way we could have `VersionFixups=force-numeric,ignore-leading-zeros,convert-from-hex` or something like that in the future. This got me thinking, so rather than devices having two quirks, e.g.

    VersionFormat=quad
    VersionFixups=force-numeric

Could we just say that `VersionFormat=quad` strips away the non-semver chars? It would be very strange indeed to have a version like `1.abc.3.4` where we want to treat that as a semver. I could imagine a `001.abc.003.004` but that would be `plain` as we can resort to using `strcmp` to compare versions. This of course means we can't do things like the fictional `ignore-leading-zeros,convert-from-hex` in the future, but maybe those don't make sense at all if I'm struggling to invent other examples.

closed 2018-12-05 by superm1