fwupd/fwupd #891: Boot order lock prevents updates

davidstrauss, 2018-12-06:
I recently diagnosed why updates were failing on my ThinkPad T580 with Fedora 29. Everything in the diagnostics looked fine. In fact, they *were fine*, as the problem was elsewhere. Having Boot Order Lock enabled in my firmware/BIOS configuration apparently reset the UEFI boot order and removed the one-time changes by fwupd before the subsequent boot. The fact that it did this was silent.

Changes seemed to be allowed to the boot order:
```shell
$ efibootmgr -v
[straussd@t580 ~]$ efibootmgr -v
BootNext: 0001
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0010,0011,0012,0013,0017,0018,0019,001A,001B,001C,001D,001E,001F,0001
Boot0000* Fedora	HD(1,GPT,5efce5ad-b6f2-4b75-af83-14bea923cc22,0x800,0x64000)/File(\EFI\fedora\shimx64.efi)
Boot0001* Linux Firmware Updater	HD(1,GPT,5efce5ad-b6f2-4b75-af83-14bea923cc22,0x800,0x64000)/File(\EFI\fedora\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017  USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018  USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019  NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A  NVMe1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a401)
Boot001B  ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001C* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001D  PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001E  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001F  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0020* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,1,0)
Boot0021* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,0,0)
Boot0022* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0023* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
```

It's just that my firmware would reset the configuration (with no warning) on each subsequent boot.

I'd consider one or more of these to mitigate the issue:
- Update the issue template to ask users if they have any boot order locking enabled.
- If the system has rebooted into the main OS since fwupd made the boot order change and the change hasn't applied, warn the user that Boot Order Lock may be interfering with applying the update.
- Perform the switch at the GRUB (or similar) level and chainload the updater. I feel like this would be a regression in the architecture, though.

superm1, 2018-12-06:
I'd say "most" people don't know to come here to file issues.  A good mitigation should be something that happens without people needing to know to come here.

I'd wonder is there any hints that the OS can get from the BIOS through some interface that boot order lock is enabled?  If so, then it would make sense to use them and mark the UEFI device as not updatable and set `UpdateError`.  Or even better is there a way to toggle boot order lock from OS via any interface?

davidstrauss, 2018-12-07:
> I'd say "most" people don't know to come here to file issues. A good mitigation should be something that happens without people needing to know to come here.

Agreed. I just wanted to be open to improvements anywhere along the way that would have saved me time.

> I'd wonder is there any hints that the OS can get from the BIOS through some interface that boot order lock is enabled?

I'm not sure on this one.

> Or even better is there a way to toggle boot order lock from OS via any interface?

I'd guess this isn't possible, as the purpose of the feature seems to be preventing manipulation of the boot order from the OS.

superm1, 2019-02-28:
I'm going to close this as won't fix.  Not much that can be done unless the vendor provides an API to detect this or to turn it off.

closed 2019-02-28 by superm1

betrcode, 2019-08-07:
@davidstrauss Thanks for writing this issue. I had a similar issue on my Lenovo Thinkpad X1 (20KH006JMX). I noticed that the firmware updates were not applied after reboot. `sudo fwupdmgr get-history` reported `UpdateState: needs-reboot`. 

After reading this issue, I checked my BIOS Boot Order. The "LinuxFirmwareUpdate" was **last** in the boot order in BIOS and was therefore never applied. However, unlike you, the "Boot Order Lock" was **disabled**.

Solution: I moved the "LinuxFirmwareUpdate" to the top of the Boot Order list, saved and rebooted. Then the firmware updates were applied.

FilipLaurentiu, 2025-02-20:
I have a similar issue with my ThinkStation P5. I use Arch and recently I was notified that I have 2 updates, one for the BIOS and the other for the  Intel ME. Both of them fail to install when I restart the machine.
I manage to download the ISO image for the BIOS update and install it manually, but I don't have any ISO image on the Lenovo page for the other one.
I checked in my BIOS and I don't have an `BIOS order lock` option, and neither `LinuxFirmwareUpdate` in the Boot Order List.
Is there another 'feature' that block the update ?

The only error I could find is in the `systemctl` logs

> Jan 16 17:44:48 filip-thinkstation fwupd[3786]: 15:44:48.721 FuDevice             disabling polling: error polling Bolt receiver: failed to receive: error condition
> Jan 20 17:07:33 filip-thinkstation fwupd[3766]: 15:07:33.214 FuDevice             disabling polling: error polling Bolt receiver: failed to receive: error condition
>

superm1, 2025-02-20:
Please open your own new issue.