fwupd/fwupd #893: fwupdmgr not compatible with firmware version 0.1.25

msto, 2018-12-07:
Hi,

I recently installed Fedora 29 on a Lenovo Thinkpad X1 Carbon. I read a few suggestions online to update the firmware after installation to fix an issue with suspend, however I'm encountering the below error when trying to update with `fwupdmgr`.

```
$ fwupdmgr get-updates
Not compatible with firmware version 0.1.25, requires >= 0.1.27
$ fwupdmgr update
Not compatible with firmware version 0.1.25, requires >= 0.1.27
```
Do you know how I might fix this? It looks like I need to somehow update the firmware without fwupd to a version that supports fwupd. I kept a windows partition when installing, so I tried rebooting into Windows and running system update there, but it didn't fix the issue.

Please let me know if you have any suggestions or if there's a better venue for this question elsewhere. Thank you!

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.1.4
compile-time dependency versions
	appstream-glib:	0.7.14
	gusb:	0.3.0
	efivar:	35
daemon version:	1.1.4
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
X1 Carbon Thunderbolt Controller
  DeviceId:             681db4cd121d88f9af5eb6dcda2a68c2f23e65ef
  Guid:                 2da42a33-cd30-5ef5-a8fb-2c800a4b760f
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              33.00
  Icon:                 computer
  Created:              2018-12-07

20KG0022US System Firmware
  DeviceId:             1c53551e7da69d896138fac1ae131c83ad46d923
  Guid:                 a4b51dca-8f97-4310-8821-3330f83c9135
  Guid:                 6b3879ff-921f-5bc5-a0c2-5cec28f31d2f
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.25
  VersionLowest:        0.0.1
  Icon:                 computer
  Created:              2018-12-07

UEFI Device Firmware
  DeviceId:             29749712e93b58730ecd1b8537a5503ff5b7fb27
  Guid:                 9c9d9769-32fa-4841-b550-ea998e754e99
  Guid:                 1a4eea87-292a-5d99-82bd-fd244d4e0194
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              184.55.3510
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-12-07

UEFI Device Firmware
  DeviceId:             9698faabddf0d7b18925cfbbda95f8b0d0dacc53
  Guid:                 3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
  Guid:                 7090c94a-ee40-593e-91bc-8cfd5a8c0a5d
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.8
  VersionLowest:        0.1.8
  Icon:                 audio-card
  Created:              2018-12-07

SAMSUNG MZVLB512HAJQ-000L7
  DeviceId:             04e17fcf7d3de91da49a163ffe4907855c3648be
  Guid:                 47335265-a509-51f7-841e-1c94911af66b
  Guid:                 8fd4ca73-d0ae-52e8-8977-461435c6f4cf
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Samsung Electronics Co Ltd
  VendorId:             NVME:0x144D
  Version:              4L2QEXA7
  Icon:                 drive-harddisk
  Created:              2018-12-07
```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000,0018,0019,001A,001B,001C,001D,001E,001F,0024
Boot0000* Windows Boot Manager	HD(1,GPT,cbc5903d-4432-48d5-a490-be474c8b6ac4,0x800,0x82000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* Fedora	HD(1,GPT,cbc5903d-4432-48d5-a490-be474c8b6ac4,0x800,0x82000)/File(\EFI\fedora\shimx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Regulatory Information	FvFile(478c92a0-2622-42b7-a65d-5894169e4d24)
Boot0015  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0016  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0017  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0018* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0019* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot001A* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001B* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f601)
Boot001C* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001D* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001E  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001F  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0020* USBR BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0021* USBR BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0022* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0023* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0024* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
```

```shell
# Was not already installed; no output after installing
$ efivar -l | grep fw
bash: efivar: command not found...
Install package 'efivar' to provide command 'efivar'? [N/y] y

Proceed with changes? [N/y] y

$ efivar -l | grep fw
```

```shell
$ tree /boot
/boot
├── config-4.18.16-300.fc29.x86_64
├── config-4.19.6-300.fc29.x86_64
├── efi [error opening dir]
├── elf-memtest86+-5.01
├── extlinux
│   ├── cat.c32
│   ├── chain.c32
│   ├── cmd.c32
│   ├── cmenu.c32
│   ├── config.c32
│   ├── cptime.c32
│   ├── cpu.c32
│   ├── cpuid.c32
│   ├── cpuidtest.c32
│   ├── debug.c32
│   ├── dhcp.c32
│   ├── dir.c32
│   ├── disk.c32
│   ├── dmi.c32
│   ├── dmitest.c32
│   ├── elf.c32
│   ├── ethersel.c32
│   ├── gfxboot.c32
│   ├── gpxecmd.c32
│   ├── hdt.c32
│   ├── hexdump.c32
│   ├── host.c32
│   ├── ifcpu64.c32
│   ├── ifcpu.c32
│   ├── ifmemdsk.c32
│   ├── ifplop.c32
│   ├── kbdmap.c32
│   ├── kontron_wdt.c32
│   ├── ldlinux.c32
│   ├── lfs.c32
│   ├── libcom32.c32
│   ├── libgpl.c32
│   ├── liblua.c32
│   ├── libmenu.c32
│   ├── libutil.c32
│   ├── linux.c32
│   ├── ls.c32
│   ├── lua.c32
│   ├── mboot.c32
│   ├── memdisk
│   ├── meminfo.c32
│   ├── menu.c32
│   ├── pci.c32
│   ├── pcitest.c32
│   ├── pmload.c32
│   ├── poweroff.c32
│   ├── prdhcp.c32
│   ├── pwd.c32
│   ├── pxechn.c32
│   ├── reboot.c32
│   ├── rosh.c32
│   ├── sanboot.c32
│   ├── sdi.c32
│   ├── sysdump.c32
│   ├── syslinux.c32
│   ├── vesa.c32
│   ├── vesainfo.c32
│   ├── vesamenu.c32
│   ├── vpdtest.c32
│   ├── whichsys.c32
│   └── zzjson.c32
├── grub2 [error opening dir]
├── initramfs-0-rescue-2256ea179e4d4de08491c13c56f4391d.img
├── initramfs-4.18.16-300.fc29.x86_64.img
├── initramfs-4.19.6-300.fc29.x86_64.img
├── loader
│   └── entries [error opening dir]
├── lost+found [error opening dir]
├── memtest86+-5.01
├── System.map-4.18.16-300.fc29.x86_64
├── System.map-4.19.6-300.fc29.x86_64
├── vmlinuz-0-rescue-2256ea179e4d4de08491c13c56f4391d
├── vmlinuz-4.18.16-300.fc29.x86_64
└── vmlinuz-4.19.6-300.fc29.x86_64
```

Please answer the following questions:

- Operating system and version: Fedora 29
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): Came with installation of Fedora
- Have you tried rebooting? Yes. Rebooted, rebooted into windows and tried updating from there, rebooted back to linux, still no luck
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No, I disabled it before installing Fedora

msto, 2018-12-07:
Yes, this solved it. 
```$ fwupdmgr get-updates
No upgrades for 20KG0022US System Firmware, current is 0.1.34: 0.1.27=older, 0.1.28=older, 0.1.30=older, 0.1.31=older, 0.1.34=same
```

If anyone else encounters the same problem, I originally tried running the Windows system update, which doesn't include firmware updates. Recent Lenovo machines include an app called "Lenovo Vantage"; after you skip through some setup and data requests the landing page has a "system update" option, where you can find recent BIOS updates.

closed 2018-12-07 by msto

daya, 2019-06-14:
I don't think its right to Close this issue, since not everyone has a handy Windows OS, so there is no chance to run "Lenovo Vantage" 

What's the alternative solution for folks with access to just Linux OS?

Xeddius, 2019-11-06:
This issue affects other devices and makes fwupd un-usable on systems that don't have the bleeding edge builds of fwupd even if only a single device update requires a more recent version. It's really frustrating but the devs will not elect to fix the issues or add a workaround as it's "up to the distro" to keep everything up to date (Even though Linux Mint 19.2 is Ubuntu based and very up-to-date). I would like a flag like --skip to skip any updates that can't be applied so I can at least update my game controllers and mouse. I have a friend a few states away that came to visit and while he was on the same version of Mint and fwupd his system didn't have any devices which required newer fwupd and I was able to update some of my things.

This is a serious headache for me and I'd love if a workaround could be added. (The offending device on my machine is a Broadcom wireless card that can't be easily removed.)

hughsie, 2019-11-06:
> but the devs will not elect to fix the issues

It's not a fwupd issue. The original issue is a firmware bug in older versions of Lenovo firmware which means using capsule updates. If you update the version in Windows or with a live CD then you can update just fine with an old version of fwupd. 

> even if only a single device update requires a more recent version

If that's the case please open a new issue with full details.