fwupd/fwupd #895: Dell XPS 15 9570 - firmware update used to work, but now fwupdx64.efi can't locate device handle

madsl, 2018-12-11:
[image]
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version: 1.2.1
compile-time dependency versions
        gusb:   0.2.11
        efivar: 36
daemon version: 1.2.1
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
XPS 15 9570 System Firmware
  DeviceId:             66352e92895dc6d018071ad4ecfae249b0b8af0d
  Guid:                 e4d7bdc4-cc95-4aa0-a982-5e915fc04bf0
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Guid:                 127676e3-b872-5f45-83d2-de29b06d8dfd
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.5.0
  VersionLowest:        0.1.5.0
  Icon:                 computer
  Created:              2018-12-11
  UpdateState:          success

XPS 15 9570 TPM 2.0
  DeviceId:             f80c30caae6a1e4a4f8b403e53e5c726320e224c
  Guid:                 abe4e2d2-3fd5-5f0c-9613-7a04e8e048a2
  Summary:              Platform TPM device
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Dell Inc.
  Version:              7.2.0.1
  Icon:                 computer
  Created:              2018-12-11

KXG50ZNV1T02 NVMe TOSHIBA 1024GB
  DeviceId:             f2759da7fe8e0388c5f3601cb072f837b1070b03
  Guid:                 4d0aed03-a30c-52c6-99e7-a8977797c3d9
  Guid:                 ad9fe8f7-cdc4-52c9-9fea-31b6f4988ffa
  Serial:               78PS12Y9T8KQ
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Toshiba America Info Systems
  VendorId:             NVME:0x1179
  Version:              AADA4105
  Icon:                 drive-harddisk
  Created:              2018-12-11

XPS 15 9570 Thunderbolt Controller
  DeviceId:             3138dc1a2fe54de1b9b27321993abb82d7bb364f
  Guid:                 560774f3-821b-5ebb-9f1a-97cf34ce266f
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              33.00
  Icon:                 computer
  Created:              2018-12-11
```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0000,0001,0002
Boot0000* Gentoo        PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-08-0D-02-00-4B-A3-0C)/HD(1,GPT,b96a0957-2dd6-4b04-92fa-b4ca4cf8dbcc,0x28,0x64000)/File(\vmlinuz.efi)
Boot0001* UEFI shell    PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-08-0D-02-00-4B-A3-0C)/HD(1,GPT,b96a0957-2dd6-4b04-92fa-b4ca4cf8dbcc,0x28,0x64000)/File(\shellx64.efi)
Boot0002* Linux Firmware Updater        PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-08-0D-02-00-4B-A3-0C)/File(\efi\EFI\gentoo\fwupdx64.efi)
**This is only required if you use the UEFI plugin**
```

```shell
$ efivar -l | grep -i fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-FWUPDATE_DEBUG_LOG
0abba7dc-e516-4167-bbf5-4d9d1c739416-FWUPDATE_VERBOSE
**This is only required if you use the UEFI plugin**
```

```shell
$ find /boot
/boot
/boot/shellx64.efi
/boot/.keep
/boot/vmlinuz-4.19.8-gentoo
/boot/System.map-4.19.8-gentoo
/boot/config-4.19.8-gentoo
/boot/efi
/boot/efi/EFI
/boot/efi/EFI/gentoo
/boot/efi/EFI/gentoo/fw
/boot/efi/EFI/gentoo/fw/fwupd-e4d7bdc4-cc95-4aa0-a982-5e915fc04bf0.cap
/boot/efi/EFI/gentoo/fwupx64.efi
/boot/efi/EFI/gentoo/fwupdx64.efi
/boot/vmlinuz-4.19.6-gentoo
/boot/System.map-4.19.6-gentoo
/boot/config-4.19.6-gentoo
/boot/vmlinuz.efi
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version: gentoo
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): portage
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? no

superm1, 2018-12-11:
Bug in efivar 36 I would expect. Can you please move to 37?

madsl, 2018-12-11:
You were right. Updating to efivar-37 makes it work. Thanks!

closed 2018-12-11 by superm1