fwupd/fwupd #900: Add support for blacklisting individual firmware items

superm1, 2018-12-14:
Right now fwupd supports blacklisting whole plugins and whole devices from firmware upgrades.  This is a big hammer that (potentially) prevents security updates from flowing in (permanently).

When someone downgrades to a firmware, there is usually a good reason for doing so (IE a bug).  So fwupd should offer to blacklist the "current" or "last" firmware during a downgrade command.  This allows preventing upgrades back to that same firmware, but more importantly lets future upgrades still be offered.

hughsie, 2018-12-14:
I guess we have to store this somewhere, I guess the history database might make sense. e.g. a "pinned" boolean on the last downgrade which bubbles up to the session as a flag, or something.

superm1, 2020-08-10:
closed in https://github.com/fwupd/fwupd/pull/2305

closed 2020-08-10 by superm1