fwupd/fwupd #906: Lenovo ThinkPad X1 Yoga 1st: after BIOS update, custom logo is discarded

Depau, 2018-12-24:
I used to have my custom logo image, set by using the official ISO image. I used to extract it using geteltorito, copy it to a fat32-formatted partition on a USB drive and added `logo.gif` under `/Flash`.

Once it's installed this way, when the BIOS is further updated, the installer asks whether you want to keep the currently installed logo (even if you recreate the USB drive without the logo).

I noticed Lenovo uploaded a firmware image some days ago and decided to install it using `fwupdmgr` (from GNOME Software). The image was flashed successfully (except for a minor bug I'll describe in another issue), however the custom logo was discarded without asking me whether I wanted to keep it.

I think the default option should be to keep any installed logos: if somebody installed one, they probably still want it and know how to remove it manually if needed.

---

```shell
$ fwupdmgr --versionclient version:	1.2.1
compile-time dependency versions
	gusb:	0.3.0
	efivar:	35
daemon version:	1.2.1
```

```shell
$ fwupdmgr get-devices
20FRS0070D System Firmware
  DeviceId:             c66f3753f0a35b3874d65546d4b7b10896d44ce2
  Guid:                 81cba30b-c2d4-4e66-bcf3-69d81ba890b0
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Guid:                 078d2ed4-fd5c-5fc7-a488-092cfab7db2f
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.40
  VersionLowest:        0.1.31
  Icon:                 computer
  Created:              2018-12-24

UEFI Device Firmware
  DeviceId:             6f92d954409bb3c145828fbbc0346c3ca12f2a9a
  Guid:                 cb3c1682-5386-4591-bd29-7c1f441b7ccb
  Guid:                 663b9134-3d22-582a-809f-4433a9cc50cb
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.55.3510
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-12-24

UEFI Device Firmware
  DeviceId:             8f659ac2737a491fb05138ae0de6f5e05d949998
  Guid:                 4c2e5b5c-6467-43af-afeb-7bc72d96b9c3
  Guid:                 1f713eb1-ef3d-5890-9c50-45fdd5e5ca95
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.18
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-12-24
```

```shell
$ efibootmgr -v
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0002,0001,0017,0018,0019,001A,001B,001C
Boot0000* rEFInd Boot Manager	HD(1,GPT,7543c7f0-d985-4f2f-9320-e4197535c01b,0x800,0x82000)/File(\EFI\refind\refind_x64.efi)
Boot0001* Windows Boot Manager	HD(1,GPT,7543c7f0-d985-4f2f-9320-e4197535c01b,0x800,0x82000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0002* arch	HD(1,GPT,7543c7f0-d985-4f2f-9320-e4197535c01b,0x800,0x82000)/File(\EFI\arch\grubx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,1,0)
Boot001E* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,0,0)
Boot001F* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0020* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
```

```shell
$ efivar -l | grep fw
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
```

```shell
$ tree /boot
(skipped because too much junk, and it's kinda unrelated. I'll add it if needed)
```

Please answer the following questions:

- Operating system and version: Arch Linux, rolling
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman
- Have you tried rebooting? not relevant
- Are you using an NVMe disk? nope
- Is secure boot enabled (only for the UEFI plugin)? nope

hughsie, 2019-01-01:
I've written to Lenovo asking for a comment about this functionality. I'll let you know when there's something public I can share. Thanks.

t-8ch, 2019-01-04:
It would also be nice to be able to set a logo in the first place.
Otherwise users would have to reflash the current EFI version manually to apply the logo.
(They may have forgotten how to do so after getting used to the convenience of fwupd)

evacchi, 2019-05-11:
ping, sorry to bump this up, Fedora 30 released flickerfree boot officially, and it would be nice to customize the EFI image :-)

hughsie, 2019-05-11:
Still nothing back from Lenovo about this.

t-8ch, 2019-09-23:
/unstale

hughsie, 2019-09-23:
I've pinged Lenovo about this just now.

superm1, 2019-09-23:
@hughsie our "missing firmware" repos, what do you think about just moving stuff that's obviously a firmware bug over to one of those as needed?

hughsie, 2019-09-23:
I did that myself just now. I think it's the right thing to do.