fwupd/fwupd #909: After upgrade to 1.2.2, fwupd failed to start for SQL error

Xuanwo, 2019-01-02:
```shell
$ fwupdmgr --version
client version: 1.2.2
compile-time dependency versions
        gusb:   0.3.0
        efivar: 35
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
```

```shell
$ fwupdmgr get-devices
Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached
```

Please answer the following questions:

- Operating system and version: Archlinux
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes

`fwupd.service` failed to start for following error:

```
Jan 02 10:00:02 thinkpad-x1-carbon systemd[1]: Starting Firmware update daemon...
Jan 02 10:00:02 thinkpad-x1-carbon fwupd[16536]: Failed to get PCR0s: missing executable tpm2_pcrlist in PATH
Jan 02 10:00:02 thinkpad-x1-carbon fwupd[16536]: Failed to load engine: Failed to prepare SQL: no such column: protocol
Jan 02 10:00:03 thinkpad-x1-carbon systemd[1]: fwupd.service: Main process exited, code=exited, status=1/FAILURE
Jan 02 10:00:03 thinkpad-x1-carbon systemd[1]: fwupd.service: Failed with result 'exit-code'.
Jan 02 10:00:03 thinkpad-x1-carbon systemd[1]: Failed to start Firmware update daemon.
```

hughsie, 2019-01-02:
Can you upload /var/lib/fwupd/pending.db to this issue please.

Xuanwo, 2019-01-02:
[pending.db.gz](https://github.com/hughsie/fwupd/files/2720365/pending.db.gz)

This is the `pending.db` with gzip ( *github do not allow *.db attchment.* )

kaihendry, 2019-01-02:
https://s.natalian.org/2019-01-02/pending.db

Nikki1993, 2019-01-02:
[pending.db.tar.gz](https://github.com/hughsie/fwupd/files/2720735/pending.db.tar.gz)

gardar, 2019-01-02:
Same issue for me on Arch, removing the /var/lib/fwupd/pending.db seems to solve the issue.

joakim-tjernlund, 2019-01-02:
I didn't save a copy of pending.db, sorry.

philmmanjaro, 2019-01-02:
@hughsie: your commit seems to fix the issue. https://forum.manjaro.org/t/70565/5

closed 2019-01-04 by hughsie

swidz, 2019-01-05:
$ fwupdmgr --version
client version:	1.2.3
compile-time dependency versions
	gusb:	0.2.9
	efivar:	34
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.freedesktop.fwupd': timed out

fwupdmgr get-devices
Error calling StartServiceByName for org.freedesktop.fwupd: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.freedesktop.fwupd': timed out

after deleting the pending.db file which in my case was not located in /var/lib/fwupd but in /var/local/lib/fwupd 

$ fwupdmgr --version
client version:	1.2.3
compile-time dependency versions
	gusb:	0.2.9
	efivar:	34

Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached

But the file pending.db is getting recreated in /var/local/lib/fwupd 

Please answer the following questions:

    Operating system and version: Debian 9.6 
    How did you install fwupd: build with meson and ninja
    Have you tried rebooting? Yes
    Are you using an NVMe disk? Yes

[pending.db.zip](https://github.com/hughsie/fwupd/files/2729939/pending.db.zip)