fwupd/fwupd #910: fwupd needs pango ?

joakim-tjernlund, 2019-01-02:
Seem like fwupd needs pango, see:
 https://bugs.gentoo.org/671096
Seems a bit odd that a cli/daemon service needs X11 related pkgs ?

hughsie, 2019-01-02:
You need pango if you want to generate the BMPs required for the firmware upgrade splash.

closed 2019-01-02 by hughsie

joakim-tjernlund, 2019-01-02:
I see, would be nice if BMP support is configurable. A mininal system will not have X11 at all.

hughsie, 2019-01-02:
You only need pango support for _building_ fwupd, on the runtime deployment it's not required.

joakim-tjernlund, 2019-01-02:
I see, but on Gentoo one builds pkg from source so one still needs X11.

hughsie, 2019-01-02:
Right, I don't think it makes sense to build the system using the dependencies you're going to be using for deployment. fwupd requires gcc for building, but not for runtime.